PowerBroker2/ELMduino #3: can it run on ESP32?

vfebert, 2020-02-10:
Hi,
was really happy to find this one, intending to do a "My own CO2 footprint" project.
Could you please help me in providing an Arduino example running on an ESP32 - which has Bluetooth build in.  When I compile the sample sketch, it simply gives "serial3 unknown" error. Sorry for being only a hobby programmer....

sabifa, 2020-02-10:
+1 would also like to know :)

I've tried to combine the example code here and your posted comment in [this thread](https://forum.arduino.cc/index.php?topic=563742.0#7) and come up with this:
```c++
#include "BluetoothSerial.h"
#include "ELMduino.h"

#define ELM_BT SerialBT
#define ESP_BLUETOOTH_NAME "ESP32"

BluetoothSerial SerialBT;
ELM327 myELM327;

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  Serial.begin(115200);
  ELM_BT.begin(ESP_BLUETOOTH_NAME, true);

  Serial.println("Attempting to connect to ELM327...");

  if (!ELM_BT.connect("OBDII"))
  {
    Serial.println("Couldn't connect to OBD scanner");
    while (1);
  }

  myELM327.begin(ELM_PORT);

  Serial.println("Connected to ELM327");
  Serial.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  Serial.println("Type and send commands/queries to your ELM327 through the serial monitor");
  Serial.println();
}

float rpm = 0;

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = tempRPM;
  }
  else
  {
    Serial.print(F("\tERROR: "));
    Serial.println(myELM327.status);
    delay(100);
  }
}
````

I couldn't test it yet because I do not have my esp32 and OBD adapter here yet but maybe you can test it and report back @vfebert.

I guess we have to replace `ELM_BT.connect("OBDII")` with the actual MAC address of the adapter.

PowerBroker2, 2020-02-10:
> When I compile the sample sketch, it simply gives "serial3 unknown" error.

Yeah, the current example isn't very portable and isn't general enough. I'll update the example to be easier to understand and add a separate example for the ESP32.

> I guess we have to replace ELM_BT.connect("OBDII") with the actual MAC address of the adapter.

Actually, no. The [<BluetoothSerial>.connect() method is overloaded](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/src/BluetoothSerial.h) to handle several different input arguments. If the input is a String, the ESP32 will decipher the input as the remote device's name. If the input is a uint8_t array, it will decipher the input as the remote device's MAC address. The name of my ELM327 happens to be "OBDII" and I figured it would be easiest to use the name as opposed to trying to find the ELM327's MAC address.

In the end, it's all up to how you want to connect to the ELM327.

PowerBroker2, 2020-02-10:
I haven't done extensive testing of the library in it's current state, but I should be able to get some testing done before the end of the week...

PowerBroker2, 2020-02-10:
Should be fixed in [release 1.1.4](https://github.com/PowerBroker2/ELMduino/releases/tag/1.1.4)

vfebert, 2020-02-11:
thanks so much guys - looking forward to 1.1.4  and promise to do some testing as much as my time allows.

pniewiadowski, 2020-02-16:
I was trying to test this today.
The connection with BT OBD2 was successful but then I've got some errors while trying to read the response.

This is what I got on serial monitor:
`19:05:21.539 -> ets Jun  8 2016 00:22:57
19:05:21.539 -> 
19:05:21.539 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:05:21.539 -> configsip: 0, SPIWP:0xee
19:05:21.539 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:05:21.539 -> mode:DIO, clock div:1
19:05:21.539 -> load:0x3fff0018,len:4
19:05:21.539 -> load:0x3fff001c,len:1216
19:05:21.539 -> ho 0 tail 12 room 4
19:05:21.539 -> load:0x40078000,len:9720
19:05:21.573 -> ho 0 tail 12 room 4
19:05:21.573 -> load:0x40080400,len:6352
19:05:21.573 -> entry 0x400806b8
19:05:23.706 -> Attempting to connect to ELM327...
19:05:49.302 -> Connected to ELM327
19:05:49.302 -> Ensure your serial monitor line ending is set to 'Carriage Return'
19:05:49.302 -> Type and send commands/queries to your ELM327 through the serial monitor
19:05:49.302 -> 
19:05:49.302 -> Received: ?>
19:05:49.302 -> RPM: 0
19:05:49.302 -> Header NOT found
`

Do you know what can it be?

PowerBroker2, 2020-02-16:
@pniewiadowski Can you post your entire sketch?

Regardless, thanks for testing - I'll do some debugging myself today and/or tomorrow and get back to all of you.

PowerBroker2, 2020-02-16:
I did some testing today and **was able** to get both my ESP32 to connect to my ELM327 and successfully query telemetry (RPM and speed).

Although I didn't have to change any parsing logic, I did do some updates to the library. If you're still having trouble, please try reinstalling/updating the library, post your entire sketch, and post example printouts from the Serial Monitor.

Sowmya-Sukhanand, 2020-02-17:
@PowerBroker2 Hii,even i tried today updating the library and then running the code with ESP32 and OBD2 via bluetooth MAC address.Iam not able to get the output.It is just printing(In Debug Mode)

Received:>?
HEADER NOT FOUND
RPM:0

Please help me with this!!!

PowerBroker2, 2020-02-17:
You seem to be connecting to the ELM327 fine, but it doesn't seem to understand the rpm query for some reason. Can you use a Bluetooth Serial Monitor app (android or laptop) to connect to the ELM327 and try the RPM query manually?

I don't have the problem with my device...strange

PowerBroker2, 2020-02-17:
On second thought, you can use this instead to access the ELM327 through the Serial Monitor without a third party app:

``` C++
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

#define DEBUG_PORT Serial
#define ELM_PORT   SerialBT

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("ESP32test", true);

  DEBUG_PORT.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);
  }

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}

void loop()
{
  if(DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();

    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }

  if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if(c == '>')
      DEBUG_PORT.println();

    DEBUG_PORT.write(c);
  }
}```

pniewiadowski, 2020-02-17:
The first sketch I was testing came from this repo example:
https://raw.githubusercontent.com/PowerBroker2/ELMduino/master/examples/ESP32_test/ESP32_test.ino
This is the output of serial monitor:

19:05:21.539 -> ets Jun 8 2016 00:22:57
19:05:21.539 ->
19:05:21.539 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:05:21.539 -> configsip: 0, SPIWP:0xee
19:05:21.539 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:05:21.539 -> mode:DIO, clock div:1
19:05:21.539 -> load:0x3fff0018,len:4
19:05:21.539 -> load:0x3fff001c,len:1216
19:05:21.539 -> ho 0 tail 12 room 4
19:05:21.539 -> load:0x40078000,len:9720
19:05:21.573 -> ho 0 tail 12 room 4
19:05:21.573 -> load:0x40080400,len:6352
19:05:21.573 -> entry 0x400806b8
19:05:23.706 -> Attempting to connect to ELM327...
19:05:49.302 -> Connected to ELM327
19:05:49.302 -> Ensure your serial monitor line ending is set to 'Carriage Return'
19:05:49.302 -> Type and send commands/queries to your ELM327 through the serial monitor
19:05:49.302 ->
19:05:49.302 -> Received: ?>
19:05:49.302 -> RPM: 0
19:05:49.302 -> Header NOT found

And it is keep looping display over and over same text:
19:05:49.302 -> Received: ?>
19:05:49.302 -> RPM: 0
19:05:49.302 -> Header NOT found

Now I took the above sketch:
https://github.com/PowerBroker2/ELMduino/issues/3#issuecomment-587134687
and this is what I am able to see in serial monitor:

21:23:49.401 -> >ets Jun  8 2016 00:22:57
21:24:52.205 -> 
21:24:52.205 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:24:52.205 -> configsip: 0, SPIWP:0xee
21:24:52.205 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:24:52.240 -> mode:DIO, clock div:1
21:24:52.240 -> load:0x3fff0018,len:4
21:24:52.240 -> load:0x3fff001c,len:1216
21:24:52.240 -> ho 0 tail 12 room 4
21:24:52.240 -> load:0x40078000,len:9720
21:24:52.240 -> ho 0 tail 12 room 4
21:24:52.240 -> load:0x40080400,len:6352
21:24:52.240 -> entry 0x400806b8
21:24:54.375 -> Attempting to connect to ELM327...
21:25:16.826 -> Connected to ELM327
21:25:16.826 -> Ensure your serial monitor line ending is set to 'Carriage Return'
21:25:16.863 -> Type and send commands/queries to your ELM327 through the serial monitor
21:25:16.863 -> 

PS. I repeated test with example ESP32_test.ino sketch, after updating to latest ELMduino library - same situation.
PS2. In both sketches I have to comment out this two lines:
```
//pinMode(LED_BUILTIN, OUTPUT);
//digitalWrite(LED_BUILTIN, HIGH);
```

PowerBroker2, 2020-02-17:
Hmm, interesting.

Can you try sending the following commands with [this sketch](https://github.com/PowerBroker2/ELMduino/issues/3#issuecomment-587134687)?:

- AT E0
- AT SP 0
- 010D

Entering 010C into the serial monitor should return something like "010D 00" where 00 is your speed in hex in kph. Try that and let me know the results.

Also, make sure your serial monitor is in carriage return line ending mode!

Sowmya-Sukhanand, 2020-02-18:
Thanks for the reply @PowerBroker2 .
My question is that the header file <ELMduino.h> itself is not detected.So what can be done for that?

PowerBroker2, 2020-02-18:
Then it means you don't have the library installed...

Sowmya-Sukhanand, 2020-02-18:
I do have the library installed with the latest version 1.1.7
[image]

PowerBroker2, 2020-02-18:
In that case the header file <ELMduino.h> _is_ detected and compiled. You might need to go into ELMduino.cpp to add some print statements to figure out what's going on. Without being there with you to test on your equipment there's not much I can do. It works with my ESP32 with my ELM327. If it's not working with your setup I'll need more info as to what exactly is going on with your hardware. For instance, is the library actually sending the AT commands? Can you confirm this with serial prints? Can you confirm the AT commands are formed properly? What is the exact response of the ELM327 for each AT command? Can you confirm it responds with "OK>"? etc, etc...

I'm more than happy to adjust the library so that all can easily use it, but I need very detailed testing results when I can't replicate the errors.

Sowmya-Sukhanand, 2020-02-19:
There is no problem with the hardware,both esp32 as well as elm327 because they connect well via bluetooth also whenever i send AT commands via serial monitor to elm327 it responds appropriately.
Anyways i'll confirm with the serial prints in the library to check if it actually sending the commands or no.

vfebert, 2020-02-23:
Hi,  rainy sunday in Germany, means a bit of time to test. Purchased an original VEEPEAK BLE, installed 1.18, used the latest sketch from the above. Good news: got a connection, but unfortunately no ">" promt, so no answer to "AT E0" - happy to receive any comment that might help a hobby programmer..

pniewiadowski, 2020-02-23:
@PowerBroker2 sorry for late response.
I took the sketch you told me and this is the output of the serial monitor.

`ets Jun  8 2016 00:22:57
18:41:34.434 ->
18:41:34.434 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:41:34.434 -> configsip: 0, SPIWP:0xee
18:41:34.434 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:41:34.478 -> mode:DIO, clock div:1
18:41:34.478 -> load:0x3fff0018,len:4
18:41:34.478 -> load:0x3fff001c,len:1216
18:41:34.478 -> ho 0 tail 12 room 4
18:41:34.478 -> load:0x40078000,len:9720
18:41:34.478 -> ho 0 tail 12 room 4
18:41:34.478 -> load:0x40080400,len:6352
18:41:34.478 -> entry 0x400806b8
18:41:36.645 -> Attempting to connect to ELM327...
18:42:01.471 -> Connected to ELM327
18:42:01.471 -> Ensure your serial monitor line ending is set to 'Carriage Return'
18:42:01.471 -> Type and send commands/queries to your ELM327 through the serial monitor
18:42:01.471 ->
18:42:22.251 -> AT E0
OK

18:42:22.290 -> >AT SP 0
OK

18:42:32.296 -> >010D
SEARCHING...
UNABLE TO CONNECT

18:42:41.259 -> >010D
SEARCHING...
UNABLE TO CONNECT`

PowerBroker2, 2020-02-23:
@vfebert Since your OBD scanner module is a BLE version, you'll have to use the ESP32's BLE capability. Currently, my example is only for normal Bluetooth serial and will not work for BLE. That being said, I'm sure the ESP32 can make that BLE connection, I'll just have to do a little research on how that works. I hope that helps.

@pniewiadowski Did you do that test while the car was fully on and running?

PowerBroker2, 2020-02-23:
@vfebert Take a look at [this tutorial on ESP32 BLE](https://randomnerdtutorials.com/esp32-bluetooth-low-energy-ble-arduino-ide/). Since I don't have a BLE OBD scanner I can't do any real testing to modify my examples. Since you have one of those modules, can you do the testing to figure out how to get the ESP32 to connect to your BLE OBD scanner?

pniewiadowski, 2020-02-24:
@PowerBroker2 yes, you were right. I have switch the ignition on and now I have this:

19:26:41.863 -> ets Jun  8 2016 00:22:57
19:26:41.863 ->
19:26:41.863 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:26:41.863 -> configsip: 0, SPIWP:0xee
19:26:41.863 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:26:41.863 -> mode:DIO, clock div:1
19:26:41.863 -> load:0x3fff0018,len:4
19:26:41.863 -> load:0x3fff001c,len:1216
19:26:41.863 -> ho 0 tail 12 room 4
19:26:41.863 -> load:0x40078000,len:9720
19:26:41.863 -> ho 0 tail 12 room 4
19:26:41.863 -> load:0x40080400,len:6352
19:26:41.900 -> entry 0x400806b8
19:26:44.031 -> Attempting to connect to ELM327...
19:27:09.312 -> Connected to ELM327
19:27:09.312 -> Ensure your serial monitor line ending is set to 'Carriage Return'
19:27:09.312 -> Type and send commands/queries to your ELM327 through the serial monitor
19:27:09.312 ->
19:27:15.089 -> AT E0
OK

19:27:15.089 -> >AT SP0
OK

19:27:18.881 -> >010D
SEARCHING...
7E8 03 41 0D 00

Seems it is connecting and receiving data.

miguelos6, 2020-02-26:
@vfebert I think this project is using BLE
https://hackaday.io/project/167968-rear-car-speedometer

miguelos6, 2020-02-28:
@vfebert I think this project is using BLE
https://hackaday.io/project/167968-rear-car-speedometer

@pniewiadomski - that's an output from Serial Monitor sketch, right ? did you manage to get results from example sketch using RPM ?

@PowerBroker2 - for me it seems everyone is able to connect & get data using  simple Serial Monitor sketch, but not the example ESP32_test. Also - I tried to connect using MAC and int values : 
  //if (!ELM_PORT.connect("001DA520BC21"))
  //if (!ELM_PORT.connect("127324437537"))
  if (!ELM_PORT.connect("V-LINK"))
only the last one seem to work

Another thing - when I tested the ESP32_test in version <2.0 I was getting the same as @pniewiadomski, see :
	>
	Header NOT found
	RPM: 0
	Received: 410C0000

With 2.0 I receive only 0 values all the time

	RPM: 0

Don't have that much time to test with car, but maybe I'll try and modify CPP to print serial output here & there to tell you more
maybe when I manage to start obdsim on rpi I could test it while sitting at home

BTW - should we configure BT PIN for connection ?

miguelos6, 2020-02-28:
Ok I managed to start odbsim on win10 and tested it working with Torque (Android) - proved to work. Still querying it with rpm funtion from ELMduino it behaves weird - displays only value when it's changed in obdsim, then zeroes it out, check below (don't bother for DPF prints, these are my custom functions)

> 18:18:25.177 -> load:0x40078000,len:8896
> 18:18:25.177 -> load:0x40080400,len:5816
> 18:18:25.177 -> entry 0x400806ac

> 

> 18:18:25.177 -> load:0x40080400,len:5816
> 18:18:25.177 -> entry 0x400806ac
> 18:18:26.572 -> Attempting to connect to ELM327...
> 18:18:35.451 -> Connected to ELM327
> 18:18:38.765 -> RPM: 0
> 18:18:38.765 -> DPF soot %: 0
> 18:18:38.765 -> DPF stat: 0

> 18:19:01.559 -> RPM: 1417
> 18:19:01.559 -> DPF soot %: 0
> 18:19:01.559 -> DPF stat: 0
> 18:19:03.902 -> RPM: 0
> 18:19:03.902 -> DPF soot %: 0
> 18:19:03.902 -> DPF stat: 0
> 18:19:06.186 -> RPM: 0
> 18:19:06.186 -> DPF soot %: 0
> 18:19:06.186 -> DPF stat: 0

> 18:19:35.836 -> RPM: 1557
> 18:19:35.836 -> DPF soot %: 0
> 18:19:35.836 -> DPF stat: 0
> 18:19:38.088 -> RPM: 1557
> 18:19:38.088 -> DPF soot %: 0
> 18:19:38.088 -> DPF stat: 0
> 18:19:40.410 -> RPM: 0
> 18:19:40.410 -> DPF soot %: 0
> 18:19:40.410 -> DPF stat: 0
> 18:19:42.676 -> RPM: 0
> 18:19:42.676 -> DPF soot %: 0
> 18:19:42.676 -> DPF stat: 0

PowerBroker2, 2020-02-28:
**Don't try testing the custom PIDs until you get RPM working.**

Simply run the example for ESP32 with the library version 2.0.0 and let me know what happens.

miguelos6, 2020-02-29:
@PowerBroker2 👍 seemed I was trying too much, and it was failing always because other things were added. With running engine & just by changing OBD name it works : 

> 09:15:18.900 -> ets Jun  8 2016 00:22:57
> 09:15:18.900 -> 
> 09:15:18.900 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> 09:15:18.900 -> configsip: 0, SPIWP:0xee
> 09:15:18.900 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> 09:15:18.900 -> mode:DIO, clock div:1
> 09:15:18.900 -> load:0x3fff0018,len:4
> 09:15:18.900 -> load:0x3fff001c,len:1044
> 09:15:18.938 -> load:0x40078000,len:8896
> 09:15:18.938 -> load:0x40080400,len:5816
> 09:15:18.938 -> entry 0x400806ac
> 09:15:19.705 -> Attempting to connect to ELM327...
> 09:15:42.185 -> Connected to ELM327
> 09:15:42.329 -> 	ERROR: 7
> 09:15:42.477 -> 	ERROR: 6
> 09:15:42.722 -> 	ERROR: 7
> 09:15:42.970 -> 	ERROR: 7
> 09:15:43.080 -> RPM: 0
> 09:15:43.118 -> 	ERROR: 6
> 09:15:43.371 -> 	ERROR: 7
> 09:15:43.592 -> RPM: 1004
> 09:15:43.667 -> RPM: 996
> 09:15:43.773 -> RPM: 995
> 09:15:43.881 -> RPM: 991
> 09:15:43.956 -> RPM: 1007
> 09:15:44.069 -> RPM: 1002
> 09:15:44.182 -> RPM: 1007
> 09:15:44.256 -> RPM: 1003

pniewiadowski, 2020-02-29:
@PowerBroker2 unfortunatelly the ESP32 example doesn't work for me.
ERROR: 7 - this is what I got:

> 16:33:18.053 -> ets Jun  8 2016 00:22:57
16:33:18.053 ->
16:33:18.053 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
16:33:18.053 -> configsip: 0, SPIWP:0xee
16:33:18.089 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
16:33:18.089 -> mode:DIO, clock div:1
16:33:18.089 -> load:0x3fff0018,len:4
16:33:18.089 -> load:0x3fff001c,len:1216
16:33:18.089 -> ho 0 tail 12 room 4
16:33:18.089 -> load:0x40078000,len:9720
16:33:18.089 -> ho 0 tail 12 room 4
16:33:18.089 -> load:0x40080400,len:6352
16:33:18.089 -> entry 0x400806b8
16:33:20.246 -> Attempting to connect to ELM327...
16:33:42.581 -> Connected to ELM327
16:33:42.723 -> ERROR: 7
16:33:42.982 -> ERROR: 7
16:33:43.227 -> ERROR: 7
16:33:43.476 -> ERROR: 7
16:33:43.731 -> ERROR: 7
16:33:43.962 -> ERROR: 7
16:33:44.210 -> ERROR: 7
16:33:44.468 -> ERROR: 7
16:33:44.715 -> ERROR: 7

PowerBroker2, 2020-02-29:
@miguelos6 Glad it's working for you!

@pniewiadowski It's fairly common for the ELM327 to have errors in its response for a short amount of time after booting. Let it run for 60 seconds and if it still gives you an error. Can you verify you're using release 2.0.0 and testing with the car on and running?

vfebert, 2020-03-02:
got it running on my ESP32 together with VEEPAK BLE and a C-Class Mercedes.  Thanks you very much for V 2.0.0

miguelos6, 2020-03-02:
@vfebert got anything more but rpm ?

PowerBroker2, 2020-03-02:
@vfebert Could you please post your code? I'd like to incorporate BLE into the library (or at least an example)

vfebert, 2020-03-03:
will do as soon as I wrote some lines. So far, I have only used the sample code posted above, and queried throught the Arduino serial monitor: fuel tank level, for example works fine.

PowerBroker2, 2020-03-03:
You were able to connect to it via Bluetooth serial or did you have to modify the example for BLE? I'm interested only in how you were able to connect the ESP32 to the BLE OBD scanner (if that makes sense).

vfebert, 2020-03-04:
did not (yet) modify the sample code. Typed in the commands into the Arduino serial monitor. Works pretty well ! Right now I am modifying the code to automatically transfer data from the Veepak into MQTT (home automation).  Seeed to have been a good idea to buy an original OBD dongle: https://www.veepeak.com/product/obdcheck-ble/

Sowmya-Sukhanand, 2020-03-09:
@PowerBroker2 Thanks for the library updation.The obd works!!!

hepetry, 2020-03-15:
I really liked your library, where I used your example on an ESP32 card and it worked immediately. Unfortunately I'm not a programmer, and I would like to add the PID's (04 ENGINE_LOAD), (17 THROTTLE_POSITION) (99 ENGINE_REFERENCE_TORQUE). I started with PID 04, changing the example as shown below and had a strange response. Can you help me?
/////////////////////////////////////////////////////// Load
  if (myELM327.queryPID(34, 4))
  {
    int32_t tempLoad = myELM327.findResponse();

    Serial.print("Engine Load: ");
    for (byte i = 0; i < PAYLOAD_LEN; i++)
      Serial.write(myELM327.payload[i]);
    Serial.println();
    
    if (myELM327.status == ELM_SUCCESS)
    {
      load = tempLoad;
      Serial.print("Load: "); Serial.println(load);
......
  }
the answer I got was:
Connected to ELM327
RPM: 843
Engine Load: 7F2211
load: 0

PowerBroker2, 2020-03-15:
Try using

- `myELM327.queryPID(SERVICE_01, ENGINE_LOAD)` 
- `myELM327.queryPID(SERVICE_01, THROTTLE_POSITION)` 
- `myELM327.queryPID(SERVICE_01, ENGINE_REFERENCE_TORQUE)` 

inside your if statements for each query

hepetry, 2020-03-15:
it worked perfectly .. thank you very much !!!

george0305, 2020-03-20:
Regarding the ESP32, i used it for two weeks now with the elm32 obd and it worked fine. Thank you for that! However, for a few days now it stopped connecting and i get the Phase 2 error. I can find the elm on phone bluetooth. What can it possibly be?

PowerBroker2, 2020-03-20:
When you look on you phone, what's the name of the ELM327? Is it "OBDII"? Also, what version of the library are you using? Are you using the latest release?

george0305, 2020-03-21:
Thank you for your time, yes, on my phone it`s OBDII. I checked that. The version is 2.0.7, i had the arduino implemented in the car for 2 weeks and everything went fine until now. I didn`t change anything, just let it run. Yesterday i recompiled and uploaded but still the same problem. Below is the connection part, the one in your example and i added a display:
`
... 
Serial.begin(115200);
 display.begin(SSD1306_SWITCHCAPVCC, 0x3C);  // required to run SSD1306
         
   
  
  ELM_PORT.begin(ESP_BLUETOOTH_NAME, true);

  Serial.println("Attempting to connect to ELM327...");
  
      display.clearDisplay();
      display.setTextSize(1);
      display.setTextColor(WHITE);
      display.setCursor(10, 20);
      display.println("Connecting...");
      display.display();
     
 
      
  if (!ELM_PORT.connect("OBDII"))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 1");
    while (1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
  
`

PowerBroker2, 2020-03-21:
Hmm, there have been several bug fixes since that release. Try using release 2.0.10 and see if it fixes things for you.

george0305, 2020-03-21:
Hi, so i updated the library and i tested. Results:
-Still not connecting by itself but i managed to get it connected like so:
 - pair the phone with the OBDII
 - unpair the phone
 - restart the ESP32
 - it connects.
From what i can see, im not sure and possibly be wrong, its not pairing or its waiting for the pairing pin. I tried to search how the bluetooth serial connects and i tried a setpin but no success.
Can it be an ELM327 problem? If so, i can try tomorrow or on Monday to get a new one.

PowerBroker2, 2020-03-21:
The problem might lie in the fact that your phone is paired. Because of this, the ELM might be bound to the phone without allowing your Arduino to connect to it. Try **turning off bluetooth** for all other devices (including your phone) and then see if the Arduino can connect to the ELM.

george0305, 2020-03-21:
I tried it, i dont usually leave bt on, so that isnt the option. If i manage to connect it to the elm by the steps above and then restart the esp it wont reconnect to the elm (phone bt off).

PowerBroker2, 2020-03-21:
Bizzare, you might need to go into the library's function `int8_t ELM327::sendCommand(const char *cmd)` and add print statements to determine what (if anything) is being returned by the ELM327 when being configured by the Arduino.

george0305, 2020-03-21:
Ok, i will try to test it and i will come back these days. Thank you a lot!

vfebert, 2020-03-22:
my 2 cents on function:  Last version 2.0.10 works without any issues on my end. Again, BIG THANKS
#1 smartphone bluetooth always off when testing
#2 have spend 30$ on a good adaptor from Veepeak, not over Amazon, directly from manuf to avoid getting a bad clone.  Nice overview here: https://www.carscanner.info/choosing-obdii-adapter/ (and no, I am not working for that company ;-)
#3 trying to be close to the adaptor (means: sitting in my car with the laptop testing). Reason: from 5meter away, I got a connection (took longer), but no data. 
hope this helps?

PowerBroker2, 2020-03-28:
Thanks @vfebert! Since it seems to be working for the vast majority of everyone (including BLE users apparently), I'm going to close this issue. If anyone still has issues, please make a new issue for your individual problem and I'll get to it as soon as I can.

closed 2020-03-28 by PowerBroker2

vfebert, 2020-03-29:
big thanks to you for making it working perfectly now - My application based on your SW is up and running now also:  "Visualize your daily CO2 footprint based on your car fuel consumption, heating, electricity, water"

PowerBroker2, 2020-03-29:
Glad to help! Do you have a link to your project build?

vfebert, 2020-03-30:
not yet - working on it - will keep you posted

vsaravind007, 2020-04-04:
Hi @PowerBroker2, thanks for all the hard work on the library. I'm using an ESP32 Dev kit with Bluetooth ELM327 module. I'm using your example & the connection goes through correctly, but all I get is ERROR 7, even after 3 minutes:
[image]

Tried sending the AT commands with the basic sketch, it works fine. Any ideas ? 

_Thanks_

PowerBroker2, 2020-04-04:
Looks like you're getting a timeout error. So I can keep track of everyone's issues separately, **please make a new issue** describing your exact hardware setup, testing environment, library version, and complete code.

vsaravind007, 2020-04-04:
Sure, thanks!

vfebert, 2020-06-07:
Hi PowerBroker,

 

finally I can answer your question: „do you have a repository for your solution?“ – it was about the project to read out mileage and fuel consumption fully automated.

 

Here it is: https://myco2.org/

 

The repository is: https://github.com/VolkerMyCO2/CarMileageFuel 

Hope you like it and everything is ok for you.  If not, just let me know.

 

Regards,

Volker