PowerBroker2/ELMduino #7: Show & Clear Diagnostic Trouble Codes

miguelos6, 2020-02-27:
Hi
Are you planning to implement show & clear diagnostic trouble codes ?
I guess these are services 03 and 04 (hex)

PowerBroker2, 2020-02-27:
Yeah, that shouldn't be too difficult. Can you provide some example queries and their respective responses?

PowerBroker2, 2020-02-28:
Try and see if you can show/clear diagnostic codes with release [2.0.0](https://github.com/PowerBroker2/ELMduino/releases/tag/2.0.0) (use queryPID()) and let me know.

miguelos6, 2020-03-04:
I'm not sure if you need to query for all trouble codes in loop or it's just one query that returns whatever is there. 
mode 3 in this doc ->
https://python-obd.readthedocs.io/en/latest/Command%20Tables/   
Can't test it as I don't have any error code in car currently

PowerBroker2, 2020-03-04:
Checking and clearing trouble codes will be up to the user to program specific commands and queries using `queryPID()` (**or** `sendCommand()`) with `findResponse()` similar to how it's done for custom PID queries.

Speaking of which, were you able to get the custom queries to work with the current lib version?

closed 2020-03-05 by PowerBroker2