PowerBroker2/ELMduino #9: ESP8266

jostfelix, 2020-03-20:
I want to read PIDs over my ESP8266 WiFi Modul.
How can i implement your ELM Duino to read Pids over the wifi connection ?
When i enter 010C in my serial monitor iam get the right answer from the ELM327 so the connection should be fine ?
But now i dont no know how i can use ur ELMDuino sketch to simplifz things and get the data to be print in to mz serial monitor  ?

`#include <ESP8266WiFi.h>

const char* ssid     = "WiFi_OBDII";
const char* password = "your-password";

//IP Adress of your ELM327 Dongle
IPAddress server(192, 168, 0, 10);

boolean last_in = false;

WiFiClient client;

void setup() {
  Serial.begin(115200);
  delay(10);

  // Connecting to ELM327 WiFi
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid);
  // WiFi.begin(ssid, password);  //Use this line if your ELM327 has a password protected WiFi

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("Connected to Wifi");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  if (client.connect(server, 35000)) {
    Serial.println("connected");
    Serial.println("<ATZ");
    // Make a Request
    client.println("ATZ");
  }
  else {
    Serial.println("connection failed");
    ESP.reset();
  }
}

void loop() {
  if (!client.connected()) {
    Serial.println("CLIENT DISCONNECTED - RESET!");
    client.stop();
    ESP.reset();
  }
  // if there are incoming bytes available
  // from the server, read them and print them:
  if (client.available()) {
    if (!last_in) {
      Serial.println("");
      Serial.print(">");
      last_in = true;
    }
    char c = client.read();
    /*Serial.print(" 0x");
    Serial.print(c, HEX);*/
    Serial.print(c);
  }

  // as long as there are bytes in the serial queue,
  // read them and send them out the socket if it's open:
  while (Serial.available() > 0) {
    if (last_in) {
      Serial.println("");
      Serial.print("<< ");
      last_in = false;
    }
    char inChar = Serial.read();
    if (client.connected()) {
      client.print(inChar);
    }
  }
}`

PowerBroker2, 2020-03-20:
Try this:

```C++
#include <ESP8266WiFi.h>
#include "ELMduino.h"

const char* ssid = "WiFi_OBDII";
const char* password = "your-password";

//IP Adress of your ELM327 Dongle
IPAddress server(192, 168, 0, 10);
WiFiClient client;
ELM327 myELM327;

void setup()
{
  Serial.begin(115200);

  // Connecting to ELM327 WiFi
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid);
  // WiFi.begin(ssid, password); //Use this line if your ELM327 has a password protected WiFi

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("Connected to Wifi");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  if (client.connect(server, 35000))
    Serial.println("connected");
  else
  {
    Serial.println("connection failed");
    ESP.reset();
  }

  myELM327.begin(client);
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
  {
    Serial.print(F("\tERROR: "));
    Serial.println(myELM327.status);
    delay(100);
  }
}

```

Felix-Jost, 2020-03-21:
Thanks for your Response !
now i can read the data but only rpm and kph(Speed).
if i want to Read other PIDS its dosent work proberly.
So the rpm works good but i cant get the COOLANT temperature?

I hope this is Right ?:float tempCOOLANT = myELM327.queryPID(SERVICE_01,ENGINE_COOLANT_TEMP);

void loop()
{
  float tempRPM = myELM327.rpm();
  float tempBATTERY = myELM327.queryPID(SERVICE_01,CONTROL_MODULE_VOLTAGE);
  float tempCOOLANT = myELM327.queryPID(SERVICE_01,ENGINE_COOLANT_TEMP);
  float tempINTAKE = myELM327.queryPID(SERVICE_01,INTAKE_AIR_TEMP);
  float tempOIL = myELM327.queryPID(SERVICE_01,ENGINE_OIL_TEMP);
  float tempLOAD= myELM327.queryPID(SERVICE_01,ENGINE_LOAD);
  float tempSPEED1= myELM327.kph();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;

    tft.setCursor(50, 95); // Set position (x,y)
    tft.setTextColor(ST7735_WHITE);  // Set color of text. We are using custom font so there is no background color supported
    tft.setTextSize(2);  // Set text size. Goes from 0 (the smallest) to 20 (very big)
    tft.print(rpm);  // Print a text or value
    
    Serial.print("RPM: "); Serial.println(rpm);

  

     coolant = (uint32_t)tempCOOLANT;

    tft.setCursor(5, 15); // Set position (x,y)
    tft.setTextColor(ST7735_GREEN);  // Set color of text. We are using custom font so there is no background color supported
    tft.setTextSize(2);  // Set text size. Goes from 0 (the smallest) to 20 (very big)
    tft.print(coolant);  // Print a text or value
     Serial.print("COOLANT: "); Serial.println(coolant);

PowerBroker2, 2020-03-21:
I was able to help another user successfully query non-rpm/non-speed PIDs in [issue #4](https://github.com/PowerBroker2/ELMduino/issues/4). Keep in mind, you have to check `myELM327.status` after each `myELM327.queryPID()` call. Also keep in mind that you may have to do some extra math on the return value of `myELM327.queryPID()` to get the correct reading.

Here's an example:
```C++
// modified test + my custom PIDs
// based purely on library

#include "BluetoothSerial.h"
#include "ELMduino.h"

#define ELM_PORT SerialBT
#define ESP_BLUETOOTH_NAME "ESP32"

BluetoothSerial SerialBT;
ELM327 myELM327;

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  Serial.begin(115200);
  ELM_PORT.begin(ESP_BLUETOOTH_NAME, true);

  Serial.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("V-LINK"))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 1");
    while (1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }
  
  Serial.println("Connected to ELM327");
  
  for (int a = 0; a < 10; ++a)
  {
    digitalWrite(LED_BUILTIN,HIGH); delay(100);
    digitalWrite(LED_BUILTIN,LOW); delay(100); 
  }
}

void loop()
{
  int32_t rpm = -1;
  int32_t soot = -1;
  int32_t burn = -1;
  float litersLeft = -1;

  /////////////////////////////////////////////////////// RPM
  float tempRPM = myELM327.rpm();
  
  Serial.print("Payload received for rpm: ");
  for (byte i = 0; i < PAYLOAD_LEN; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();
  
  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (int32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
      printError();

  /////////////////////////////////////////////////////// Soot
  if (myELM327.queryPID(34, 13162))
  {
    int32_t tempSoot = myELM327.findResponse();

    Serial.print("Payload received for soot: ");
    for (byte i = 0; i < PAYLOAD_LEN; i++)
      Serial.write(myELM327.payload[i]);
    Serial.println();
    
    if (myELM327.status == ELM_SUCCESS)
    {
      soot = tempSoot;
      Serial.print("Soot: "); Serial.println(soot);
    }
    else
      printError();
  }

  /////////////////////////////////////////////////////// Liters
  if (myELM327.queryPID(34, 4906))
  {
    int32_t tempLitersLeft = myELM327.findResponse() / 64.0;

    Serial.print("Payload received for liters: ");
    for (byte i = 0; i < PAYLOAD_LEN; i++)
      Serial.write(myELM327.payload[i]);
    Serial.println();

    if (myELM327.status == ELM_SUCCESS)
    {
      litersLeft = tempLitersLeft;
      Serial.print("Liters: "); Serial.println(litersLeft);
    }
    else
      printError();
  }

  /////////////////////////////////////////////////////// Burns
  if (myELM327.queryPID(34, 8434))
  {
    int32_t tempBurn = myELM327.findResponse();

    Serial.print("Payload received for DPF burns: ");
    for (byte i = 0; i < PAYLOAD_LEN; i++)
      Serial.write(myELM327.payload[i]);
    Serial.println();

    if (myELM327.status == ELM_SUCCESS)
    {
      burn = tempBurn;
      Serial.print("DPF burns: "); Serial.println(burn);
    }
    else
      printError();
  }

  for (int a = 0; a < 5; ++a)
  {
    digitalWrite(LED_BUILTIN,HIGH);
    delay(200);
    digitalWrite(LED_BUILTIN,LOW);
    delay(200); 
  }
}

void printError()
{
  if (myELM327.status == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.status == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.status == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.status == ELM_GARBAGE)
    Serial.println(F("\tERROR: ELM_GARBAGE"));
  else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.status == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_GENERAL_ERROR)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));
  }

  delay(500);
}
```

PowerBroker2, 2020-03-23:
Seeing as @Felix-Jost was able to get it to work for an ESP8266, I'm closing this ticket.

closed 2020-03-23 by PowerBroker2