PowerBroker2/ELMduino #10: ELM327 with USB cable

aster94, 2020-03-29:
Hi!

I would guess that your library works using bluetooth commands, isn't it?

Have you ever saw this EML327 adapter?
[image]
I would be curios to know if it the same command you send via BT works the same via Serial, do you have any idea?

PowerBroker2, 2020-03-29:
The commands would be the same, but you would need a USB host interface to drive the device. You can't plug one of those directly to Serial, if that makes sense.

aster94, 2020-03-29:
actually i read that these adapter uses a CH340 so I was thinking about getting rid of the CH340 and use directly the TTL signal with your library.

I know i could use BT but I would like the fastest communication possible

BTW a few years ago I wrote a library that uses the [keyword protocol](https://github.com/aster94/Keyword-Protocol-2000) but soon i will get a new motorbike that uses the can bus so your library would be perfect

PowerBroker2, 2020-03-29:
If you can access the TTL UART lines  of the ELM327 directly, the library should work perfectly for you

PowerBroker2, 2020-03-30:
I'm going to close this for now, but if you have more issues with this you can reopen it.

closed 2020-03-30 by PowerBroker2