PowerBroker2/ELMduino #11: ESP32 Connection

Felix-Jost, 2020-03-30:
I used the following to connect to my ELM327, to this Point ist al working good but the readout is ERROR: 7 and it wont be displaying my rpm.
I also check my ELM327 with torque app and it all Works fine.

#include "BluetoothSerial.h"

#include "ELMduino.h"

#define ELM_PORT SerialBT

#define ESP_BLUETOOTH_NAME "ESP32"

BluetoothSerial SerialBT;

ELM327 myELM327;

uint32_t rpm = 0;

void setup()

{

  pinMode(LED_BUILTIN, OUTPUT);

  digitalWrite(LED_BUILTIN, HIGH);

  Serial.begin(115200);

  ELM_PORT.begin(ESP_BLUETOOTH_NAME, true);

  Serial.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("Carly Adapter"))

  {

    Serial.println("Couldn't connect to OBD scanner - Phase 1");

    while (1);

  }

  if (!myELM327.begin(ELM_PORT))

  {

    Serial.println("Couldn't connect to OBD scanner - Phase 2");

    while (1);

  }

  

  Serial.println("Connected to ELM327");

}

void loop()

{

  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)

  {

    rpm = (uint32_t)tempRPM;

    Serial.print("RPM: "); Serial.println(rpm);

  }

  else

  {

    Serial.print(F("\tERROR: "));

    Serial.println(myELM327.status);

    delay(100);

  }

}

PowerBroker2, 2020-03-30:
Is this related to issue #9? If so are you trying to use an ESP32 or ESP8266? Are you sure the OBD scanner's bluetooth name is `Carly Adapter`? Also, what do you mean when you say it's been working to this point yet gives you errors? Have you always been getting the errors? Was there a time you weren't getting the errors?

Sorry for the barrage of questions, but when submitting issues you need to provide as much clarity and detail as possible so that I can more easily help.

Felix-Jost, 2020-03-30:
In issue #9 its all works fine with the Wifi adapter and ESP8266 thx for that.

but now I'm using a ESP32 now so its not related to issue#9.
I'm using 2 OBD ELM327 Adapter  (OBDII and Carly Adapter) names of the adapter.
Its connect to the ELM327 but if I want to get the rpm is says Error 7.

PowerBroker2, 2020-03-30:
It looks like the Carly Adapter requires a BLE connection. The code you provided is "normal" bluetooth. To be honest, I'm not sure how to connect to a BLE device since I don't have a BLE OBD scanner. Take a look at some ESP32 BLE examples and see if you can connect and communicate with the Carly Adapter.

jostfelix, 2020-03-30:
I am also using a standard OBDII adapter ELM327 like zou can buy everywhere, i tried both of them ...

jostfelix, 2020-03-30:
[image]

PowerBroker2, 2020-03-30:
What code did you use for the standard adapter?

jostfelix, 2020-03-30:
#include "BluetoothSerial.h"
#include "ELMduino.h"

#define ELM_PORT SerialBT
#define ESP_BLUETOOTH_NAME "ESP32"

BluetoothSerial SerialBT;
ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  Serial.begin(115200);
  ELM_PORT.begin(ESP_BLUETOOTH_NAME, true);

  Serial.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 1");
    while (1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }
  
  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
  {
    Serial.print(F("\tERROR: "));
    Serial.println(myELM327.status);
    delay(100);
  }
}

PowerBroker2, 2020-03-30:
[Please edit your code so that it's properly formatted and tabulated](https://help.github.com/en/github/writing-on-github/creating-and-highlighting-code-blocks)

jostfelix, 2020-03-30:
[image]

jostfelix, 2020-03-30:
So the name of the ELM327 is OBDII so thats correct,

PowerBroker2, 2020-03-30:
Are you testing with the car on and running?

jostfelix, 2020-03-30:
Yes , i will try it tomorrow again.

PowerBroker2, 2020-03-30:
Ok, also are you using the latest release of the library? Did you edit the library yourself?

It's bizarre because I haven't had trouble getting an ESP32 to communicate like this.

Felix-Jost, 2020-03-31:
When cam compiling this sketch its says LED_BUILTIN was not declared ?
How should I declare it ?

PowerBroker2, 2020-03-31:
It's declared in the Arduino Core for you. It works for all Arduinos as far as I'm aware. What board are you compiling for?

PowerBroker2, 2020-03-31:
Just out of curiosity, why do you use two GitHub accounts?

Felix-Jost, 2020-04-01:
Iam Compiling for a ESP32 Dev Module

PowerBroker2, 2020-04-01:
You can just delete those two lines in the sketch

PowerBroker2, 2020-04-04:
Haven't heard a response, assuming everything is working. If you have an update, feel free to reopen.

closed 2020-04-04 by PowerBroker2