PowerBroker2/ELMduino #12: Connection problem

tomek8308, 2020-04-04:
Hey.
 I have a problem logging in ESP32 to LM327. After uploading the code, it worked for several days, then ESP32 stopped connecting with LM. I noticed that you need to log in to another LM237 device using the pairing code "1234" and the connection is made but only until the first restart. Then repeat the whole procedure - enter the pairing code "1234" on another device. Something is working incorrectly I have two ESP32 and LM237 and in all cases it is so bad for this I exclude a hardware problem.
I check the operation on the unmodified test code and the adapter name is OBDII.
How it looks can be seen in the video:  https://youtu.be/d-oBFSECpDg 

#include "BluetoothSerial.h"
#include "ELMduino.h"

#define ELM_PORT SerialBT
#define ESP_BLUETOOTH_NAME "ESP32"

BluetoothSerial SerialBT;
ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  Serial.begin(115200);
  ELM_PORT.begin(ESP_BLUETOOTH_NAME, true);

  Serial.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 1");
    while (1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }
  
  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
  {
    Serial.print(F("\tERROR: "));
    Serial.println(myELM327.status);
    delay(100);
  }
}

PowerBroker2, 2020-04-04:
What sort of OBD scanner is it? Is it a classic bluetooth device or is it a BLE version?

When you try the [ESP32_test.ino example](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino), are you able to send [AT commands](https://github.com/PowerBroker2/ELMduino#list-of-standard-pids)?

tomek8308, 2020-04-04:
Yes, it's a classic ELM327. I have two different models on both there is the same effect. If I can connect, I can send AT commands and send PID and receive responses. Everything worked fine and I tested for several days. Suddenly he stopped connecting by accident, I noticed that the connection is only after sending the pairing pin 1234 on another device. I had a second ESP32 - started working but the situation repeated after a few days. I think the problem lies in pairing ELM237 with ESP32, but I don't know how to solve it because I'm not a programmer. I decided to take care of this project and I liked it, I managed to display DPF data and temperature. It would be great if it worked again.

PowerBroker2, 2020-04-04:
From the video I can see it failed the second phase of connection. The ESP32 test sketch overrides this "second phase" and allows you to send queries and AT commands manually. In order to debug, I need to know if you can still get proper AT and query responses from the ELM327 using the ESP32 test sketch. That'll be a big help to determine what went wrong.

Also, when you say "I decided to take care of this project", does that mean you took it over from someone?

tomek8308, 2020-04-04:
Tomorrow I will check the possibility of sending AT commands in test mode. As for the project, I saw on YouTube a video of the engine parameters display in ESP32 + Oled and I decided to do something similar myself. After the first successes, there was a connection problem that I couldn't solve myself. So I came back to basics and I start again with ESP32_test.ino to get a stable connection with ELM327 but my knowledge is too small.

PowerBroker2, 2020-04-04:
Interesting, could you post a link to the video?

PowerBroker2, 2020-04-05:
I found a bug in the code and applied a fix in version 2.0.12. Update your library and try using the ESP32 test sketch and post what happens when you send the following ELM327 commands:

```
AT E0 //Echo off
AT S0 //Spaces off
AT SP0 //Mode --> Automatic
010C //Query RPM
```

tomek8308, 2020-04-05:
Ok. I updated the library and uploaded the ESP32 test. I connected for a moment but couldn't send AT commands. I will have more time later so I will check again.

PowerBroker2, 2020-04-05:
Can you post a picture of the serial monitor after you input the commands (ensure the serial monitor is in newline and carriage return mode)

tomek8308, 2020-04-05:
I connected ESP32 but does not accept AT commands and no response.
[image]
After entering pin 1234 on another device, I get the correct automatic response "rpm"
[image]
I also used to check the code:

` #include "BluetoothSerial.h"

BluetoothSerial SerialBT;

#define DEBUG_PORT Serial
#define ELM_PORT   SerialBT

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("ESP32test", true);

  DEBUG_PORT.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);
  }

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}

void loop()
{
  if(DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();

    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }

  if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if(c == '>')
      DEBUG_PORT.println();

    DEBUG_PORT.write(c);
  }
} `

And no shoulder to answer AT commands
[image]

And after entering the pairing code "1234" on another device, the connection is stable and the correct responses to AT commands follow.
[image]
The connection is always stable but only until ESP32 is restarted. ELM327 will not reconnect if it is not paired with the "1234" code on the external device
It seems to me that the problem lies in the connection of ELM327 with ESP32. ELM327 necessarily wants to pair with the "1234" pin and does not receive a response from ESP32 - is it possible?

PowerBroker2, 2020-04-05:
Ok, I think what's happening is that once the ESP32 restarts, one of your other devices (i.e. phone or laptop) automatically connects to the ELM327. Since the ELM327 can only be connected to one bluetooth device at a time, the ESP32 can't reconnect to it. Try turning off bluetooth and/or unpairing the ELM327 on all devices other than your ESP32. Once you do this, the ESP32 should be able to regain connection to the ELM327 after restarting.

Unfortunately, there is no way to send a security authentication pin via an ESP32's built-in bluetooth serial. (BLE is a different story, but we're not using BLE here)

tomek8308, 2020-04-05:
OK. I will try to turn off bluetooth on all devices in the car tomorrow, maybe some are pairing and blocking the connection.

tomek8308, 2020-04-06:
Unfortunately, no device blocked the bluetooth connection.
However, I was able to solve the problem.
ESP32 worked properly for the first few days so I decided that ESP32 has a problem.
1. I have cleaned the Flash memory
2. I uploaded the bootloader again
3. I have uploaded the ESP32 test
Connection problem has resolved ESP connects correctly with ELM327 all data is transferred.
Thank you for your help.

closed 2020-04-06 by PowerBroker2

maksumon, 2020-07-15:
@tomek8308 Would you please let me know the steps and files needed to flash the memory and upload bootloader on ESP32?

nomidaepapi, 2023-06-06:
> Unfortunately, no device blocked the bluetooth connection. However, I was able to solve the problem. ESP32 worked properly for the first few days so I decided that ESP32 has a problem.
>
> 1. I have [...]

Could you pleaase explain what you have done to get it working?