PowerBroker2/ELMduino #13: ERROR 7 On ESP32 Wrover Board

vsaravind007, 2020-04-04:
Arduino IDE Version: **1.8.5**
ELMduino Library Version: **2.0.11** (Latest at the time of opening this issue)
ESP32 Build Tools Version: **1.0.4** (Latest at the time of opening this issue)
ESP32 Board Used: **Wroom 32 ESP32 Dev Board**
ELM327 Module: Generic Bluetooth ELM327 

**Issue description**
Getting ERROR 7 on the serial console while running the ESP32_Bluetooth_Serial.io example, Used the file ESP32_Bluetooth_Serial.ino that came with the library, only changes were updating the Bluetooth name of the ELM327 module & commenting the following lines since LED_BUILTIN is not defined for this particular ESP32 board:

```
pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, HIGH);
```

No issues with compilation or connection with the ELM327 module, ESP32 is connecting fine with the module quickly  its a generic known working device, Torque works fine.

Tried sending AT commands using the minimal example & it worked fine, module responds back with values for AT commands. 

Attaching the screenshot of the serial monitor:
[image]

PowerBroker2, 2020-04-04:
I'll fix the example code so that the compilation error isn't a problem anymore. Curious, are you testing with the car on and running?

vsaravind007, 2020-04-04:
Yes, the car is running, its a Skoda Octavia MK2 TDI. Does it have something to do with the OBD2 protocol?

PowerBroker2, 2020-04-04:
It might actually have something to do with the OBD interface if you're car was made on or before 2008. When was your car made?

vsaravind007, 2020-04-04:
Its a 2008 model, Arduino with a CANBus Shield works fine and responds to all PIDs correctly.

PowerBroker2, 2020-04-05:
I found a bug in the code and applied a fix in version 2.0.12. Try testing the new ESP32 Bluetooth Serial example. If it still doesn't work for you, try the ESP32 test sketch and post what happens when you send the following ELM327 commands:

```C++
AT E0 //Echo off
AT S0 //Spaces off
AT SP0 //Mode --> Automatic
010C //Query RPM
```

vsaravind007, 2020-04-05:
Hi @PowerBroker2, thanks for the quick update, I'll check with the latest version & report back.

PowerBroker2, 2020-04-08:
Did you get it to work?

vsaravind007, 2020-04-08:
Apologies for the later response, couldn't test it yet. Will be testing it today or tomorrow, I'll make sure to update here.

pascaltippelt, 2020-04-19:
Hi!
I am having the same issue. I am using a ESP32 (glued on a TTGO T4, but it is a box standart ESP32). When I try to connect over BT to my OBD-Stick (verified working with Android Smartphone + Torque) I only get "Error 7":

Outpup with my own program:

[image]

Output with you ESP32 test program

[image]

But as you can see it works with Torque

[image]

I use a 2000 smart car wich uses the ISO14230-4 KWP fast protocoll.

PowerBroker2, 2020-04-19:
@pascaltippelt Please create a new issue instead of hijacking this one. In your new issue, please make sure to post your entire sketch properly formatted in addition to the screenshots you have here.

vsaravind007, 2020-04-19:
@PowerBroker2 My issue has been resolved, thanks for the library and the efforts! I'm closing the ticket :)

closed 2020-04-19 by vsaravind007

PowerBroker2, 2020-04-19:
@vsaravind007 Can I ask how you solved it?