PowerBroker2/ELMduino #14: home-testing Elmduino

miguelos6, 2020-04-10:
Hi,
Thank you for ELMduino. I'm already using it in my car with custom PIDs (discussion already closed as testing went fine). Since I don't drive the car much lately and the condition I'm waiting for has not occurred since I did my setup I cannot really prove it works. I mean - 4 out of 5 values I'm reading are ok, changing etc. But the one I'm most interested in (DPF burn status 0/1) happens once 300-700km driven. 
I did already small instructable & shared with community. The only one person testing this solution said It didn't show the status properly (DPF burn was 0, while Torque showed 1 at the same time)

My question is - is there any quick ESP32/Arduino Bluetooth emulator that we program to feed other ESP32 running Elmduino to test ? I'm specially intersted in this one custom PID

PowerBroker2, 2020-04-10:
Other than [the simple ELM327 simulator I wrote in Python](https://github.com/PowerBroker2/ELM_Simulator/blob/master/ELM_Simulator.py), the only other such simulator I know of is [this one](https://www.instructables.com/id/Arduino-OBD2-Simulator/). Maybe if you google, something else might pop up.

The Python sim should work for you since its easy to edit and change for your specific PIDs/responses.

PowerBroker2, 2020-04-12:
Due to lack of response, I'm assuming it's been all sorted out

closed 2020-04-12 by PowerBroker2