PowerBroker2/ELMduino #15: Error 7 / Timeout with ESP32

pascaltippelt, 2020-04-20:
**Arduino IDE Version:** 1.8.12
**ELMduino Library Version:** 2.0.12 (Latest at the time of opening this issue)
**ESP32 Board Used:** TTGO T4 (based on ESP32 Dev Board)
**ELM327 Module:** Generic Bluetooth ELM327
**Car:** MCC Smart build in 2000

**Issue description:**

I try to use your [test-sketch for ESP32 ](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino). No changes. I get the following console output:

[image]

My engine is of cause running, I testet the adapter using the "Torque" App for Android. The adapter works well, its name is "OBDII".

I believe the connection to the ECU may fail. Any idea how to figure that out?

If I use the [ESP32_Bluetooth_Serial.ino](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino)-Sketch, I can see another error-message:
[image]

PowerBroker2, 2020-04-20:
> I try to use your [test-sketch for ESP32 ](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino). No changes. I get the following console output:

It looks like you're using Bluetooth to connect to the ELM327, using Bluetooth to view debug statements on a separate device (your Android), and yet not changing the example code. Is this correct or am I misunderstanding a part of the setup?

Also, have you _ever_ been able to get successful readings from the ELM327 using your ESP32?

pascaltippelt, 2020-04-20:
Hi, thanks for your reply.

No, you missunderstood my setup. I use the Android device with OTG to power my ESP32 while I am in the car an see the Serial output. Your sketch is really unchanged.

I have never had been able to get successful readings from the ELM327 using my ESP32. But I used my Android-Phone before with the same ELM327 and it connects and works like a charm.

PowerBroker2, 2020-04-20:
Oooh, I understand now.

It seems that you have successfully connected and initialized the ELM327, but just having trouble getting valid response from data queries. I think I have a trick or two to help fix this.

The first one would be specifying exactly what protocol to initialize the ELM327 with. Currently it defaults to "automatic", which _should_ work for all car types, but obviously isn't the case. Right now, the library allows the calling of the `(class).initializeELM(protocol);` function with the following as possible parameters:

```
  * Protocol - Description
  * 0        - Automatic
  * 1        - SAE J1850 PWM (41.6 kbaud)
  * 2        - SAE J1850 PWM (10.4 kbaud)
  * 4        - ISO 9141-2 (5 baud init)
  * 5        - ISO 14230-4 KWP (5 baud init)
  * 6        - ISO 14230-4 KWP (fast init)
  * 7        - ISO 15765-4 CAN (11 bit ID, 500 kbaud)
  * 8        - ISO 15765-4 CAN (29 bit ID, 500 kbaud)
  * 9        - ISO 15765-4 CAN (11 bit ID, 250 kbaud)
  * A        - ISO 15765-4 CAN (29 bit ID, 250 kbaud)
  * B        - User1 CAN (11* bit ID, 125* kbaud)
  * C        - User2 CAN (11* bit ID, 50* kbaud)
```

In your case, you would want to call `(class).initializeELM('6');` directly after calling `(class).connect("OBDII");`.

That being said, I'd like to make it easier for you to do something like this by adding a custom protocol parameter to the `(class).begin()` function so that you don't have to call `(class).initializeELM()` manually.

Does this all make sense?

PowerBroker2, 2020-04-20:
Another thing to test: [try this sketch](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino). Once you run the sketch, try entering the following in the serial monitor (make sure to include carriage returns):

```
AT E0
AT S0
AT SP0
010C
```

If the above commands do not work, try these:

```
AT E0
AT S0
AT SP6
010C
```

PowerBroker2, 2020-04-20:
In [release 2.1.0](https://github.com/PowerBroker2/ELMduino/releases/tag/2.1.0), you can do the following:

```C++
  if (!myELM327.begin(ELM_PORT, ISO_14230_FAST_INIT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }
```

pascaltippelt, 2020-04-20:
Thank you for your work, I will test the code _tomorrow_. My neigborhood might be a bit annoyed starting and stopping the engine every 20 minutes...

I have a question about [your erlier reply](https://github.com/PowerBroker2/ELMduino/issues/15#issuecomment-616692979): How does this work? Shouldn't there be something like 
```cpp
if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
```
inside the loop() of the sketch to submit the commands over Bluetooth?

PowerBroker2, 2020-04-20:
I apologize, I linked the wrong sketch. Use this one instead:

```C++
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

#define DEBUG_PORT Serial
#define ELM_PORT   SerialBT

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("ESP32test", true);

  DEBUG_PORT.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);
  }

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}

void loop()
{
  if(DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();

    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }

  if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if(c == '>')
      DEBUG_PORT.println();

    DEBUG_PORT.write(c);
  }
}
```

pascaltippelt, 2020-04-21:
Good Morning!

So I tried the [new sketch](https://github.com/PowerBroker2/ELMduino/issues/15#issuecomment-616750607) this morning. I believe the Connection with the ELM works, as I get results:
[image]
I am really sorry for the poor quality of the Android Terminal, all those "^M" wich have this gray background are [ASCII Chars](https://www.torsten-horn.de/techdocs/ascii.htm). "^M" reperesents a carriage return.

But: The last command, the "010C" returns without those ASCII Chars a "SEARCHING" and then it looks like "401D0". 

I will get the newer version of your library today and try to specify the protocoll manually - lets see if this helps.

pascaltippelt, 2020-04-21:
> You did mention you had the ELM327 working with Torque. Are you disabling your phone's Bluetooth before turning your ELM327 and Arduino back on for project testing? I'm concerned about possible cross-talk between the app and your Arduino.

No, the Android has disabled Bluetooth. No other active Bluetooth devices in the area.

> This means it's _almost_ working. A correct response would look something like "SEARCHING...410C541B". [See this site for reference](http://dthoughts.com/blog/2014/11/06/obd-scanner-using-elm327/). I'm not sure why the OBD scanner isn't returning a response in the correct format...
> 
> Do you have a laptop you could do your debugging with? It would be a lot easier to see what's going on with a normal serial monitor.

I will borrow my girlfriends tablet tomorrow and check if I can better results in the terminal (putty or terra term). I will also connect to the ELM directly (with the tablet) and use the 
`AT E0
AT S0
AT SP0
010C`
there.

By the way: Yesterday I got 2 other ESP32 modules (without a display, completely naked). They show the same strange behaviour.

Maybe it is a fault of my ELM. If nothing works, I will go and get another one, so I can absolutely say it is not a hardware issue.

PowerBroker2, 2020-04-21:
> Maybe it is a fault of my ELM. If nothing works, I will go and get another one, so I can absolutely say it is not a hardware issue.

Yeah, I think it _might_ be the OBD scanner if it doesn't work when specifying the correct protocol. Not sure why it would work with Torque, though.

> And I will try an other car as well. Might be the pre-OBD-II-standart of my damn old car :-D

Maybe, but I should still work with your old car, lol

pascaltippelt, 2020-04-23:
Just to inform you: I am waiting for new hardware for further testing, will reply soon.

pascaltippelt, 2020-04-26:
Okay thats the state at the moment: Yesterday I got a brand new ELM OBDII adapter and attached it to my girlfrieds car (1-series bmw). I got a connection and was able to read the RPM. I was also able to get the RPM from my smart car.

Conclusion: Your software seems corret.

BUT: After that I (beeing an idiot) used this brand new adapter with torque. And after that tha adapter did not work any longer with any car. I am really puzzled abaout this situation. Somehow the torque app seems to modify the adapter in a way I don't understand and makes it unusable for your library.

Is there maybe a AT-code to fully reset a ELM327?

As there is no real issue with your library, I think I will close this issue...

PowerBroker2, 2020-04-26:
Interesting...

You can do `AT Z` to reset all settings. Maybe that will do it.

closed 2020-04-26 by PowerBroker2

martindo72050, 2020-08-07:
I'm having a similar issue with my arduino mega. I get the connection, but then get timeout errors. I'm able to read rpm using AT commands. I'm also new to this so it may be my inexperience. 
[image]

T-T-dev, 2020-09-18:
I am working on something similar myself; ESP32 with general ELM327 bluetooth adapter. I can pair just fine, but when I try to read rpm; I get a TIMEOUT error. Did you have further success on your end?

PowerBroker2, 2020-09-18:
What happens if you use the test sketch and try manually inserting AT commands/queries?

MGC1STOKI, 2021-04-25:
Im having similar issues with my WIRED ELM327 where i just get error 7.
Without defining which OBD protocol im using i managed to receive data but only after 30secs or so please see this log : 

[Success after 30secs.txt](https://github.com/PowerBroker2/ELMduino/files/6371995/Success.after.30secs.txt)

This was not reliably repeatable but when defining which protocol i want the ELM327 to use, in my case '6', in an attempt to fix the issue it no longer provides data after any amount of time.

Using the Torque app on my phone with a bluetooth adapter revealed what the correct protocol was so i know that ive got the right one for my car.

PowerBroker2, 2021-04-25:
Are you using the latest version of the library?

zxrxth, 2022-04-26:
Hi, i tried this code that i got from https://forum.arduino.cc/t/elmduino-library-for-car-hacking/593482/16#msg4679253

#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

//String MACadd = "00:10:CC:4F:36:03";                        //enter the ELM327 MAC address
uint8_t address[6]  = {0x00, 0x10, 0xCC, 0x4F, 0x36, 0x03};  //enter the ELM327 MAC address after the 0x

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);

  if (!ELM_PORT.connect(address))            //"OBDII" replaced by "address"
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while (1);
  }

  if (!myELM327.begin(ELM_PORT,true, 2000, AUTOMATIC))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
  {
    myELM327.printError();
  }
}

void printError()
{
  Serial.print("Received: ");
  for (byte i = 0; i < myELM327.recBytes; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.nb_rx_state == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.nb_rx_state == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.nb_rx_state == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.nb_rx_state == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.nb_rx_state == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.nb_rx_state == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.nb_rx_state == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(100);
}

Problem:
ELM_TIMEOUT

Some description:
It was successfully connected based on output on the serial monitor and green light on ELM 327. But there is no blue light on ESP32, only red light which seems that the ESP is only turned on but there's no connectivity. Correct me if I'm wrong. But I'm not sure if this is related to the ELM_TIMEOUT. I also tried your suggested solution but nothing happened. When i did the AT command, nothing comes out too. Is there any error on the code? I appreciate your time for looking into it.