PowerBroker2/ELMduino #19: Make Lib Non-Blocking

PowerBroker2, 2020-04-20:
See title

PowerBroker2, 2020-05-19:
Been hesitant to add this feature. If anyone wants this to be added to the lib, please let me know here in the comments.

patfelst, 2020-05-20:
I just discovered this library yesterday, and will get a chance to try out in the next week or two so will give you feedback then. Would you use some kind of scheduler?

PowerBroker2, 2020-05-20:
Maybe. I was thinking about doing something similar to what I accomplished in [this serial communication library](https://github.com/PowerBroker2/SerialTransfer). Maybe I could implement some sort of command queuing system? What are your thoughts?

av4625, 2020-06-06:
I was the guy on the Arduino forum that mentioned this too you back when I started using the library. I since got an ESP32 so I can read the OBD data on one core and do the stuff that I didn't want to get blocked (like flash a led at specific intervals) on the other core. Finally my project is getting close to completion and I will let you know if using the two cores of an ESP32 is a decent work around.

patfelst, 2020-06-07:
Hi Aaron, I'm interested in what you're doing with dual cores on the ESP32, keep me in the loop!

av4625, 2020-06-16:
Forgot to get back here. The two cores on the ESP32 work a treat. I can get my data using Elmduino on core 0 as that is the core that does bluetooth stuff so makes sense? Then I can do my other stuff on core 1 that is timing specific.

The one thing that bothers me is that you can have true non blocking code in the tasks that run on specific cores like you can in the `loop` function even though I think it is basically like a task on core 1. If you have true non blocking tasks you "starve" the `IDLE` task, this is the task that does the cleanup of deleted tasks and the like. I found that a putting `delay(1)` in your tasks loop is faster than calling 'taskYIELD()` in the loop (weirdly?). This stops watchdog from complaining that you are "starving" any `IDLE` task

PowerBroker2, 2020-06-16:
That's really neat. @av4625 - Can you post an example sketch that uses 2 cores and ELMduino.h? I can add it as an example sketch in the lib.

av4625, 2020-06-17:
Yea I can fork and make a pull request if you like?

PowerBroker2, 2020-06-17:
Yeah, that works too

av4625, 2020-06-22:
Created the pull request

PowerBroker2, 2020-06-22:
Merged - thank you!

patfelst, 2020-07-04:
I appreciate @av4625 Aaron's dual core example, but if I don't want to go down that path, can you confirm this is the bit that makes the library blocking?
https://github.com/PowerBroker2/ELMduino/blob/39dc2fb703f9c09e65db1fef2bec575f9564e749/src/ELMduino.cpp#L526-L528
If I wanted to make this non-blocking, would I remove this "receive" code from the SendCommand method? And a new receive method would be needed which would check for available bytes, and skip to next code if none?
Thanks again

PowerBroker2, 2020-07-04:
Yeah, that's the blocking code - you'd have to redesign the parsing system and add other infrastructure (i.e. command/response queues, extra status codes, etc).

patfelst, 2020-07-04:
thanks. Instead of queues, couldn't you just have a flag that says don't send another command until a the current one is complete - i.e. still waiting for a response?

quichedood, 2021-12-23:
Hi',

I'm working since a few years on a data logger based on Arduino.
I made many versions with different hardware, screens, functionalities.

Anyway the last one (v7) : [https://github.com/quichedood/DAWA-7.x](url)
Demo : [https://youtu.be/l66AxoFr9uk](url)

I setup a "stn2100" (ELM327 like) for the hardware part and give a try with your library.
It works pretty well as I can get for example speed of my car.

As you can guess, the "blocking" part of the library is the main drawback in my case as some data should be updated in realtime.

I don't know how much work this feature represent, probably a lot ... but if you read this and are interested by this project please tell me :)

PowerBroker2, 2021-12-23:
I'm still interested in finding a good way to make this lib non-blocking, but other projects and life stuff are higher priority for me right now. From the looks of your linked project, you're an impressive software dev - if you want to fork and find a good way to make the lib non-blocking without changing the API too much and submit a PR, I'd be happy to merge it.

quichedood, 2021-12-29:
I'm not an impressive software dev but it works :D
I'll try to find a way to make your lib non blocking but that's not guaranteed ... as you said life stuff take up much time !

Anyway, I'll keep you informed about developments.

patfelst, 2021-12-30:
Hi there, just letting you know I'm having a go at making a non-blocking version, just in case anyone else is. I should be able to make a PR in the next couple of days.

quichedood, 2021-12-31:
That's a pretty good news, I've just ordered an OBD simulator but you'll probably have the job done before I receive it :).
Anyway, I'll be happy to test/help/debug your version.

patfelst, 2022-01-01:
I'm about to submit a pull request. I've tested this on my own car and it is working ok. This represets about 3 days full time work for my coding / dev speed 😄. Things to note:

1. @PowerBroker2 consider if you want to keep the blocking version of the API. This shouldn't be too hard, but might be confusing for new people using the library. For now I have replaced the blocking code with non-blocking.
2. I tried hard to keep the API unchanged as far as I could. After querying a PID, the application code needs to check if a message is still being received if the response is !ELM_SUCCESS, see examples `else if (myELM327.nb_rx_state != ELM_GETTING_MSG)`. An error has occurred if the response is none of these two values.
3. I have only implemented the `rpm` and `engineCoolantTemp` PID functions as examples so you guys can review the code. Unfortunately the trade-off for non-blocking is that the PID functions are a few more SLOCs. If and when it's accepted, then we can implement all the other PID functions. All the others are commented out.
4. I use PlatformIO and the C++ Intellisense extension to Visual Studio Code. It has a automatic formatting feature, so some changes are purley due to the formatter. It wasn't worth me manually undoing these so they wouldn't show up as changes.
5. I added a function `batteryVoltage` to read vehicle battery voltage. This is not actually a PID, it's just an AT command. This can be used as an example of how to issue AT commands and get response in a non-blocking fashion.
6. I created `sendCommand_Blocking` as it is handy for issuing blocking AT commands like in the `initializeELM` function.
7. There is one change that is unrelated to non-blocking. In PID queries, I added a `num_responses` to the query string. This is normally set to "1" for most PIDs, unless you kmow there is a multi-line response. It means the OBD scanner won't wait around to see if more responses are coming from the vehicle's ECU once it has received `num_responses` responses. This has two positive effects: 1) faster frame rates, I have achieved up to 15 Hz querying two PIDs. 2) no duplicate responses which @PowerBroker2 you see in your scanner. This also means the code in `findResponse` to identify `secondHeadIndex` is pretty much redundant (I left the code unchanged however as it does no harm).
8. I have provided 3 examples. The one called `examples/ESP32_M5Stack_Core2_non_block/non blocking M5Stack Core2.ino` could be removed, but I think quite a few people would appreciate this. In `examples/non_blocking_2/non blocking example 2.ino` I have put in a simple scheduler to show how to query a number of PIDs at different rates.

patfelst, 2022-01-01:
Hi guys, please see pull request https://github.com/PowerBroker2/ELMduino/pull/102

quichedood, 2022-01-01:
Impressive, that was fast !
I'll give a try next week as soon as I can on my car.

PowerBroker2, 2022-01-01:
PR merged - I have some additional updates to roll in before tagging out a new version. The edits I have in mind are updating the legacy example sketches and built-in functions to the new API.

quichedood, 2022-01-02:
I did a very quick try on my car tonight and RPM are updated without blocking anything else !
I'll try this week with more data requests.

patfelst, 2022-01-03:
some more updates in PR https://github.com/PowerBroker2/ELMduino/pull/103

PowerBroker2, 2022-01-06:
Fixed in [release 3.0.0](https://github.com/PowerBroker2/ELMduino/releases/tag/3.0.0)

closed 2022-01-06 by PowerBroker2

quichedood, 2022-01-08:
First, thanks to both of you for your amazing work :)

I made some tests on my car, here are the results :

**If I request every 0,1 sec a single PID (RPM for example) I receive an updated value every 1 sec approximately.**
I've read that there are some technics to go faster but that seem pretty slow, isn't it ? I'm pretty sure I saw someone here speaking about ~15Hz data rate ... 

**If I request every 0,1 sec two different PID (RPM and engine coolant temp) I don't get anything.**
Here is the code I use :
`tempOBD = myELM327.rpm();
if (myELM327.nb_rx_state == ELM_SUCCESS && tempOBD > 0.0)
{
  g_rpm = (uint32_t)tempOBD;
}
else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
{
  myELM327.printError();
}

tempOBD = myELM327.engineCoolantTemp();
if (myELM327.nb_rx_state == ELM_SUCCESS && tempOBD > 0.0)
{
  g_engineCoolantT = (uint32_t)tempOBD;
}
else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
{
  myELM327.printError();
}`

Is there a better way to request multiple PID (even if that's only every second for example) ?
Maybe I have to use a queue and continuously : request PID 1, wait result of PID 1, request PID 2, wait result of PID 2, ...

PowerBroker2, 2022-01-08:
Your code tags are wrong. I usually fix them for people but I've decided to stop. [See here for proper code tags](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting).

As for the code, you're pinging both PIDs at the same time, when you have to ping one PID and wait until you either get a response or error before pinging another (or even before pinging the same PID again).

patfelst, 2022-01-09:
No prolem @quichedood. Please read the updated readme on the main page. With regard for a single PID taking 1.0 seconds, I'm not sure what's happening there. Yes I have achieved 15Hz rate when querying two PIDs sequentially.

I did provide an example that showed how to query multiple PIDs using a variable which kept track of which PID to query (i.e. a state machine) however PowerBroker2 didn't think it was needed. The [example](https://github.com/patfelst/ELMduino/blob/8780d5fb05bf097b2e0d1d0268ca075110b51a36/examples/non_blocking_2/non%20blocking%20example%202.ino) is still in my original pull request, have a look and see if it helps.

This is not your question / problem, but there actually is a way to call multiple PIDs in a single query. It makes the decoding more complex though becuase the payload buffer contains the responses to all PIDs. Perhpas that is a pull request for another holiday period! You can try it yourself you're adventurous, e.g. 
```c++
      char OBD_command[8] = {0};
      sprintf(OBD_command, "01%02X%02X1", INTAKE_MANIFOLD_ABS_PRESSURE, ENGINE_RPM);
      send_command(OBD_command);
```
The response will look like:
`
payload = 41 0B 63 0C 12 C6 (99 kPa and 1201 RPM)
`
So you'd need to extract the 0x63 which is the manifold pressure, and the (0x12 << 8 + 0xC6) / 4 = 1201 which is the RPM.

quichedood, 2022-01-09:
> Your code tags are wrong. I usually fix them for people but I've decided to stop. [See here for proper code tags](https://docs.github.com/en/github/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks#syntax-highlighting).

Sorry for that, I edited the post.

> As for the code, you're pinging both PIDs at the same time, when you have to ping one PID and wait until you either get a response or error before pinging another (or even before pinging the same PID again).

Crystal clear.

@patfelst : I'll dig about the refresh rate but before anything else I'll try your example with "switch" statement.
Multiple PIDs in a single query is nice, that should probably be faster.