PowerBroker2/ELMduino #20: Make Lib use a WiFi Telnet

pascaltippelt, 2020-04-26:
Hi,

it is me again... I finally got one thing working: Use a ESP32 to connect it to a WiFi OBD adapter:

```c++
#include <WiFi.h>

#include <HTTPClient.h>

#include <elapsedMillis.h>

elapsedMillis printTime = 1000;

//#define ELM_PORT client
#define DEBUG_PORT Serial

WiFiClient telnet;
const int telnetPort = 35000;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  
  WiFi.begin("WiFi_OBDII", NULL);
  while (WiFi.status() != WL_CONNECTED) {
     delay(500);
     Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
    
  
  if (!telnet.connect("192.168.0.10", telnetPort)) {
     Serial.println("connection failed");
     return;
  } else {
    telnet.print("AT Z");
    telnet.print('\r');
  }  
}

void loop() {
  // put your main code here, to run repeatedly:
  //EMPFANGEN
  while (telnet.available() > 0) {
     char c = telnet.read();

     if(c == '>') {
        Serial.println();        
     } else {
      Serial.write(c);
     }
  }
  //SENDEN
  if (printTime > 1000) {
  if (telnet.connect("192.168.0.10", telnetPort)) {
    telnet.print("AT RV");
    telnet.print('\r'); 
    printTime = 0;
  } else {
    Serial.println("connection failed");
  }}  
}
```
[This is my fast and dirty 'does it work' code]

Basically those adapters provide an open WiFi network ("WiFi_OBDII") with a telnet server at port 35000.

This telnet server passes through the serial interface. If you send a AT Z, you will get your response.

May it be possible to use this connection for your (by the way great) library?

Best regards

Pascal

PowerBroker2, 2020-04-26:
Thank you for the kind compliments on the lib, I'm glad it's working for you :)

I'll see what I can do to incorporate WiFi into the library and add a new example.

PowerBroker2, 2020-04-26:
Actually, I think I got ahead of myself - I already have an ESP32 WiFi example [here](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_WiFi/ESP32_WiFi.ino). Does it not work for you?

pascaltippelt, 2020-04-27:
Well, your programming skills are faster than the light. Yes, it works, I can finally get values from my old car. :blue_car: :+1:

There is one minor issue with those cheap WiFi adapters: They sometimes close the telnet connection (maybe if there were no requests for a certain time or so). So does you lib check if the telnet client is still connected before a request and, if not, tries to reconnect? Or should this be done manually?

pascaltippelt, 2020-04-27:
Here you can see the first succesfull run of my implementation (still slow as I only poll data once a second).

https://youtu.be/nSUs2budr2M

PowerBroker2, 2020-04-27:
> There is one minor issue with those cheap WiFi adapters: They sometimes close the telnet connection (maybe if there were no requests for a certain time or so). So does you lib check if the telnet [...]

To keep the library as flexible and simple as possible, any automatic reconnects should be done at sketch-level. You could test for connection issues in the error printing function.

> Here you can see the first succesfull run of my implementation (still slow as I only poll data once a second).
> 
> https://youtu.be/nSUs2budr2M

Looks good! Are you having problems with the connection if you poll faster than 1Hz?

pascaltippelt, 2020-04-28:
No I don't have problems getting more values. It is for being able so "see" the single values.

Everything fine!

closed 2020-04-28 by PowerBroker2

pascaltippelt, 2020-04-29:
So in the loop() I would check Telnet state and then get my value like this?
```c++
if (!WiFi.connected()) {WiFi.connect();}
if (!client.connected()) {client.connect(server, 35000);}
//get Values...
```

By the way... When exactly is it required to `myELM327.flushInputBuff()` ?

PowerBroker2, 2020-04-29:
I don't have much experience in WiFi programming, but it looks ok to me. Googling might produce some good example code.

`myELM327.flushInputBuff()` is automatically called so you don't have to. If you experience a lot of ELM327 status errors, you could try calling that function after printing the error, but idk if that would help.