PowerBroker2/ELMduino #21: Small ELM327::initializeELM improvements

ziotibia81, 2020-04-29:
I try to suggest some small improvements to ELM327::initializeELM :

A previously used software may have done settings adjustments  that prevent ELMduino from communicating properly with ELM327. So it is a practice for many software to send ATZ as a first command.

ELMduino initialize protocol using AT SP. This write the protocol to the ELM327 eeprom every time. Anoter practice is to use AT TP (or AT TP A), that doesn't store default, and fallback to AT SP if previous command fails (some ELM327 doesn't support AT TP).

PowerBroker2, 2020-04-29:
I added the `AT Z` call during initialization, but I don't fully understand the usage of `AT TP`. Do you have any examples of this call being used?

ziotibia81, 2020-04-29:
My explanation is not so clear, you are right! ;-)

AT SP5 write 5 to elm eeprom as default protocol and select it as current protocol. 

AT TP5 select protocol 5 as current protocol but don't write to eeprom the default protocol. 
This to extend eeprom life.

http://www.elmelectronics.com/DSheets/ELM327DS.pdf

PowerBroker2, 2020-04-29:
ok, I see. I'll think about it.

closed 2020-05-09 by PowerBroker2