PowerBroker2/ELMduino #22: Make PAYLOAD_LEN redefinable.

IW3IJQ, 2020-05-02:
Now PAYLOAD_LEN is fixed to 40.
Some ECU have custom pids with custom reply.
For example mine reply with 36, 44 and 126 characters on ELM side.

PowerBroker2, 2020-05-03:
Added this functionality in [the current **dev branch**](https://github.com/PowerBroker2/ELMduino/tree/dev). Can you test to see if it works for you?

Once the dev branch is validated, I'll merge it to master and re-release.

IW3IJQ, 2020-05-03:
Thanks!
I have updated my code. 
Now I can't test it, but I will give you feedback as soon as possible.

ELM expect 7 bytes max queries and replies. To leave this limit  it need "AT AL" setting.
I send this after initialization. 
You could think of sending it automatically according to payload size.

PowerBroker2, 2020-05-03:
That's a good point, I might add `AT AL` in `begin()` automatically. I'm guessing, however, this change won't affect normal PID queries (i.e. speed or rpm)?

IW3IJQ, 2020-05-03:
ELM327 datasheet indicate only that this command remove the 7 bytes limit. So I think that's ok for normal PID queries.

PowerBroker2, 2020-05-03:
Added `AT AL` in the dev branch

PowerBroker2, 2020-05-12:
Fixed in [release 2.1.3](https://github.com/PowerBroker2/ELMduino/releases/tag/2.1.3). Thanks!

closed 2020-05-12 by PowerBroker2