PowerBroker2/ELMduino #23: How can i set the pin number (1234)?

Tinkerpete, 2020-05-11:
How can i set the pin number (1234)?

PowerBroker2, 2020-05-11:
You shouldn't need to. Can you describe your hardware setup and post your entire code (in tags)?

Tinkerpete, 2020-05-11:
Thank you for your answer. I have tried all the examples from ELMDuino but it shows only:

'Couldn't connect to OBD scanner - Phase 2'

My ELM 327 is 9 or 10 years old. I can connect it with my android smartphone and torque. But i must provide '1234'.

This is one of the codes i have tried:

```
#include "ELMduino.h"

#define ELM_PORT Serial1

ELM327 myELM327;

char *pin = "1234"; //<- standard pin would be provided by default

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  Serial.begin(115200);
  ELM_PORT.begin(115200);
 
  Serial.println("Attempting to connect to ELM327...");

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
    printError();
}

void printError()
{
  Serial.print("Received: ");
  for (byte i = 0; i < PAYLOAD_LEN; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();
 
  if (myELM327.status == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.status == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.status == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.status == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(100);
}
```

PowerBroker2, 2020-05-11:
When I connect my ELM327 to my phone or laptop, I also need to use the pin, but not with my Arduinos. What Arduino board are you using?

Tinkerpete, 2020-05-11:
My test board is a TTGO T_Display, a little ESP32 board with a little OLED display. I compile the code with the Arduino IDE and 'ESP 32 Dev module'.

https://www.banggood.com/LILYGO-TTGO-T-Display-ESP32-CP2104-WiFi-bluetooth-Module-1_14-Inch-LCD-Development-Board-p-1522925.html?rmmds=search&cur_warehouse=UK

PowerBroker2, 2020-05-11:
A couple of things:

1. Are you sure you've unpaired the ELM327 from all other devices before attempting to connect to it via your ESP32?
2. Does your ESP32 have built-in Bluetooth or are you using an HC-05/6?
3. Are you making sure the car is running when you try to connect the ESP32 to the ELM327?

Tinkerpete, 2020-05-11:
for 1: No, i will test tomorrow.
for 2: It has built in bluetooth, because i can find it with my smartphone.
for 3: No, the key is turned but the engine did not run. But this works with 'torque'.

PowerBroker2, 2020-05-11:
> for 2: It has built in bluetooth, because i can find it with my smartphone.

In that case, you will want to use [this sketch](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino) to see if your ELM327 can respond to AT commands you send via your Serial Monitor. If the AT commands work, you can then run [this sketch](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino) **while the car is running** to test if you can get data.

pascaltippelt, 2020-05-12:
I had strange issues with ESP32 + ELM327 + Bluetooth too. Until now, I found no way to re-use a bluetooth ELM327 adapter that was used by torque before.

Tinkerpete, 2020-05-12:
Did you mean with 'Serial Monitor' the function in the Arduino IDE?

Tinkerpete, 2020-05-12:
@PowerBroker2: I have tried it again, but i think this codes from you are the same, i have tried before! No new thoughts for me. It has not worked.

Tinkerpete, 2020-05-12:
@pascaltippelt: I have read from your problems in an forum, i thinks so. I will try to reset the ELM327, but i have not find a way in the moment. I will try to connect with a notebook and then use a serial monitor.

PowerBroker2, 2020-05-12:
Pushed an update to the library today. See [release 2.1.3](https://github.com/PowerBroker2/ELMduino/releases/tag/2.1.3). One of the things in this new release is the automatic resetting of all settings to factory defaults upon startup. This might clear any weird settings Torque might be applying.

Please update your library and see if it fixes things for you.

Tinkerpete, 2020-05-12:
@PowerBroker2: I have tried the new library in a quick test only with the ESP32. It did not work. But i will tomorrow make a second test with my notebook for debugging.

Tinkerpete, 2020-05-12:
Interesting behavior without ELM327: It shows 'Couldn't connect to OBD scanner - Phase 2' after ca. 30 seconds! The same message when the ELM327 is in the near! This means the test '!ELM_PORT.connect("OBDII")' did not work.

george0305, 2020-05-13:
This is the same issue i`m having, posted in March about it in another issue. Coming back, it still persists and for now the only solution i got is:
Pair the ELM327 with the phone
After pairing, unpair it quickly
Restart the ESP32 so it searches for ELM327 again.
It should pair.
However, the next time you want to pair it / if you restart the ESP, you have to do the same steps again.

Tinkerpete, 2020-05-13:
@george0305: Thank you for the interesting information. I have tried ist, but it did not work. Because i was in hurry, i will try it tomorrow again.

@All: Today i could make a connection to the ELM327 over my win 10 notebook. I must transmit the PIN '6789' to get the connection. Then i could submit 'AT Z' over a 'Bluetooth Serial Terminal' and the ELM327 answer me with 'ELM327 v1.4'.

PowerBroker2, 2020-05-13:
After connecting and submitting `AT Z`, did you try to then unpair your notebook and then retry with your ESP32?

Tinkerpete, 2020-05-13:
@PowerBroker2: I think so, but i must give the car to a other person and i will try tomorrow again. I have found with Google, that i can flash the ELM327 with the tool 'Deep OBD for BMW and VAG' to Firmware 1.5. But i did not know how.

Tinkerpete, 2020-05-14:
Has anyone a idea how i could flash a new  firmware to the ELM327? My trys with the Deep OBD for BMW and VAG' not work.

pascaltippelt, 2020-05-14:
> Has anyone a idea how i could flash a new firmware to the ELM327? My trys with the Deep OBD for BMW and VAG' not work.

As far as I understand the topic, this is not possible. Those cheap ELM327 adapters are all based on a leaked firmware and most are unlicensed - well illegally copied by the mostly asian manufacturer.

A original ELM Chip is way more expensive than those 10$ things.

ziotibia81, 2020-05-18:
> Interesting behavior without ELM327: It shows 'Couldn't connect to OBD scanner - Phase 2' after ca. 30 seconds! The same message when the ELM327 is in the near! This means the test '!ELM_PORT.connect("OBDII")' did not work.

Hi!
Current release of arduino-esp32 (1.0.4) have a bug on Bluetooth serial.
You have to use sources from master branch:

https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

ziotibia81, 2020-05-18:
> How can i set the pin number for the ELM 327 over bluetooth with the ESP32 (1234)?

@Tinkerpete to set pin you have only to call `setPin()` after BT init.
See row 24 here: https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino

Tinkerpete, 2020-05-18:
@ziobia81: Thank you for your info! I have copied the 'BluetoothSerial.cpp' and 'BluetoothSerial.h' to my local installation. I have insert setPin before and after 'begin' with '6789', what is my PIN for the ELM327. Now it stopps with "Couldn't connect to OBD scanner - Phase 1"!!!

IW3IJQ, 2020-05-18:
> Pushed an update to the library today. See [release 2.1.3](https://github.com/PowerBroker2/ELMduino/releases/tag/2.1.3). One of the things in this new release is the automatic resetting of all [...]

@PowerBroker2 I think that you accidentally removed `sendCommand(RESET_ALL)` in this commit:
https://github.com/PowerBroker2/ELMduino/commit/395d0cad778e571260f1a21d6312eed5a95f7d36

PowerBroker2, 2020-05-18:
> > Pushed an update to the library today. See [release 2.1.3](https://github.com/PowerBroker2/ELMduino/releases/tag/2.1.3). One of the things in this new release is the automatic resetting of all settings to factory defaults upon startup. This might clear any weird settings Torque might be applying.
> > Please update your library and see if it fixes things for you.
> 
> @PowerBroker2 I think that you accidentally removed `sendCommand(RESET_ALL)` in this commit:
> [395d0ca](https://github.com/PowerBroker2/ELMduino/commit/395d0cad778e571260f1a21d6312eed5a95f7d36)

Oops, I'll go ahead and add that back in, Thanks!

PowerBroker2, 2020-05-18:
Thanks all for the help - all things should be fixed in [release 2.1.4](https://github.com/PowerBroker2/ELMduino/releases/tag/2.1.4)

Tinkerpete, 2020-05-20:
@PoerBroker2: I have tested the new version with no luck. Which Bluetooth library did you use for testing?

PowerBroker2, 2020-05-20:
There was a slight mixup - [release 2.2.0](https://github.com/PowerBroker2/ELMduino/releases/tag/2.2.0) is actually the version you want to use. If it's still not working, please post your entire sketch and a description of exactly what's going wrong.

patfelst, 2020-05-21:
> @Tinkerpete
> 
> > Interesting behavior without ELM327: It shows 'Couldn't connect to OBD scanner - Phase 2' after ca. 30 seconds! The same message when the ELM327 is in the near! This means the test '!ELM_PORT.connect("OBDII")' did not work.
> 
> Hi!
> Current release of arduino-esp32 (1.0.4) have a bug on Bluetooth serial.
> You have to use sources from master branch:
> 
> https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

I'm having the same issue, it keeps crashing on the ELM_PORT.connect(address) call. I'm using a [OBDLink MX+ Professional OBD2 Scanner](https://www.scantool.net/obdlink-mxp/) and just before it crashes, the OBD and Host LEDs flash, so bluetooth begin is working. About every 10 seconds it reboots, the LEDs flash in a continuous cycle.

I updated the bluetooth library from master branch, but it's still crashing. Here's the screengrab of the crash. I realise this isn't an issue with your library but I've no idea how to solve this. Tried three different ESP32 dev boards, different USB cables, different power supplies.
[image]

Code is here https://pastebin.com/S0em3wnD

PowerBroker2, 2020-05-21:
@patfelst , yours is an unrelated problem, please submit a new issue for this.

Tinkerpete, 2020-05-21:
I tried everything again. With this

```#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  SerialBT.setPin("6789");
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
    printError();
}

void printError()
{
  Serial.print("Received: ");
  for (byte i = 0; i < myELM327.recBytes; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();
  
  if (myELM327.status == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.status == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.status == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.status == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(100);
}´´´´

and this

´´´#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

#define DEBUG_PORT Serial
#define ELM_PORT   SerialBT

void setup()
{
  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("ESP32test", true);
  ELM_PORT.setPin("6789");

  DEBUG_PORT.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);
  }

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}

void loop()
{
  if(DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();

    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }

  if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if(c == '>')
      DEBUG_PORT.println();

    DEBUG_PORT.write(c);
  }
}´´´

I tried 'SerialBT.setPin("6789");' once before and once after 'ELM_PORT.begin("ArduHUD", true);'. I'm still not sure if I should try this now with the current library from here https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial with the default library installed.

I suspect that my ELM327 is too old and that it only works with newer ELM327.

Translated with www.DeepL.com/Translator (free version)

PowerBroker2, 2020-05-21:
Apparently you should use [the version of the BT library linked here](https://github.com/PowerBroker2/ELMduino/issues/23#issuecomment-630153538) to get `setPin()` to work.

> I suspect that my ELM327 is too old and that it only works with newer ELM327.

IDK if that's really the case, but new ELM327 clones cost ~$10, so verifying this should be relatively cheap.

PowerBroker2, 2020-05-21:
Also, are you sure `6789` is the correct pin and `OBDII` is the correct device name?

Tinkerpete, 2020-05-21:
'6789' is correct, i have verified with my smartphone. My smartphone shows in the bluetooth settings 'CHX' for the ELM327? Is this the correct name for the code? I have tried it, but it did not work. I will try in the next days with my notebook...

PowerBroker2, 2020-05-21:
In that case, you might try `ELM_PORT.connect("CHX")`

patfelst, 2020-05-22:
I found an alternative Bluetooth classic library [here](https://github.com/gmag11/BTClassicSPP), I might try it on the weekend.

george0305, 2020-05-23:
Hi! Coming with a big update, i managed to make it work. Here`s what i did:
No setpin needed, from the library ziotibia81 suggested(Thank you!) https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial i replaced the 2 files. Still wasnt working so there is an examples files with bt_remove_paired_devices. Modify the #define REMOVE_BONDED_DEVICES 0 to 1, upload and let it delete them. After that, upload your sketch again and it should work. Mine is now connecting by itself again.

ext2ext, 2020-05-24:
> @ALL: Today i could make a connection to the ELM327 over my win 10 notebook. I must transmit the PIN '6789' to get the connection. Then i could submit 'AT Z' over a 'Bluetooth Serial Terminal' and the ELM327 answer me with 'ELM327 v1.4'.

Hi, guys.
I have a successful connection to my OBD dongle, but could not get a response on my 'AT Z' command using a terminal program. I use ESP8266_WiFi.ino.
Thank you for any help.

PowerBroker2, 2020-05-24:
@ext2ext , please create a new issue with your particular problem.

Tinkerpete, 2020-05-24:
@george0305: Good found and idea! I have tried 'bt_remove_paired_devices.ino'. First it did not work by me. Then i commented out line 65 ('// count = PAIR_MAX_DEVICES;') and then it removed 1 bounded device! But ELMduino examples did not work.

PowerBroker2, 2020-05-24:
> But ELMduino examples did not work.

What do you mean? If something "doesn't work", please post your updated code and a detailed description of what exactly is going wrong. More details helps me fix things faster.

Tinkerpete, 2020-05-24:
@PowerBroker2: It is like before in this issue. I did not think that your code has a problem! I think my old ELM327 could be the problem.

PowerBroker2, 2020-05-25:
Sorry, trying to keep up with what everyone is saying is confusing, lol. Definitely try with a newer ELM327. If it isn't a problem specific to the library, I'm going to go ahead and close this. That being said, if you do find a related bug to your original post, feel free to reopen.

closed 2020-05-25 by PowerBroker2