PowerBroker2/ELMduino #27: ESP32 crashes during startup

patfelst, 2020-05-22:
Hi there, thanks for your work on this library. Unfortunately in my environment / setup, the ESP32 crashes during startup, I get the following backtrace from the serial debug monitor:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060330  A0      : 0x800e6ed0  A1      : 0x3ffc83c0  
A2      : 0x00000030  A3      : 0x0000002c  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffc8690  
A10     : 0x00000003  A11     : 0x00060323  A12     : 0x00060320  A13     : 0x3ffbc480  
A14     : 0x00000000  A15     : 0x3ffc68e0  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000030  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffc83c0 0x400e6ecd:0x3ffc83d0 0x400e5019:0x3ffc86e0 0x400d2462:0x3ffc87a0 0x400d24ee:0x3ffc87d0 0x400d1865:0x3ffc87f0 0x400d2f3f:0x3ffc8830 0x4008e0bd:0x3ffc8850

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
```
Pasting this into the [Exception Decoder](https://github.com/me-no-dev/EspExceptionDecoder) I get this:
```
PC: 0x400014fd
EXCVADDR: 0x00000030

Decoding stack results
0x400e6ecd: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1529
0x400e5019: sprintf at ../../../.././newlib/libc/stdio/sprintf.c line 644
0x400d2462: ELM327::initializeELM(char) at /Users/patrickfelstead/Documents/Arduino/libraries/ELMDuino/src/ELMduino.cpp line 101
0x400d24ee: ELM327::begin(Stream&, unsigned short, char) at /Users/patrickfelstead/Documents/Arduino/libraries/ELMDuino/src/ELMduino.cpp line 39
0x400d1865: setup() at /Users/patrickfelstead/Documents/OBD_bluetooth/OBD_bluetooth.ino line 26
0x400d2f3f: loopTask(void*) at /Users/patrickfelstead/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/cores/esp32/main.cpp line 14
0x4008e0bd: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
Easier to view with color coding, see screen grab:

[image]

As you can see it is crashing due to sprintf function in ```ELM327::initializeELM(char protocol)```

I think it's because you specified a string with %s for the auto search protocol character. So I made this change and **bingo - no more crashing!!**

Was:
```C++
const char * const TRY_PROT_H_AUTO_SEARCH     = "AT TP A%s"; // OBD
```
changed to
```C++
const char * const TRY_PROT_H_AUTO_SEARCH     = "AT TP %c"; // OBD
```
Now I'm at the point where I see `Connected to ELM327` which is great! But it's not reading RPM.  I haven't had the time to debug further, but at least I'm communicating with my ELM clone dongle, so I can try some basic commands like reading voltage with `AT RV` etc. I believe I can use your ESP32_test.ino sketch to do that.

PowerBroker2, 2020-05-22:
Can you update from master and see if it fixes the problem? Notice that I also switched around the input arguments on `begin()`, so now it's: `bool ELM327::begin(Stream &stream, char protocol, uint16_t payloadLen)`

patfelst, 2020-05-23:
Yep fixed now. I am calling `myELM327.begin(SerialBT, '6'))` as I know my BMW uses CAN bus 11-bit at 500kbps.
When specifying protocol = '0' or default (also '0') I've had to increase the timeout to 5000ms otherwise it will connect, but times out when querying a PID. When I specify protocol 6, the 1000ms timeout is ok.

patfelst, 2020-05-23:
I'm also finding reconnecting after resetting the ESP32 or uploading a new program is quite unreliable, I have to press the bluetooth pair button on my OBD dongle otherwise the EML327.begin fails. I suspect either I have to close the OBD connection, or close the bluetooth connection before uploading again. I will investigate.

patfelst, 2020-05-23:
Sorry one last thing, in your code update I notice that here https://github.com/PowerBroker2/ELMduino/blob/c84c0c4d7b188ad3c764175fde8e1eb265e009e5/src/ELMduino.cpp#L114 you are not using the `command` variable, you've hard coded `SET_PROTOCOL_TO_H_SAVE`, I fixed that even though it never gets executed as the for me the function returns in the AUTO search code above.

patfelst, 2020-05-23:
> I'm also finding reconnecting after resetting the ESP32 or uploading a new program is quite unreliable, I have to press the bluetooth pair button on my OBD dongle otherwise the EML327.begin fails. I [...]

ok I think I solved this in two steps, and am now getting really reliable connections every time I reboot, or upload a program:
1) I had my iPhone paired with the OBD scanner. On every reboot, the iPhone was jumping in and connecting to the scanner over bluetooth and locking out the ESP32 - oops.

2) by putting in a 1 second delay between starting the ESP32 bluetooth module and the connecting with the scanner.

So this is my setup() now:

```C++
void setup() {
  
  Serial.begin(115200);

  // Start the ESP32's bluetooth classic module and advertise as "Boost"
  Serial.println("Starting ESP32 bluetooth classic module");
  SerialBT.begin("Boost", true);
  delay(1000);
  
  // Try and connect to the OBD scanner over bluetooth
  sprintf(bt_scanner_mac, "%02X:%02X:%02X:%02X:%02X:%02X", address[0], address[1],address[2],address[3],address[4],address[5]);
  Serial.printf("Trying bluetooth connection with OBD scanner %s\n", bt_scanner_mac);
  bool connected = SerialBT.connect(address);
  
  if (!connected) {
    SerialBT.end();
    SerialBT.unpairDevice(address);
    Serial.printf("Couldn't make bluetooth connection with OBD scanner %s. "
            "Disconnected from OBD scanner, and unpaired it.\n", bt_scanner_mac);
  }

  if (!myELM327.begin(SerialBT, '6')) {
    Serial.println("Couldn't start OBD scanner CAN protocol 6");
    while (1);
  }

  Serial.println("Connected to ELM327");
}
```

Also I'm not sure if it helps, but to end the program, I'm searching for a keypress on the serial monitor, and if I press 'd' it closes the bluetooth connection.
```C++
  if(Serial.available()) {
    char c = Serial.read();
    if (c == 'd') {

      // if (myELM327.sendCommand("STPC") == ELM_SUCCESS)
      //   Serial.printf("Closed Protocol %s\n", myELM327.payload);

      Serial.println("Closing bluetooth connection");
      SerialBT.end();
      Serial.println("Halting program ... ");
      while(1);

    }
```

PowerBroker2, 2020-05-25:
Everything should be fixed in [2.2.1](https://github.com/PowerBroker2/ELMduino/releases/tag/2.2.1). Thanks!

closed 2020-05-25 by PowerBroker2

patfelst, 2020-05-25:
excellent, thank you!

patfelst, 2020-05-25:
> 2\. by putting in a 1 second delay between starting the ESP32 bluetooth module and the connecting with the scanner.

PS With further testing I found 1 second delay is not needed for reliable operation during restart