PowerBroker2/ELMduino #28: How to send an AT command and get respond via WiFi?

ext2ext, 2020-05-24:
I have a successful connection to my OBD dongle, but could not get a response on my 'AT Z' command using a terminal program. I use ESP8266_WiFi.ino.

Connected to Wifi
IP address: 
192.168.0.1
connected

Thank you for any help.

PowerBroker2, 2020-05-24:
> I have a successful connection to my OBD dongle

What do you mean by that? How do you know?

> but could not get a response on my 'AT Z' command using a terminal program. I use ESP8266_WiFi.ino.

Are you using an ESP8266? What response does it give when you type in `AT Z`? How are you sending AT commands when ESP8266_WiFi.ino doesn't support this?

Also, are you doing this testing with your car on and running?

ext2ext, 2020-05-24:
> > I have a successful connection to my OBD dongle
> What do you mean by that? How do you know?
I just powered up my OBD scanner at home. It creates a WiFi network and I got IP address using your ESP8266_WiFi.ino example (see log above).

> > but could not get a response on my 'AT Z' command using a terminal program. I use ESP8266_WiFi.ino.
> Are you using an ESP8266?
Yes, I am. ESP8266 01.

>  What response does it give when you type in `AT Z`?
Nothing. Expecting FW version in the response.

> How are you sending AT commands when ESP8266_WiFi.ino doesn't support this?
That why I'm asking about it. I see that guys did it successful using your BT example.
Used Arduino's Serial monitor and another terminal software.

> Also, are you doing this testing with your car on and running?
My experiments are going without a car. I want to work with the configuration of elm327 chip. On my understanding it should respond to me without a car connection. In my mind, It should work as "any PC (W10) terminal program -> COM port -> USB to RS232 TTL UART adapter -> ESP8266 -> WiFi network -> ELM327 chip".

PowerBroker2, 2020-05-24:
Ok, I think I understand what you're trying to do now. The code that was being referred to was for an ESP32 connected to an ELM327 over standard bluetooth. Instead, you'll need to use a sketch similar to this:

```C++
#include <ESP8266WiFi.h>
#include "ELMduino.h"

const char* ssid = "WiFi_OBDII";
const char* password = "your-password";

//IP Adress of your ELM327 Dongle
IPAddress server(192, 168, 0, 10);
WiFiClient client;
ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  Serial.begin(115200);

  // Connecting to ELM327 WiFi
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid);
  // WiFi.begin(ssid, password); //Use this line if your ELM327 has a password protected WiFi

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("Connected to Wifi");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  if (client.connect(server, 35000))
    Serial.println("connected");
  else
  {
    Serial.println("connection failed");
    ESP.reset();
  }

  myELM327.begin(client);

  Serial.println("Connected to ELM327");
  Serial.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  Serial.println("Type and send commands/queries to your ELM327 through the serial monitor");
  Serial.println();
}

void loop()
{
  if(Serial.available())
  {
    char c = Serial.read();

    Serial.write(c);
    ELM_PORT.write(c);
  }

  if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if(c == '>')
      Serial.println();

    Serial.write(c);
  }
}
```

Let me know if you can get AT responses using the above code. If it works, I'll add it to the examples in the library.

ext2ext, 2020-05-24:
You lost the declaration:
ESP8266_WiFi:65:5: error: 'ELM_PORT' was not declared in this scope

PowerBroker2, 2020-05-24:
It's updated, try it again

ext2ext, 2020-05-25:
> It's updated, try it again
Perfect! Now it works how I expect.
Thank you so much!

closed 2020-05-25 by PowerBroker2