PowerBroker2/ELMduino #29: Some basic setup questions

cmanger2020, 2020-05-27:
Hey! I really appreciate the support. I'm new to the arduino universe, but I'm a big car enthusiast and very interested in making some basic displays for a track car I'm building. I bought one of the inexpensive bluetooth ELM327 modules and desoldered the bluetooth component so I could pass the serial TX/RX data to the arduino through pins 2 and 3 like in one of the examples. 

I have a (probably very basic, simple) question about the baud rate settings. How did you determine to use 115200 for both the serial and ELMserial ports? I found a few resources that said ELM327's have a baud rate of 38400, but if I used those values, it obviously didn't work. I was just curious about those settings. 

The other issue I had was when I did get the serial port to work (after it printed the "connected to elm327"), I would get around 10 values printed for the RPM (all which read 0), then it would go into the timeout error and just print that endlessly. 

I was wondering if you could help point me in the right direction. I don't have any experience with data communication, so it's proving to be a bit of a steep learning curve, so any advice/guidance is very much appreciated.

PowerBroker2, 2020-05-27:
I've only communicated with ELM327 modules wirelessly, so I don't know much about the physical wiring aspect. Although you seem to have gotten things to work (at least mostly), I'm very much interested in seeing pictures of what pads correspond to the ELM327's TX, RX and GND signals.

As for the baud, I honestly can't remember if my ELM327 happened to have a default baud of 115200 or if I used AT commands to change the baud from 38400.

Curious, did you have to edit any of the example sketch or the lib to get RPM readouts?

Lastly, the timeout error does occur from time to time with one of my personal projects (power cycling fixes it), but I've never been able to make heads or tails of why it happens. It's possible that using a baud of 115200 might introduce some comm errors. If you're experiencing timeout errors more than you care to deal with, try using AT commands to switch to a lower baud ([more here](https://www.elmelectronics.com/wp-content/uploads/2016/07/ELM327DS.pdf)).

patfelst, 2020-05-27:
sorry to hijack this issue, but did you see on the [ELM front page](https://www.elmelectronics.com/), they are shutting down in 2 years. Not that it really impacts this library as it's been cloned so many times. 
> Having to live with, and make adjustments for Covid-19 has reminded us of a great many things, however. Foremost is that life is very precious and our time here is very limited. For this reason, we have decided to wind down our business, and close it effective June 2022.

Personally I'm using the [OBD Link MX](https://www.scantool.net/obdlink-mxp/) which has a [STN1100 chip](https://www.scantool.net/downloads/98/stn1100-frpm.pdf), which uses a hybrid of the ELM "AT" commands, but also enhanced "ST" commands and I'm modifying your library to add a few of the ST commands.

da40flyer, 2020-06-03:
I am having the same issue.  After the rpm values scroll for a short time, it goes to an endless TimeOut Error.  I have not edited the sketch or the library at all.    I am using a VEEPEAK wifi adapter talking to an esp32 board. 
Suggestions?

PowerBroker2, 2020-06-03:
You might be able to jog the module back to life it if you send a string of a few carriage returns and/or call the `myELM.begin()` method again.

da40flyer, 2020-06-04:
I went into the sketch and simply changed it to:   Serial.begin(**38400**);    and it now stays connected.

PowerBroker2, 2020-06-08:
@cmanger2020 , due to lack of response I assume you have things working without further questions. I'm going to close the issue for now, but if you need more clarification feel free to reopen.

closed 2020-06-08 by PowerBroker2

siwar-w, 2022-03-15:
I can't find tx and rx on my elm327 v1.5 any help please