PowerBroker2/ELMduino #30: how to read voltage in sketch on Arduino IDE

lysiong, 2020-05-31:
I been scratch my head regarding how to put a "read voltage"(ATRV) command in sketch in order to display the result in serial monitor.

I already try ATRV command on ESP32_test.ino sample & it work, it show ">atrvatrv12.1V" on serial monitor. But when I issue "ATRV" command in sketch on ESP32_Bluetooth_Serial.ino sample, it show 0.
the command in sketch as below 

    Serial.print("Batt: "); 
    myELM327.sendCommand("ATRV");
    Serial.println(myELM327.findResponse());
    Serial.println("V");

it will show below result: 
Batt: 0
V
in serial monitor. 

By the way, I still at beginner stage for my Arduino command. FYI,  I'm using OBDLink LX as my OBDII bluetooth dongle & ESP32.

PowerBroker2, 2020-05-31:
What do you get if you use this code?

```C++
Serial.print("Batt: "); 
myELM327.sendCommand(READ_VOLTAGE);
Serial.write(myELM327.payload, myELM327.recBytes)
Serial.println("V");
```

lysiong, 2020-06-01:
Thanks for reply, I will try tonight & let you know the result.

lysiong, 2020-06-01:
@PowerBroker2 ,
seem like the code not working, it stuck at below code

Serial.write(myELM327.payload, myELM327.recBytes)

error message below

ESP32_Bluetooth_Serial:56:23: error: invalid conversion from 'char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

 Serial.write(myELM327.payload, myELM327.recBytes)

                       ^

I was using the example from your library ESP32_Bluetooth_Serial.ino 
remark out the below 3 lines

float tempRPM = myELM327.rpm();
 rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);

& add in the command that you give after the below command

if (myELM327.status == ELM_SUCCESS)

Serial.print("Batt: "); 
myELM327.sendCommand(READ_VOLTAGE);
Serial.write(myELM327.payload, myELM327.recBytes);
Serial.println("V");

Just found up if I modify the code become like below, it works !!
Serial.print("Batt: "); 
myELM327.sendCommand(READ_VOLTAGE);
Serial.write(myELM327.payload);
Serial.println(" ");

Result in serial monitor as below
Batt: 12.2V  <--I not sure where the V coming from ........

But if I change the code to below
Serial.print("Batt: "); 
myELM327.sendCommand(READ_VOLTAGE);
Serial.write(myELM327.recBytes);
Serial.println(" ");

Result in serial monitor as below
Batt:   <--I not sure what is generate out, is like a square symbol. 

So the data is in payload? what is recBytes store actually?

PowerBroker2, 2020-06-01:
Sorry about the error, I forgot the ESP32 handles `Serial.write()` strangely. Try this instead:

```C++
Serial.print("Batt: "); 
myELM327.sendCommand(READ_VOLTAGE);
Serial.write((const uint8_t*)myELM327.payload, (uint8_t)myELM327.recBytes)
Serial.println("V");
```

`myELM327.payload` is a buffer for the characters returned from the ELM327 and `myELM327.recBytes` is the total number of characters returned from the ELM327 and stuffed in the buffer `myELM327.payload`.

lysiong, 2020-06-01:
seem like the result is same for below 2 command,

Serial.write(myELM327.payload);
Serial.write((const uint8_t*)myELM327.payload, (uint8_t)myELM327.recBytes)
Batt: 12.1vv <-- ya, double V somehow, I just remove the print V command.

but if I print the result to LCD screen using (myELM327.payload), then the result is different.
LCD screen will show below
ATRV12.1v  <--show the ATRV command in front although I didnt specify it.

PowerBroker2, 2020-06-01:
Looks like the library is working as intended - the LCD portion depends on your code. Can you confirm?

lysiong, 2020-06-01:
for the LCD portion, I just code it as below
lcd.print(myELM327.payload); 

maybe need to add the filter for filter up the ATRV wording? I'm still very new to this coding, will take sometime to dig it or you have any idea?

PowerBroker2, 2020-06-01:
Maybe. Can you post your entire sketch?

lysiong, 2020-06-02:
I'm simplify my code as below, LCD library take from https://github.com/Bodmer/TFT_eSPI
the LCD will display result as below
OBD voltage: ATRV12.4V

#include <TFT_eSPI.h> 
#include "BluetoothSerial.h"
#include "ELMduino.h"

TFT_eSPI lcd = TFT_eSPI();  

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial
ELM327 myELM327;

void setup() 
{
  lcd.init();
  lcd.fillScreen(BLK);
  lcd.setRotation(1);

  lcd.setTextSize(3);

  Serial.begin(115200);
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect("OBDLink LX"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
 lcd.setCursor(0, 0, 1);
    lcd.print("OBD voltage: ");
    lcd.print(myELM327.payload);
}

PowerBroker2, 2020-06-02:
That sketch doesn't send any manual AT commands. Please update your comment with the correct sketch.

lysiong, 2020-06-02:
> 
> 
> That sketch doesn't send any manual AT commands. Please update your comment with the correct sketch.

@PowerBroker2 sorry, I forgot to add in the below line. Please find the update comment above.
myELM327.sendCommand(READ_VOLTAGE);

PowerBroker2, 2020-06-02:
Try `lcd.write(myELM327.payload, myELM327.recBytes)`

lysiong, 2020-06-03:
Will try tonight & let you know the results

lysiong, 2020-06-03:
using both command below also come out the same error
lcd.write(myELM327.payload, myELM327.recBytes)
lcd.print(myELM327.payload, myELM327.recBytes)

no matching function for call to 'TFT_eSPI::write(char*&, uint16_t&)'

I will try to figure out as your library is actually working.

PowerBroker2, 2020-06-03:
Ok, sounds good. You'll probably find more help if you post on the Arduino forum.