PowerBroker2/ELMduino #31: How can I use ELMduino over Uart?

beebrain, 2020-06-02:
I have ELM32 but the Bluetooth module on ELM is broken.
I extracted the RX and TX line and connected them to my laptop. It's work.
I would like to know. How to connect them over arduino with your library?

PowerBroker2, 2020-06-02:
Using hardware or software serial, connect RX to TX both ways between the Arduino and ELM327. Then use the appropriate library example.

closed 2020-06-02 by PowerBroker2

nices0325, 2020-12-16:
would you let me know the example link??? I have same situation with 'beebrain' plz...

beebrain, 2020-12-17:
> would you let me know the example link??? I have same situation with 'beebrain' plz...

For my situation, I have changed the baud rate to 34800. Try to change the baud rate.