PowerBroker2/ELMduino #32: Hi,can you tell me BluetoothSerial.h source，thank you

zhangxueh, 2020-06-03:
Hi, Thank you for your code, but I have a problem. I look at your code in "examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino", it have a  "BluetoothSerial.h" ,I can't find the source on the Internet. Can you tell me its link and it's library? or can you provide the code? Thank you.

PowerBroker2, 2020-06-03:
If you're compiling for an ESP32, you already have the library as part of the NodeMCU core. If you're not compiling for an ESP32, the library will not work. [This, however, is the link to the BluetoothSerial.h file on GitHub](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/src/BluetoothSerial.h)

closed 2020-06-03 by PowerBroker2

zhangxueh, 2020-06-04:
OK, thank you