PowerBroker2/ELMduino #33: Esp32 connection

philo-create, 2020-06-11:
Hello,Firstly i would  like to thank you for all the  hard  work you have  put  into  making  this  library.unlike  the previous  esp32  issue #11. I havent  been  able  to  get a bluetooth connnection between   the esp32 and obd2 scanner .while  running  the [ ESP32_Bluetooth_Serial.ino code](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino)  provided in the  example, i get  the following  output "Couldn't connect to OBD scanner - Phase 2".In the  previous  issuses  you  recommended  that  people olso  try out the esp32  test.ino example.i have  tested  that example  and  realised  that i get  a "Connected to ELM327" output  even when  the  is  no  obd2  scanner  present.

I am using  the LILYGO® TTGO T-Call V1.3 ESP32 Wireless Module GPRS Antenna SIM Card SIM800L Board and a  standard  elm32 bluetooth obd2 scanner

[image]

PowerBroker2, 2020-06-11:
Try the [ESP32_test.ino](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino) sketch again, but with the ELM327 on and present. Also ensure that no other bluetooth devices are paired with the ELM327 and that your car is on and running (with the ELM327 plugged into the car of course). If you get `Connected to ELM327`, type in the following commands into the serial montior (carriage return line ending selected):

```
AT Z
AT E0
AT S0
AT SP0
010C
```

Let me know what the responses for each of these commands are.

philo-create, 2020-06-11:
Hello again.I just  followed  the  instructions  you  gave  me  and  i   didn't get  any  response back.so basically  nothing  happens  and  after  a while the board  reboots. below  is  a picture of  what  happens when  i eneter  the  commands one after another
[image]

[image]

I just  also want  to let  you  know  that  i have  tested the  obd2  scanner  with  the  torque android  app and  it works  fine .

PowerBroker2, 2020-06-11:
2 things:

1. What is the name of the ELM327 device when you connect to it on your Android? Is it's name "OBDII"? If not, what is it?
2. Are you turning off your phone's bluetooth when attempting to connect to the ELM327 with your ESP32?

philo-create, 2020-06-12:
The name  of the  ELM327 device  that i am using  for  testing  is  called  (OBDll) .I have  onother  ELM327 device  called  "OBDll bluetooth",for  simplicity i will refer to this  device as  NO2.

Is setting  up device  NO2 as  easy as changing  the ` if (!ELM_PORT.connect("OBDII"))` line of code  or do i have to dig deeper into the library .

PowerBroker2, 2020-06-12:
Yes, you can simply change that line and it should work fine for the other module

philo-create, 2020-06-12:
Thank you, device No2 works .it  would  have  taken me  forever  to realise  that changing  the  board  rate  would  fix  it..still dont  understand why  the  other  device doesnot  work.
 I might  be  asking  for too much in the  next  question so i  understand  if you  choose  not  to answer.
from  my understanding after  going  through  your  library I kinda  assume that  if  i want  for  example  speed in kph i would  write  the  code  like this
 ` uint32_t kph = 0;
 
  float speed = myELM327.kph();

  if (myELM327.status == ELM_SUCCESS)
  {
    kph= (uint32_t)speed;
    Serial.print("speed: "); Serial.println(kph);
  }
  else
    printError();
}
`
How  do  i get  fuel level ? so bassically want  to know  how to make use of the List of standard PIDs:?

philo-create, 2020-06-12:
Thank you, device No2 works .it would have taken me forever to realise that changing the board rate would fix it..still dont understand why the other device doesnot work.
I might be asking for too much in the next question so i understand if you choose not to answer.
from my understanding after going through your library I kinda assume that if i want for example speed in kph i would write the code like this
```

 uint32_t kph = 0;

float speed = myELM327.kph();
{
if (myELM327.status == ELM_SUCCESS)
{
kph= (uint32_t)speed;
Serial.print("speed: "); Serial.println(kph);
}
else
printError();
}
```

How do i get fuel level ? so bassically want to know how to make use of the List of standard PIDs:?

PowerBroker2, 2020-06-12:
Yes, except I would make kph a float.

In order to use the PIDs in the list, you can do something like this:

```C++
// https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01
float fuelLevel = 0; // in %
if (myELM327.queryPID(SERVICE_01, FUEL_TANK_LEVEL_INPUT))
	fuelLevel = myELM327.findResponse() * 100.0 / 255.0;
```

More info [here](https://github.com/PowerBroker2/ELMduino/issues/4).

closed 2020-06-12 by PowerBroker2

philo-create, 2020-06-12:
Thank you. I will test everything and keep you informed

patfelst, 2020-06-12:
> The name of the ELM327 device that i am using for testing is called (OBDll) .I have onother ELM327 device called "OBDll bluetooth",for simplicity i will refer to this device as NO2.
>
> Is setting up [...]

I know you're not trying to optimise here but simply get your code working, but if you look at the Arduino bluetooth library, you'll see that using the OBD scanner's MAC address is much faster than using its name (10 seconds vs up to 30 seconds). Have a look [this comment](https://github.com/espressif/arduino-esp32/blob/b92c58d74b151c7a3b56db4e78f2d3c90c16446f/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino#L31). I'm using the "address" method and it works very reliably.

Also I would recommend finding out why your program was crashing, there is probably a very simple explanation, probably an issue with a string variable or a timeout not long enough. You can use [this tool](https://github.com/me-no-dev/EspExceptionDecoder) which is very easy to use within the Arduino IDE. It will show you the exact line of source code where the program is crashing.

philo-create, 2020-06-15:
@PowerBroker2. I have tested the following  code and i have only been able to get 0 as an output
```
// https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01
float fuelLevel = 0; // in %
if (myELM327.queryPID(SERVICE_01, FUEL_TANK_LEVEL_INPUT))
	fuelLevel = myELM327.findResponse() * 100.0 / 255.0;
```

PowerBroker2, 2020-06-15:
I can't help unless you provide as much detail as possible on the context of the problem since I don't know what steps you've taken, what your physical setup is, or what your entire code looks like. Please post your entire sketch.

Also, have you verified the connection does in fact work for the example sketches? What were the responses to the manual AT commands as recommended in earlier posts?

philo-create, 2020-06-17:
I tested the example schetch and it does work .The  responses to the manual AT commands are.
```
AT Z =ELM327 V1.5
AT E0 =AT EOOK
AT S0 =OK
AT SP0 =OK
010C = 410COBDA
```
Below is the entire code i am using
```

#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

//uint32_t rpm = 0;
//const uint8_t FUEL_TANK_LEVEL_INPUT  = 47;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(38400);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  Serial.println("Attempting to connect to ELM327...");
  if (!ELM_PORT.connect("OBDll bluetooth"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  
float fuelLevel = 0; // in %

if(myELM327.queryPID(SERVICE_01, FUEL_TANK_LEVEL_INPUT)){
  fuelLevel = myELM327.findResponse() * 100.0 / 255.0;
  Serial.print("FuelLevel: "); Serial.println(fuelLevel);  
  
}
else
    printError();

}

void printError()
{
  Serial.print("Received: ");
  for (byte i = 0; i < myELM327.recBytes; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();
  
  if (myELM327.status == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.status == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.status == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.status == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(100);
}

```

PowerBroker2, 2020-06-17:
Looks like your ELM327 is connected and responding correctly. I'm curious, if you run the test sketch again and enter `012F`, what do you get as a response?

philo-create, 2020-06-17:
I tested it and i got `NoData`  as   a response

PowerBroker2, 2020-06-17:
In that case, you might need to use a different PID. Check the header file/wiki page and find one that works for fuel level.

patfelst, 2020-06-17:
Hey @PowerBroker2 I wrote a bit of code to query available PIDs for your vehicle, and a google spreadsheet to decode it. Would you be interested in me sharing it?

PowerBroker2, 2020-06-18:
A separate issue for this would be better, but yes, I'm interested

maksumon, 2020-07-13:
Hello, @PowerBroker2 Thanks for the excellent library. But unfortunately, I could not manage to use it yet as it is failing to connect on phase 2. Then I tried the [ESP32_test.ino](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino), it connects but no response from the AT commands you have mentioned above.

[image]

This is the device I am using and it works fine with Torque and any other OBDII application. I have also made sure that no other Bluetooth device was connected to the OBDII device.

Do you know what can be the issue or should I buy and try another OBDII device?

PowerBroker2, 2020-07-13:
Sometimes there are issues with baud rates. Try using 38400 baud instead of 115200. If you still have trouble, try all other possible bauds. I've seen one case where the ELM327 only worked at 9600.

maksumon, 2020-07-13:
Ow, okay, I have tried 38400 and 115200 but no luck... Tomorrow I will try all other baud rates and let you know. Thanks for the prompt reply.

maksumon, 2020-07-14:
@PowerBroker2 I have tried all the baud rates from 9600 to 115200 with 2 ESP32 boards but no luck :(. I don't have any spare ELM327 device to try out.

PowerBroker2, 2020-07-14:
If it doesn't respond to manual AT commands, it probably means the ELM327 is automatically connecting to a previously paired device (i.e. whatever device you had running Torque). Make sure to do testing with NO other bluetooth devices enabled other than the ELM327 and your Arduino.

Also, someone with a similar problem was able to resolve it by [following these steps](https://github.com/PowerBroker2/ELMduino/issues/12#issuecomment-609814007).

maksumon, 2020-07-15:
But no other devices are connected at the moment I was testing... I even tried to disconnect the EMP327 form the OBD port and reconnect.

I have tried with 3 different ESP32 but all cannot connect.

I will try following the steps mentioned and let you know.

Thanks

maksumon, 2020-07-15:
@PowerBroker2 I tried connecting my laptop with the OBDII and it was successful. I sent the command AT Z and it printed ELM327 V2.1, was that something to do with the version of ELM327 to connect with ESP32?

PowerBroker2, 2020-07-15:
If no other bluetooth devices are enabled during testing, the car is running, and you're using the [EP32_test.ino sketch](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino) and still do not get responses from the ELM327 at any baud, it might mean that the ESP32 isn't paired to the ELM327 in the first place.

This may be due to your ELM327's bluetooth name not being "OBDII". When you open your laptop and choose your ELM327 as a bluetooth device, what is the device name? Is it "OBDII"? If not, you will have to change it in the test sketch. NOTE: you can replace the device name in the test sketch with the ELM327's MAC address instead.

maksumon, 2020-07-15:
@PowerBroker2 The EML327 device name is `OBDII` when I connect from phone and laptop

PowerBroker2, 2020-07-15:
If the test sketch doesn't work, all I can say is the problem doesn't lie with the library (the test sketch doesn't use the library at all) - the problem is elsewhere. Either it's your ELM327 or your ESP32...

maksumon, 2020-07-15:
Tomorrow I will give it a try with Arduino and HC-06 and will let you know

maksumon, 2020-07-16:
@PowerBroker2 I have not tried with Arduino yet but wanted to make sure, do I need to run AT commands to make HC-06 as master and bind with the ELM327?

PowerBroker2, 2020-07-16:
I'm not sure if it's possible to connect to the ELM327 with an HC-06 - I think you need an HC-05, since an HC-06 can't be used as a master. If you use an HC-05, you can follow the setup directions [here](https://forum.arduino.cc/index.php?topic=600014.msg4159554#msg4159554) to connect to your ELM327.

maksumon, 2020-07-17:
@PowerBroker2 Thanks for the info brother. I will give it a try at the weekend and let you know.

kalinkaata, 2024-10-11:
> Sometimes there are issues with baud rates. Try using 38400 baud instead of 115200. If you still have trouble, try all other possible bauds. I've seen one case where the ELM327 only worked at 9600.

Sorry, I'm late in the party having these issues with ELM327...How do you exactly set the baudrate? All I'm seeing in these examples is setting the baudrate for the debug serial port, but not for the ELM serial port....Do these ports they speed settings?