PowerBroker2/ELMduino #34: Supported PIDs decoder

patfelst, 2020-06-18:
In Issue https://github.com/PowerBroker2/ELMduino/issues/33 somebody was wanting to use a PID which happened to be unsupported by their vehicle. To get a list of PIDs supported by your vehicle, run this code. I have this function in the main loop().
```c++
/*
-----------------
  Detect if user has pressed 'd' key on serial monitor keyboard, if yes
  End the bluetooth connection and halt the processor
  If 'P' is pressed, outputs the PIDs that the vehicle supports.
-----------------
*/
void quit_prog_if_keypressed() {
  if (Serial.available()) {
    char c = Serial.read();
    if (c == 'd') {
      Serial.println("Disconnecting bluetooth connection");
      SerialBT.disconnect();
      //Serial.println("Closing bluetooth connection");
      //SerialBT.end();
      Serial.println("Halting program ... ");

      #ifdef lcd_20x4_line_char
      lcd.clear();
      lcd.setCursor(0,0);
      //         01234567890123456789
      lcd.print("Disconnect Bluetooth");
      lcd.setCursor(0,1);
      //         01234567890123456789
      lcd.print("Halting program ...");
      #endif

      while(1);
    }
    else if (c == 'P') {
      char cmd[100] = {0};
      // Print supported PIDS
      Serial.println("\n\nPrint supported PIDs *** Turn engine ON ***");
      Serial.println("Turning ECU filter off ATCRA");
      boost.send_command("ATCRA");
      Serial.println("Turning headers ON ATH1");
      boost.send_command("ATH1");
      Serial.println("Turning line feeds ON ATL1");
      boost.send_command("ATL1");
      Serial.println("Turning spaces ON ATS1\n");
      boost.send_command("ATS1");
      
      Serial.println("=================================");
      for (int i=0;i<6;i++) {
        uint8_t pid = i * 32;
        Serial.printf("Supported PIDs: 01%02X\n", pid);
        sprintf(cmd, "01%02X", pid);
        boost.send_command(cmd);
        Serial.printf("%s", boost.OBD_response);
        delay(100);
      }
      Serial.println("=================================\n");
      Serial.println("Disconnecting bluetooth connection");
      SerialBT.disconnect();
      Serial.println("Halting program ... ");
      while(1);
    }
  }
}
```
Note that the command "ATCRA" turns off the filter which prevents responses from the second ECU (with response address 7E9) in my car. If this is left on, you get duplicated responses to some PIDs (as [your code deals with](https://github.com/PowerBroker2/ELMduino/blob/d6a693685d474ff509341ca83a0a0a34c285265b/src/ELMduino.cpp#L634)).

FYI to apply the filter to prevent responses from the 7E9 ECU, issue this command during initialize:
`send_command("ATCRA7E8"); // Set CAN hardware filter to receive from address 7E8 only`

Sample output from a 2019 BMW is:
```
Supported PIDs: 0100
7E8 06 41 00 BF BF A8 93 
7E9 06 41 00 98 18 80 01 

Supported PIDs: 0120
7E8 06 41 20 A0 07 B1 19 
7E9 06 41 20 80 00 00 01 

Supported PIDs: 0140
7E8 06 41 40 FE D0 85 11 
7E9 06 41 40 C0 00 00 00 

Supported PIDs: 0160
7E8 06 41 60 64 00 00 01 

Supported PIDs: 0180
7E8 06 41 80 00 04 00 0D 

Supported PIDs: 01A0
7E8 06 41 A0 04 00 00 00 
```

And then paste the above output into this google spreadsheet I put together. It's not that pretty in terms of user input....but anyway, put it in cell J16 (the black cells). Then edit cell B15 to point to the group of 32 PIDs you're interested in. Or alternatively, just paste a single line response into Cell J16.

https://docs.google.com/spreadsheets/d/1R2C8_kY6fuFpd4JJyIrWOXqAQGgxm2MHYHXRt3qiKCc/edit?usp=sharing

Where column E says "Yes" it means that PID is supported by the vehicle.

philo-create, 2020-06-18:
Thanks for the code @patfelst ..how did  you  declare  `boost`

patfelst, 2020-06-18:
I probably should have mentioned I created a new class called `BOOST` which is adapted from the ELM327 class PowerBroker2 created, it contains a whole stack of extra variables and methods for my boost gauge project, and I restructured some things like the command and response buffers. Other than that, it's very similar.

For the above code snippet, just replace `boost.OBD_response` with `ELM327.payload`.

patfelst, 2020-06-18:
Hi again, I've some requests to give write access to the google spreadsheet. Sorry but I'm not giving public write access to it for obvious reasons, but you should be able to make a copy of it under menu "File/Make a copy" or download as an Excel file.

Yusupoff, 2020-09-28:
Hello! I am writing through a translator.
I have a Russian car and it has its own PIDs
Sending a command: myELM327.queryPID("220005")
I get this answer:
```
>053 
0: 62 00 05 58 54 41 
1: 32 31 39 30 31 30 46 
2: 30 33 32 31 37 37 35 
3: 31 35 2D 31 32 2D 32 
4: 30 31 34 30 33 31 35 
5: 33 37 30 36 32 39 34 
6: 32 34 31 EE E5 17 00 
7: 31 B4 13 00 6B B4 03 
8: 00 98 00 05 00 F2 3A 
9: D9 38 03 00 0B 00 0D 
A: 00 00 00 0B 00 03 00 
B: 00 00 47 12 C3 49 00 

```
Need to process this line:
`6: 32 34 31 EE E5 17 00 `

Initialization commands:
AT@1
ATZ
ATE0
ATSP6
ATAL
ATD0
ATST32
ATSH7E0
ATCRA7E8
ATSW00
ATCAF1

I managed to fix the initialization. But the data cannot be processed

philo-create, 2020-11-06:
Hello, I am trying to read obd data from the  Toyota Quantum 2.7 GL 14 seater bus. But  I haven't been successful. Bellow is the results that I get when  I run the code above.

`Supported PIDs: 0120
7E8037F01
Supported PIDs: 0140
7E8037F0111
Supported PIDs: 0160
7E8037F0111
Supported PIDs: 0180
7E8037F0111
Supported PIDs: 01A0 `

I received  the following output when I requested  0100

`010001007f0111`

Is there a  way to get the obd readings?

closed 2021-04-06 by PowerBroker2