PowerBroker2/ELMduino #36: ELM_NO_DATA using the Wifi sketch.

jokke009, 2020-06-23:
**Describe the bug**
Using the wifi sketch in the example, I've attempted to connect an esp8266 (which I also read about in the closed issues). The ESP connects fine and the connect to server proceeds. I'm then trying some simple as a RPM read, however this always produces an ELM_NO _DATA. I thought of doing some debugging work in with a ,net program on a laptop over the wifi direct, but I though I'd ask first to see if anybody could point out the obvious.

**To Reproduce**
Using a Iegeek OBD2 wifi adapter and the simple wifi sketch provided running on an esp8266.
I haven't hooked up a serial monitor laptop, but rather used a simple ws2812 led matrix which lights different leds according to the error. 

**Expected behavior**
Getting an ELM_SUCCESS: 
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }

**Equipment**
Iegeek wifi:
https://www.bol.com/nl/p/obd2-wifi-interface-adapter-mini-elm327/9200000055136781/?country=BE&Referrer=ADVNLGOO002060-G-88680672361-S-795237665226-9200000055136781&gclid=Cj0KCQjw0Mb3BRCaARIsAPSNGpU3xTYW7FFMST7e_kB1dhrsSr1uZfIS692eeg2aOtnFi13dVnzg0-AaAkMBEALw_wcB

Nodemcu

**Code**
`void setup_wifi()
{

    WiFi.disconnect();
    WiFi.config(ip, gate, sub); 
    WiFi.mode(WIFI_STA);
    WiFi.begin(AP_SSID);

    while (WiFi.status() != WL_CONNECTED)
    {
        FlashYellow();
        delay(100);
        FlashRed();
        delay(400);
    }

    FlashYellow();
    delay(1000);
    wifiClient.connect(server, 35000);
    while (!wifiClient.connected())
    {
        FlashGreen();
        delay(100);
        FlashYellow();
        delay(400);
    }
    FlashGreen();
    myELM327.begin(wifiClient);
}

void setup()
{

    FastLED.addLeds<NEOPIXEL, NEOPIN>(matrixleds, NUMMATRIX);
    matrix->begin();
    matrix->setTextWrap(false);
    matrix->setBrightness(40);
    matrix->setTextColor(colors[0]);
    FlashRed();
#ifdef CURRENT
    IPAddress ip1(192, 168, 1, 1);
    pzem.setAddress(ip1);

#else

#endif

    //uartP.CmdReceivedPtr = DataReceived;
    delay(10);

    setup_wifi();
#ifdef OTA_MODE
    setup_ota();
#endif
}

int x = 0;
int pass = 0;
void printError()
{

     if (myELM327.status == ELM_NO_RESPONSE)
        FlashWhite(0);
        //Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
    else if (myELM327.status == ELM_BUFFER_OVERFLOW)
        FlashWhite(1);
        //Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
    else if (myELM327.status == ELM_GARBAGE)
        FlashWhite(2);
        //Serial.println(F("\tERROR: ELM_GARBAGE"));
    else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
        FlashWhite(3);
        //Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
    else if (myELM327.status == ELM_NO_DATA)
        FlashWhite(4);
        //Serial.println(F("\tERROR: ELM_NO_DATA"));
    else if (myELM327.status == ELM_STOPPED)
        FlashWhite(5);
        //Serial.println(F("\tERROR: ELM_STOPPED"));
    else if (myELM327.status == ELM_TIMEOUT)
        FlashWhite(6);
        //Serial.println(F("\tERROR: ELM_TIMEOUT"));
    else if (myELM327.status == ELM_GENERAL_ERROR)
        FlashWhite(7);
        //Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));
    delay(100);
}
void loop()
{
#ifdef OTA_MODE
    ArduinoOTA.handle();
#endif

    unsigned long currentMillis = millis();

    // if we lose wifi
    if (WiFi.status() != WL_CONNECTED)
    {
        setup_wifi();
    }

    // get rpm and update matrix
    if (currentMillis - previousMillisHeart > heartBeat)  //heartbeat = 1000ms
    {
        float tempRPM = myELM327.rpm();

        if (myELM327.status == ELM_SUCCESS)
        {
            rpm = (uint32_t)tempRPM;
            values[1] = (int)(rpm);
            values[2] = (int)(rpm / (float)2);
            values[3] = (int)(rpm / (float)4);
            values[4] = (int)(rpm / (float)8);
            values[5] = (int)(rpm / (float)16);
            values[6] = (int)(rpm / (float)32);
            values[7] = (int)(rpm / (float)64);
            UpdateBars();
        }
        else
        {
            printError();
        }
        previousMillisHeart = currentMillis;
    }

}`

**Wiring**
Using wifi.

Thanks in advance.

PowerBroker2, 2020-06-23:
Your sketch looks good to me. For debugging purposes, you can use the following to manually figure out what bytes have been received by the ESP32:

```C++
for (int i=0; i<myELM327.recBytes; i++)
  Serial.println(myELM327.payload[i]);
```

jokke009, 2020-06-28:
Hi ,
Sorry haven't been able to test it yet . Tuesday afternoon I'll be able to
my laptop in a car and test it. I've even got another car for testing and
comparing.

I'll keep you updated in any case .

jokke009, 2020-06-30:
Hi, 
Been able to test that: but no data in the payload. 
I've tried the esp8266_wifi example directly, but get the following: 
Connecting to WiFi_OBDII
..
Connected to Wifi
IP address:      
192.168.0.11     
connected
Received: 
        ERROR: ELM_TIMEOUT
Received: 
        ERROR: ELM_TIMEOUT
Received: 
        ERROR: ELM_TIMEOUT
Received: 
        ERROR: ELM_TIMEOUT
...

Anything in the library I need to fool around with settings, timeouts or such? 

Once again any help appreciated.

PowerBroker2, 2020-07-03:
1. Can you verify the IP address you're using in the sketch is correct?
2. Use [this sketch](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP8266_test/ESP8266_test.ino) to test your connectivity manually by inputting the following commands:
```
AT Z
AT E0
AT S0
AT SP0
010C
```
Report back on what the responses are for each of the above commands.

jokke009, 2020-07-03:
`
Connected to Wifi
IP address:
192.168.0.11
connected
Connected to ELM327
Ensure your serial monitor line ending is set to 'Carriage Return'
Type and send commands/queries to your ELM327 through the serial monitor

AT Z
?
>ATZ
ELM327 v1.5
>AATTEE00
OK
>ATS0
OK
>ATSP0
OK
>010C
STOPPEDNG...
>
`

PowerBroker2, 2020-07-03:
Looks like your ELM327 is at least responding to AT commands.

I do see that you're test sketch was able to get a connection using 192.168.0.11, but your sketch uses 192.168.1.1 - that might be the issue.

jokke009, 2020-07-04:
Hi, 
Yeah the AT commands seem to work using the sketch you provided last.
Note: Ignore the ip1 in the first sketch I posted.. it isn't used. It gets .11 from the wifi dongle. 

Any explanation about the 010C , command? 
For a brief period of time I'll get 'SEARCHING' then 'STOPPEDNG..' not quite sure what all that means. 

Any other useful AT commands I can go through for testing? 

In the mean time I'll try some different protocols than '0' in the constructor. See if that yields and change.

PowerBroker2, 2020-07-04:
`010C` is the query for RPM data. If you don't get a valid response it either means you car isn't running or the ELM327 is bad - or worse.

What car make/model/year are you testing on?

Also, can you ensure you're connecting to the ELM327 in your live sketch exactly the same as you did in the test sketch?

jokke009, 2020-07-05:
Hi, 
So far been only testing on a ford focus st (2019). I've got a mazda mx5 2016 I'll try later this week.

I always turn on the engine. The elm327 wifi dongle shouldn't be the problem. I tried the torque app on android and its gets all the information. 
 
I'm using your Esp8266_wifi sketch, in the hope of just getting the RPM for now. 

A bit off topic:: something I noticed with a obd2-HUD is that the mx5 shows data at a much higher refresh rate than the focus.

jokke009, 2020-07-29:
Hi, 

No RPM data received yet. 
My plan is to wireshark it. Get a program on the laptop to send/receive data to debug it. Hopefully there's a .net library out there.

closed 2020-07-29 by PowerBroker2