PowerBroker2/ELMduino #37: ELM32_test.ino ELM327 ignoring ESP32

supaplextor, 2020-07-03:
**Describe the bug**
ELM32_test.ino vs Android apps (sketch sends nothing to bluetooth peer)

**To Reproduce**
Using a generic esp32 dev board. 
This works in Android:

[image]

[image]

[image]

[image]

As you see, sending atrv works in Android (BMW 2004 330ci). I was hoping for the same behavior, send atz, atrv etc get a reply in the serial monitor.

supaplextor, 2020-07-03:
similar sketch, syslog snippet, and serial monitor screenshot: https://github.com/supaplextor/arduino-esp-sketches/tree/master/esp32/SerialBT-to-Serial-bridge

<pre>Jul  3 10:47:13 esp32 ttyUSB0-io foo = DEBUG_PORT.available(); = (4)
Jul  3 10:47:13 esp32 ttyUSB0-io char c = DEBUG_PORT.read(); = (a)
Jul  3 10:47:13 esp32 bt-serial-io int b = ELM_PORT.write(a) = (0) !!!
</pre>

write("a") is returning 0 bytes written.

PowerBroker2, 2020-07-03:
With the car up and running plus using [this sketch](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP8266_test/ESP8266_test.ino), enter the following commands and see what the responses are:
```
AT Z
AT E0
AT S0
AT SP0
010C
```

supaplextor, 2020-07-04:
Regarding https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino line 42, I'll reuse this sketch and check for bytes written. I'm holding off including this lib until the above AT commands respond over BT.

My OBD2 units do not do WiFi. Only Bluetooth. The above sketch is for TCP/IP.  Please keep this open until I get my hands on a WiFi unit.  I hope it's WPA2 or better that allow reconfiguration of ap mode,ssid,phrase etc.

I do have two esp32 and 5+ esp8266 for testing misc sketches (just dev boards w/o external hardware). I looked around, however an esp32 advertising a serial service is more involved too. :-/

On a sad note / tangent, my fuel pump is on the way out; "eyeroll" at least I know why my RPM acceleration "dips" under mild to moderate acceleration. (chasing this ghost in my car for the last few months - was ready to replace fuel filter any day now then this lol). Yesterday (manual transmission) it kept acting like it was dropping to a limp mode (1-10 seconds of poor fuel pressure etc), made it home and left it to idle, ran inside while my friend sat in the car, came outside and it was dead. It would not restart. A few hours later I tried again and it fired up fine.

supaplextor, 2020-07-04:
Using https://github.com/supaplextor/arduino-esp-sketches/tree/master/esp32/ESP32_test3 has the same write issue. Serial output log included there.

PowerBroker2, 2020-07-05:
I have no clue what you're trying to do or what your question/problem is...

supaplextor, 2020-07-05:
In light of the following within ELM32_test.ino,
<pre>#include "BluetoothSerial.h"
BluetoothSerial SerialBT;
// ...
SerialBT.write(c);</pre>

I created https://github.com/espressif/arduino-esp32/issues/4135 instead.

closed 2020-07-05 by supaplextor