PowerBroker2/ELMduino #38: more on custom PIDs - splitting response

miguelos6, 2020-07-07:
This is continuation of [https://github.com/PowerBroker2/ELMduino/issues/4](url)
some custom PIDs return values and suggested calculation for result is divided in two parts A and B  (all examples for Torque app)
see example of calculations suggested here -> [https://www.opelastraclub.com/t27293p300-pids-para-torque#406215](url)
if A*256+B makes all sense as it's simply conversion of longer response to decimal (I guess)
such a calculation :
(A*33.55/255+B*0.13/255)-12
is impossible to get with current findResponse()

Would it be possible to return A and B separately so calculation could be done on them ?

PowerBroker2, 2020-07-07:
While bit masking/shifting the output of `findResponse()` would relatively easily provide the A and B values you're looking for, I can add 2 class member variables `a` and `b` to make things even easier.

PowerBroker2, 2020-07-08:
I opted not to use class variables `a` and `b`, but rather broke out the 64-bit response into 8 individual bytes that you can then use as your application sees fit.

Changes implemented in [2.3.0](https://github.com/PowerBroker2/ELMduino/releases/tag/2.3.0).

closed 2020-07-08 by PowerBroker2

miguelos6, 2020-07-09:
Hi @PowerBroker2, sorry for re-opening this.
Could you please give an example how to apply such calculations on A & B returned separately ?
Will old sketches work with this change ?

PowerBroker2, 2020-07-09:
No worries!

Some standard PIDs return a single byte - in this case, the LSB of the response is the `A` value and `B` is undefined. Other standard PIDs return 2 bytes - in this case, the 2nd LSB is the `A` value and the LSB is the `B` value. Custom PIDs sometimes return even more than 2 bytes of data.

Because there is no set byte for the `A` value and `B` value, I can't easily/automatically interpret which byte in the response is `A` or `B`. This is why I didn't implment class variables `a` and `b`. Instead I implemented the following:
```c++
byte responseByte_0;
byte responseByte_1;
byte responseByte_2;
byte responseByte_3;
byte responseByte_4;
byte responseByte_5;
byte responseByte_6;
byte responseByte_7;
```
Where `responseByte_0` is the response LSB, `responseByte_1` is the 2nd LSB, ..., and `responseByte_7` is the response MSB.

For your application, if you get a valid response from `getResponse()`, you can then use `responseByte_1` and `responseByte_0` as your `A` and `B` values, respectively.

Does that help?

miguelos6, 2020-07-12:
Thank you for reply, but again I'm sorry for not understanding (I'm not a programmer, just using the lib to do small work for me)
I guess I need to change my code (pre 2.3.0) in order to know how to interpret returned values? ie. - now more significant bytes may contain garbage data & simply doing sth like this :
int16_t Tpid1a6d = myELM327.findResponse() 
will give unpredicted results ?
 
So far I used only A and A+B values. Only recently found a value that need to be split into separate calculations on A and B

Maybe it's an option to zero out more significant bytes and just filling A or A+B on query ?
Could you please provide an option on how to read A, B (or more) separately in sketch after findresponse? (as mentioned - I'm not a programmer:/)

PowerBroker2, 2020-07-12:
Any non-valid bytes are already zeroed out, so it's fully backwards compatible with previous sketches. The difference is now you have more flexibility when calculating complex PID responses. For your case:

```c++
myELM327.findResponse();
byte B = myELM327.responseByte_0;
byte A = myELM327.responseByte_1;
// do things with A and B
```

cyberelectronics, 2020-07-30:
Hi,
I'm trying to implement some Extended PIDs for a Hyundai Kona EV (the car not yet arrived, so I can't test it right now. I'm more a HW guy, so implementing any SW is not an easy task for me).
Your library works great with my old car and BT adapter (checked the RPM data).

I found some [examples/sniff and description on the  net](https://docs.google.com/spreadsheets/d/1-9jZafV9eZeBUnPQo7qQHbX2-_4qZfWfRVpidoF1owA/edit#gid=1831012958), but it looks like for a query (PID 220100), the response can be 6x 8bytes or even 9x 8bytes long.
[image]

pid 220100 @ 2019-03-23T10:50:56.868873
7BB 10 26 62 01 00 7E 50 07
7BB 21 C8 FF 7C 67 66 0E E5 [?=200] [?=255] [INDOORTEMP=22] [OUTDOORTEMP=11.5] [EVAPORATORTEMP?=11] [?=14] [?=229]
7BB 22 8F FF FF 8F FF 11 FF [?=143] [?=255] [?=255] [?=143] [?=255] [?=17] [?=255]
7BB 23 FF FF FF FF FF FF FF [?=255] [?=255] [?=255] [?=255] [?=255] [?=255] [?=255]
7BB 24 FF 3A D6 8F 90 43 FF [?=255] [?=58] [?=214] [?=143] [?=144] [speed?=41.6407706650093] [?=255]
7BB 25 FF 00 FF FF 00 00 00 [?=255] [?=0] [?=255] [?=255] [?=0] [?=0] [?=0]

Is there a way to parse and check a specific "responseByte_X"?
Can you give me an example code?
Thank you!

PowerBroker2, 2020-07-30:
I'm curious, what is the `7BB` section for? Usually when you query for a PID, the response will begin a slightly modified PID as a sort of "preamble". Is this the case for your PID?

Once you do get the car, use one of the test sketches to manually query the vehicle for this custom PID. Then, **record the exact response** and post it here.

cyberelectronics, 2020-07-31:
I think Header / Module ID.  ([end of this page](https://en.wikipedia.org/wiki/OBD-II_PIDs))
[Here on the first page](https://docs.google.com/spreadsheets/d/1-9jZafV9eZeBUnPQo7qQHbX2-_4qZfWfRVpidoF1owA/edit#gid=660740603) you can find `7BB` (Aircon).
Ok. I will try to record the exact response.

cyberelectronics, 2020-08-05:
Just [found this example](https://www.insideevsforum.com/community/index.php?threads/torque-pro-on-the-kona-overview-and-setup-for-interested-owners.6970/page-7#post-95577) with more info.
So from the "our dump", we can identify:
`7BB _10_ _26_ _62 01 00_ 7E 50 07`
10h = sequence 0
26h= number of bytes (len) = 38 - 3bytes(Mode&PID > 62 01 00)
Aprox. 10ms between sequences

cyberelectronics, 2020-08-26:
Here is the response, using your demo sketch:
1) Sending command 220101:
`Connected to ELM327
220101
SEARCHING...
7F22127F22127F2212
03E
0:620101FFF7E7
1:FF980000000080
2:00000F2B1F1D1D
3:1D1F1D00001BC6
4:03C52800007A00
5:0009FF00000888
6:000003D0000003
7:1D000615A30900
8:030000000003E8`

2) Sending command 220100
`220100
SEARCHING...
7F2212
7F2212
7F2212`

3) Sending command P:
``

PowerBroker2, 2020-08-27:
This is all a little confusing - all I need is the query and the exact response. Most likely, however, you will have to send and parse this particular PID response yourself since the output of the library's parser is 64bit max. You can still find utility in the library for setting up the ELM327 and polling less eccentric PIDs

cyberelectronics, 2020-08-27:
I combined 2 demo sketches, but basically I just sent the "220101" thru Arduino Serial Monitor.
```
void loop()
{
  if(DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();
    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }
 if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();
    if(c == '>')
      DEBUG_PORT.println();
    DEBUG_PORT.write(c);
  }
}
```

and the exact response was this:
```
Connected to ELM327
220101
SEARCHING...
7F2212
7F2212
7F2212
03E
0:620101FFF7E7
1:FF980000000080
2:00000F2B1F1D1D
3:1D1F1D00001BC6
4:03C52800007A00
5:0009FF00000888
6:000003D0000003
7:1D000615A30900
8:030000000003E8
> 
```

Today I will try this:
```
if (myELM327.queryPID(34, 257))   // Kona PID = 22 01 01 = 34, 257
                                                                  {
                                                                    Serial.print("Payload received: ");
                                                                    for (int i=0; i<myELM327.recBytes; i++) Serial.print(myELM327.payload[i]);
                                                                    Serial.println();
                                                                  }
```

cyberelectronics, 2020-08-28:
So here is the results for **queryPID(34, 257)** =>PID 22 0101 at **every 2 seconds**
OBD protocol was modified **from '0' (Auto) to '6'** (otherwise 'Searching...' will appear).
Original **payloadLen = 40** in **elmduino.h** (line 300)
Here the car was turned **ON** 
```
Connected to ELM327
Payload received: 03E0:620101FFF7E77F22127F2212
Payload received: 7F22127F22127F221203E0:620101FFF7E71:FF6E
Payload received: 7F22127F22127F221203E0:620101FFF7E71:FF6E
Payload received: 7F221203E0:620101FFF7E77F22127F22121:FF6E
Payload received: 7F221203E0:620101FFF7E77F22127F22121:FF6E
```
Modified **payloadLen = 200** in **elmduino.h** (line 300)
Now all frames, from **0:** to **8:** appears correctly 
The car was turned **OFF**
```
Connected to ELM327
Payload received: 03E0:620101FFF7E71:FF6E00000000802:00000E3B1C191B3:19191B00001CB94:46B92D00007C005:000AF100000AFF6:000004270000047:040007684400008:000000000003E8
Payload received: 03E0:620101FFF7E71:FF6E00000000802:00000E3B1C191B3:19191B00001CB94:0DB92800007C005:000AF100000AFF6:000004270000047:040007684600008:000000000003E8

```
From here the car was turned **ON** ( **7F2212** was sent 3 times for each query, I don't know what is this)

```
Payload received: 7F22127F22127F221203E0:620101FFF7E71:FF6E00000000832:00010E3B1B191B3:19191B00001CB94:49B92800007A005:000AF100000AFF6:000004270000047:040007685F0D018:6B0000000003E8
Payload received: 7F22127F22127F221203E0:620101FFF7E71:FF6E00000000832:000C0E3A1B191B3:19191B00001CB94:0FB928000080005:000AF100000AFF6:000004270000047:04000768610D018:6B0000000003E8
```

Strange: sometimes **7F2212** appears later in the packet, between frame **0:** and frame **1:**
```
Payload received: 7F22127F221203E0:620101FFF7E77F22121:FF6E00000000832:000C0E3A1B191B3:19191B00001CB94:2CB92D000092005:000AF100000AFF6:000004270000047:04000768750D018:6B0000000003E8
Payload received: 7F221203E0:620101FFF7E77F22127F22121:FF6E00000000832:000F0E3A1B191B3:19191B00001CB94:01B928000084005:000AF100000AFF6:000004270000047:04000768630D018:6B0000000003E8
```

herode10, 2020-12-06:
> I combined 2 demo sketches, but basically I just sent the "220101" thru Arduino Serial Monitor (using this [OBD adapter](https://www.amazon.com/vgate-BLE-iPhone-iPad-Android/dp/B071D8SYXN) + this [ESP32 TTGO](https://www.amazon.com/LILYGO-T-Display-Bluetooth-Development-Arduino/dp/B07V7FDHWR) board).
> 
> ```
> void loop()
> {
>   if(DEBUG_PORT.available())
>   {
>     char c = DEBUG_PORT.read();
>     DEBUG_PORT.write(c);
>     ELM_PORT.write(c);
>   }
>  if(ELM_PORT.available())
>   {
>     char c = ELM_PORT.read();
>     if(c == '>')
>       DEBUG_PORT.println();
>     DEBUG_PORT.write(c);
>   }
> }
> ```
> 
> and the exact response was this (where **03E** = 62 bytes length and **620101** = 40h + PID):
> 
> ```
> Connected to ELM327
> 220101
> SEARCHING...
> 7F2212
> 7F2212
> 7F2212
> 03E
> 0:620101FFF7E7
> 1:FF980000000080
> 2:00000F2B1F1D1D
> 3:1D1F1D00001BC6
> 4:03C52800007A00
> 5:0009FF00000888
> 6:000003D0000003
> 7:1D000615A30900
> 8:030000000003E8
> > 
> ```
> 
> Today I will try this:
> 
> ```
> if (myELM327.queryPID(34, 257))   // Kona PID = 22 01 01 = 34, 257
>               {
>                  Serial.print("Payload received: ");
>                  for (int i=0; i<myELM327.recBytes; i++) Serial.print(myELM327.payload[i]);
>                  Serial.println();
>               }
> ```

Is this a snipet of your code?

Should I had:
myELM327.sendCommand("AT SH 7E4");

To set the Header first?

Thanks

cyberelectronics, 2020-12-07:
@herode10  It's from the demo code, but yes you should set the Header first, using this command, otherwise more ECUs will respond with the same message.
Just finished my project ( Konassist v1.0 ) so you can check the full [code/description here](http://www.cyberelectronics.org/?p=981)

herode10, 2020-12-07:
> @herode10 It's from the demo code, but yes you should set the Header first, using this command, otherwise more ECUs will respond with the same message.
> Just finished my project ( Konassist v1.0 ) so you can check the full [code/description here](http://www.cyberelectronics.org/?p=981) also

Wow! Impressive work! Thanks for the response.

nices0325, 2021-01-12:
actually I'm not a progammer too. Can you give me a one example code to determine the MAF sensor like using A and B values..e.g. how to use 
myELM327.findResponse();
byte B = myELM327.responseByte_0;
byte A = myELM327.responseByte_1;
// do things with A and B

nices0325, 2021-01-12:
It really good quesition for me either!! plz check my code 
I added this to my cpp file
float ELM327::fuelremain()
{
	if (queryPID)(SERVICE_01, FUEL_TANK_LEVEL_INPUT))
	return (uint32_t)((responseByte_1 * 256) + responseByte_0) / 64));

	return ELM_GENERAL_ERROR;

this is my sketch
    byte B = myELM327.responseByte_0;
    byte A = myELM327.responseByte_1;

        fuelremain = (uint32_t)tempFUEL_TANK_LEVEL_INPUT;
        Serial.print("Fuel Remain : "); Serial.print(fuelrmain); Serial.println("%");

would it be work well?? sorry for my ignorance