PowerBroker2/ELMduino #40: ERROR: ELM_NO_DATA on ESP32Dev device

i82hisaj, 2020-07-30:
**Describe the bug**
Hello! First of all, many thanks for your library, what an excellent work! Nevertheless, I'm having some troubles while trying to use it with a ESP32Dev device. I'm continuously being shown the following error:

> Attempting to connect to ELM327...
> Connected to ELM327
> Payload received for rpm: NODATA
> 	ERROR: ELM_NO_DATA
> Payload received for kph: NODATA
> 	ERROR: ELM_NO_DATA

I have tested the ESP32_test.ino example, with a successful connection to the device, as follows:

> Attempting to connect to ELM327...
> Connected to ELM327
> Ensure your serial monitor line ending is set to 'Carriage Return'
> Type and send commands/queries to your ELM327 through the serial monitor
> >AT Z
> ELM327 v2.1
> >AT E0
> OK
> >AT S0
> OK
> >AT SP0
> OK
> >010C
> SEARCHING...
> 410C0B94
> 

I'm wondering if you could give me a clue about what's going on. Thanks in advance!

**To Reproduce**
Upload to a ESP32 device the code below.
Connect to ELM327 device

**Expected behavior**
Show values to .kph(), .rpm() and standard PIDs 

**Equipment**
Microcontroller: ESP32-Bit (programmable as ESP32 Dev Module), like this one: https://www.amazon.es/ESP32-Bit-Desarrollo-Bluetooth-Ethernet-ESP32-T/dp/B07GD4SGYM
OBD II Controller: OBD II Mini ELM327, like this one: https://www.amazon.es/Interfaz-Bluetooth-herramienta-an%C3%A1lisis-diagn%C3%B3stico/dp/B00WU97RH8
Car Model: Peugeot 206

**Code**
`// modified test + my custom PIDs
// based purely on library

#include "BluetoothSerial.h"
#include "ELMduino.h"

#define ELM_PORT SerialBT
#define ESP_BLUETOOTH_NAME "ESP32"

BluetoothSerial SerialBT;
ELM327 myELM327;

void setup()
{
  //pinMode(LED_BUILTIN, OUTPUT);
  //digitalWrite(LED_BUILTIN, HIGH);

  Serial.begin(115200);
  //ELM_PORT.setPin("1234");
  ELM_PORT.begin(ESP_BLUETOOTH_NAME, true);

  Serial.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  //if (!ELM_PORT.connect("00:1d:a5:68:98:8d"))
  {
[... 241 lines cut ...]
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_GENERAL_ERROR)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(500);
}`

**Wiring**
N/A

**Screenshots**
N/A

**Additional Context**
N/A

**Troubleshooting Connection Issues**
- Try 38400 baud instead of 115200 ---> Tried with no success.
- If using an HC-05 for communication, [ensure it's configured properly](https://forum.arduino.cc/index.php?topic=600014.msg4159554#msg4159554) ---> N/A
- Ensure the car is on and running during troubleshooting ---> Car running
-Also tested with BT mac address, and providing PIN code, with no success.

PowerBroker2, 2020-07-30:
I see that the ELM327 is responding to your queries with `NO DATA`. This means that for some reason the ELM327 can't get the data off the diagnostics computer and is more of a car <-> ELM issue instead of an ELM <-> Arduino issue.

What could be going wrong:
- Car isn't in the correct state (i.e. on and running), but you've already said this is the case
- The Service ID/PID are incorrect, but you seem to be using the correct ones
- Generic bug with the ELM327 or diagnostics computer, but this is very unlikely
- The ELM327 isn't correctly configured by the lib - Try the following with the test sketch again and post the results:
```
AT Z
AT E0
AT S0
AT AL
AT TP A0
010C
010D
```

patfelst, 2020-07-31:
You received a response for RPM of 410C0B94 = 741 RPM using the test example. My guess is it's timing out when you query RPM in your loop code. Try increasing the timeout value from 1000 to 5000 for example.

Also you you have a few type problems to fix, e.g. `engineLoad = engineLoadTemp/2.55;` where `engineLoad` is an integer (int32_t), you should probably make it a float.

PowerBroker2, 2020-07-31:
> You received a response for RPM of 410C0B94 = 741 RPM using the test example. My guess is it's timing out when you query RPM in your loop code. Try increasing the timeout value from 1000 to 5000 for example.

It's not timing out because a message was received from the ELM, but was the string `NO DATA`

patfelst, 2020-07-31:
ah of course, apologies. Odd that he did receive a response with the test program though.

i82hisaj, 2020-07-31:
> You received a response for RPM of 410C0B94 = 741 RPM using the test example. My guess is it's timing out when you query RPM in your loop code. Try increasing the timeout value from 1000 to 5000 for [...]

Good catch about the float/integer mismatch. I will fix it! 👍

i82hisaj, 2020-07-31:
> I see that the ELM327 is responding to your queries with `NO DATA`. This means that for some reason the ELM327 can't get the data off the diagnostics computer and is more of a car <-> ELM issue [...]

Ok, I will test what you are requesting. I will post the results as soon as possible (in a few hours). Thanks!

i82hisaj, 2020-07-31:
Hi, I could test it earlier than expected. Here are the results of the test:

> AT Z
> ELM327 v2.1
> >AT E0
> OK
> >AT S0
> OK
> >AT AL
> OK
> >AT TP A0
> OK
> >010C
> NO DATA
> >010D
> NO DATA
> >

PowerBroker2, 2020-08-01:
Seems like your ELM327 doesn't like something about the settings the lib uses, you can edit your local copy of the library to adjust what configuration settings works for you

i82hisaj, 2020-08-01:
Hello again,

Many thanks for your help. I have investigated a bit, and I have obtained additional information from the ELM327 using Torque for it (BTW, it works perfectly fine in Torque). I have noticed that my ELM327 uses ISO14230-4 KWP 5bd protocol. I have changed the declaration of the class in file "ELMduino.h" to the following:

```
	bool begin(Stream& stream, char protocol='4', uint16_t payloadLen = 40);
	bool initializeELM(char protocol='4');
```

...as the headers in the file established that this protocol was identified as '4':

```
const char ISO_14230_5_BAUD_INIT      = '4';
const char ISO_14230_FAST_INIT        = '5';
```

After that, I have restarted Arduino IDE and compiled and uploaded again my code to the ESP32. Unfortunately, I had the same results:

> ets Jun  8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:1
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1216
> ho 0 tail 12 room 4
> load:0x40078000,len:9720
> ho 0 tail 12 room 4
> load:0x40080400,len:6352
> entry 0x400806b8
> Attempting to connect to ELM327...
> Connected to ELM327
> Payload received for rpm: NODATA
> Payload received for kph: NODATA
> Payload received for Engine Load: NODATA

After that, I have also tested with protocol '5', with the same results again.

Is there any other place in the library where I can make adjustments? Thanks in advance!

i82hisaj, 2020-08-01:
Hello again,

Something else I have tested (again without success). I have also modified ELMduino.cpp to this:

```
bool ELM327::initializeELM(char protocol)
{
	char command[10] = { '\0' };
	connected = false;
	
	sendCommand(RESET_ALL);
        delay(500);

	sendCommand(ECHO_OFF);
	delay(500);

	sendCommand(PRINTING_SPACES_OFF);
	delay(500);
	
	//sendCommand(ALLOW_LONG_MESSAGES);
	//delay(100);

```
...as the test example didn't use long messages. But as said, no success with this (both with protocol 4 and 0).

PowerBroker2, 2020-08-02:
In that case, use what worked initially:

```
AT Z
AT E0
AT S0
AT SP0
```

PowerBroker2, 2020-08-05:
Closing for now, but if you still have issues, feel free to reopen

closed 2020-08-05 by PowerBroker2

i82hisaj, 2020-08-09:
Many thanks for your help with this. I'm on my holidays these days, so I have no chance to continue making tests. ;)
In any case, as it seems to be an issue with my device, no worries about closing the ticket. If I make any progress with the code, I will let you know.
Best regards!

nices0325, 2021-01-10:
I have a same issue withe my elmduino project. I have successfully connected and gotten the data but it suddenly doesn't work... what would be the problem. If you configure what was the problem and solved it plz let me know

nomidaepapi, 2023-06-16:
Hi @PowerBroker2 , I would like to request to open this ticket one more time. 

My set up is a ELM327 Bluetooth adapter and esp32 audrino.

I have two cars a Lexus GS 300, 2001 and a Toyota sienna 2014.

when I try the test example to try reading the RPM. It is unsuccessful on the Lexus GS 300 and see is NO DATA; but for the Toyota sienna 2014, it reads completely fine and displays the RPM (the Bluetooth connection is successful in both cars). I have read through many threads and try to fix this for three weeks but I’m running out of ideas.

This thread was very useful as a problem is similar to mine. I’m very strong with programming, but I do not understand what I have to do or fix in your opensource code to make it work with my Lexus. Is there any way you can help?

VinicioAM, 2023-09-23:
Hello. I'm facing the same problem. 

When I run the example 'ESP32_Bluetooth_Serial', which makes a lot of data request, I just get 'NO DATA' answer.
When I run the example 'ESP32_test', then send:

AT Z
AT E0
AT S0
AT AL
AT TP A0
010C
010D

I can receive the data (encoded), and it is correct.

Is there any way to fix it by using the example 'ESP32_Bluetooth_Serial'?
I don't have enough experience in C++ to fix it by myself.

thank so much for any help.

PowerBroker2, 2023-09-23:
If you have a different startup procedure that works while the default one doesn't, you might try making an `initializeELM()` function at the sketch level with the proper configuration commands. Should be a pretty easy copy, paste, and edit. Here is the original function for reference:

https://github.com/PowerBroker2/ELMduino/blob/636aa957f6f7d6ff01bf41fbfce004c1efc5883c/src/ELMduino.cpp#L85-L145

VinicioAM, 2023-09-23:
Yes, Mr. P_B, I did it already.

Here is one of the my attempts:

`	delay(10000);
	sendCommand_Blocking(RESET_ALL);
	delay(1000);

	sendCommand_Blocking(ECHO_OFF);
	delay(1000);

	sendCommand_Blocking(PRINTING_SPACES_OFF);
	delay(1000);

	sendCommand_Blocking(ALLOW_LONG_MESSAGES);
	delay(1000);

	sendCommand_Blocking("AT TP A0");
	delay(1000);

	sendCommand_Blocking(DISP_CURRENT_PROTOCOL);
	delay(10000);

	sendCommand_Blocking("AT SP 7");
	delay(10000);

	sendCommand_Blocking(DISP_CURRENT_PROTOCOL);
	delay(10000);`

This is exactly the same config that I do when I use 'ESP32_test', send 010C and get the encoded RPM.
I get the configuration sucessfull. Anyway, when using 'ESP32_Bluetooth_Serial', it still being delivered 'NODATA' (even with the new config). 

There is a point: to see RPM, I send 010C. Anyway, in the debug mode inside 'ESP32_Bluetooth_Serial' the message sent to ELM  by using rpm function is not exactly 010C (It is 0112). Does this hexadecimal to decimal conversion have any problem?

Second point: might it be a timeout/too fast/wrong baud rate problem? 

I have already tried change 
`	// Set data timeout
	sprintf(command, SET_TIMEOUT_TO_H_X_4MS, dataTimeout / 4);
	sendCommand_Blocking(command);
	delay(100);`
to 1000ms, 2000ms, 5000ms, 100ms, 10000ms, 50ms, 0ms... does not work.

Tried all baud rates too.. Does not work.

Any suggestion?

DerKleinePunk, 2023-09-24:
I don't know it ist possible you have the same probleme as me.

The AT Commands not Understand from my ELM it looks as an Timing Problem on ESP32.
When i use an Android Terminal it work but not work  in ESP Software.
You Receive an OK on the AT Commands ? When not you can try my Fork

https://github.com/DerKleinePunk/ELMduino
https://github.com/DerKleinePunk/ODB-ESP32-Reader My Littel Test Projekt