PowerBroker2/ELMduino #41: Timeout Issue

martindo72050, 2020-08-07:
Hello. I'm sorry to open another issue on this topic, I've been trying to figure this out for a few days now and can't. Probably my inexperience. Anyway my hardware is:

- Arduino Mega
- HC05
- Bafx ELM327

I'm able to connect and retrieve data using AT commands, but I get a timeout error after connecting using the example code. Any ideas?

PowerBroker2, 2020-08-08:
I've had timeouts occur, but only rarely. Are you able to get any data at all with the example before getting timeouts? What happens if you edit the CPP file to follow [this configuration](https://github.com/PowerBroker2/ELMduino/issues/40#issuecomment-667602916)?

martindo72050, 2020-08-09:
No data before timeouts. I'll try that and let you know

PowerBroker2, 2020-08-09:
You can use the macros from the header file, but I think you can do what you posted instead

martindo72050, 2020-08-09:
Tried it and still get the same issue

PowerBroker2, 2020-08-09:
You said you can get data using AT commands. Can you provide that list of AT commands and their responses?

martindo72050, 2020-08-09:
Yes. I actually only used the AT commands for setting up the HC05, then I got it out of AT mode and started simply typing the PIDs. Here I repeatedly typed 010C (rpm) and got this response:
[image]

PowerBroker2, 2020-08-10:
In that case, try getting rid of all AT commands in the `begin()` method and see if that fixes it

martindo72050, 2020-08-10:
Same result. Here is my current Arduino code:
#include <ELMduino.h>

#include <SoftwareSerial.h>
#include "ELMduino.h"

//SoftwareSerial mySerial(10, 11); // RX, TX
#define ELM_PORT Serial1

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  Serial.begin(38400);
  ELM_PORT.begin(38400);

  Serial.println("Attempting to connect to ELM327...");

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
  {
    Serial.print(F("\tERROR: "));
    Serial.println(myELM327.status);
    delay(100);
  }  
  delay(1000);
}

brainfoolong, 2020-11-29:
Hi, i also have the same problem. If looked into the elmduino code and i found something that i can't explain. Currently it is so: After successfull initialization, once you get a `ELM_TIMEOUT` or `ELM_UNABLE_TO_CONNECT` after any command, you are stuck in that status and can't send any other commands.

It's because in current code, the `connected` flag is set to false on the begin of every sendCommand -> https://github.com/PowerBroker2/ELMduino/blob/342fc1e581ecf44b829053a55a6a4aeaf6852b81/src/ELMduino.cpp#L508

This flag is only updated to true when you have received any success request (no timeout or unable to connect) -> https://github.com/PowerBroker2/ELMduino/blob/342fc1e581ecf44b829053a55a6a4aeaf6852b81/src/ELMduino.cpp#L554

And the problem with this is, every call to `queryPID`, which you basically use for every data you want, require a `connected = true` -> https://github.com/PowerBroker2/ELMduino/blob/342fc1e581ecf44b829053a55a6a4aeaf6852b81/src/ELMduino.cpp#L360

But this `connected ` will never be `true` again.

I've created a pull request, maybe this is helpful for someone and maybe this will be merged by the maintainer.

PowerBroker2, 2020-11-29:
Oh dang, good catch! I'll take a look at the PR. I've had trouble with many users (including myself, actually) having this timeout error issue, but could never figure it out - I hope this bug is the root of the issue.

PowerBroker2, 2020-12-01:
This should be resolved with release [2.3.2](https://github.com/PowerBroker2/ELMduino/releases/tag/2.3.2)

closed 2020-12-01 by PowerBroker2