PowerBroker2/ELMduino #42: Using MAC address when connect to ELM327 via Bluetooth

digidax, 2020-08-28:
When I have discovered by Serial Bluetooth Terminal the ODBII device with MAC 23:32:17:05:88:45
I have to set the MAC address as an array with HEX converted values for using in ELM_PORT.connect(address)

regarding to the discovered MAC above this usage would be:

uint8_t address[6] = {0x17, 0x20, 0x11, 0x05, 0x58, 0x2D};
ELM_PORT.connect(address);

because the lib cpp expects an array for a MAC and a string if the name was used.
Can I also use the DEC notation of the MAC instead of the HEX values?

PowerBroker2, 2020-08-29:
I've never tried to connect to any bluetooth device via MAC address on an ESP32 - you'll have to do some researching on your own to make sure you're doing that correctly. That being said, [here is an example sketch that apparently works using the MAC address](https://forum.arduino.cc/index.php?topic=617795.msg4679253#msg4679253).

Did you try using the Bluetooth Serial ESP32 example? If so, what were the results? If not, please try it. You shouldn't need the MAC address, only the Bluetooth device name. Also ensure that all other devices previously connected to the ELM327 are turned off before trying to connect via Arduino. Also note that you might have to try other bauds than what the example provides.

digidax, 2020-08-29:
The Bluetooth Serial ESP32 example was working fine, with Android Serial BT Terminal. Sending to ESP and sending from ESP to the App was successful. As far I seen, I have entered the MAC address correctly like shown in the [https://forum.arduino.cc/index.php?topic=617795.msg4679253#msg4679253](url) example.

>Also note that you might have to try other bauds than what the example provides.

Do you mean I have to change the baud rate DEBUG_PORT.begin(115200); of cause also in the IDE serial monitor? Do I have also  change the upload speed to the new value? r.g. all values to 9600 ?
Thanks for helping,

PowerBroker2, 2020-08-29:
> The Bluetooth Serial ESP32 example was working fine, with Android Serial BT Terminal.

No, I mean the example without any serial terminal, just the Arduino

digidax, 2020-08-30:
Thanks, have not used you example 1:1 and this is the result:
`clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[I][BluetoothSerial.cpp:510] _init_bt(): device name set
[I][BluetoothSerial.cpp:225] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:702] connect(): master : remoteName
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:327] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT
[I][BluetoothSerial.cpp:329] esp_bt_gap_cb(): Scanned device: 23:32:17:05:88:45
[I][BluetoothSerial.cpp:336] esp_bt_gap_cb(): ESP_BT_GAP_DISC_RES_EVT : EIR : OBDII : 32
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
[I][BluetoothSerial.cpp:379] esp_bt_gap_cb(): ESP_BT_GAP_DISC_STATE_CHANGED_EVT
Couldn't connect to OBD scanner - Phase 2
`

PowerBroker2, 2020-09-24:
What happens if you comment out the line where you set the PIN? Most ELM327 dongles don't need that for connection. If commenting that line out doesn't fix your problem, your ESP32 is most likely defective.

closed 2020-09-24 by PowerBroker2

digidax, 2020-09-25:
With disabled PIN the problem is not solved. What ESP32 board are you using wand where have you buy it?

PowerBroker2, 2020-09-25:
[https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1](https://www.amazon.com/gp/product/B0718T232Z/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

emanuelebiondi, 2021-01-23:
I don't know if is a solution, but i had the same problem and I've fixed reversing 
`ELM_PORT.setPin("1234");
ELM_PORT.begin("OBDII", true);`

like this:
`ELM_PORT.begin("OBDII", true);
ELM_PORT.setPin("1234");`