PowerBroker2/ELMduino #43: >ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054

sheshman, 2020-09-15:
Hi,

First of all thanks for this great library, i'm using arduino UNO+ESP32S+Bluetooth ELM327 dongle, with ESP32_test example from your library, it connects to elm327 dongle and communicating without any problem but i'm receiving this line on the terminal and didn't figure it out why it's returning this ">ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054"

When i send 010C(speed) it returns "41 0C 00 00" without any problem. But in every 3-5 seconds it returns ">ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054"

Why? and how i can fix it?
My console output as below;
`
ASSERT_WARN(1 8), in lc_task.c at line 5054010C
010C
ASSERT_WARN(1 8), in lc_task.c at line 505441 0C 00 00 

>ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054010C
010C
41 0C 00 00 

>ASSERT_WARN(1 8), in lc_task.c at line 5054010C
010C
41 0C 00 00 

>ASSERT_WARN(1 8), in lc_task.c at line 5054010D
010D
41 0D 00 

>ASSERT_WARN(1 8), in lc_task.c at line 5054ASSERT_WARN(1 8), in lc_task.c at line 5054
`

PowerBroker2, 2020-09-15:
This is really bizarre, I don't think I've seen this one before. Can you post your full code?

sheshman, 2020-09-15:
here is my(your :) ) code;
`
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

#define DEBUG_PORT Serial
#define ELM_PORT   SerialBT

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  DEBUG_PORT.begin(115200);
  ELM_PORT.setPin("1234");
  ELM_PORT.begin("AA:BB:CC:11:22:33", true);

  DEBUG_PORT.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);
  }

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}

void loop()
{
  if(DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();

    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }

  if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if(c == '>')
      DEBUG_PORT.println();

    DEBUG_PORT.write(c);
  }
}
`

PowerBroker2, 2020-09-15:
> here is my(your :) ) code;

That isn't my code since you edited it, lol

It seems that [the warning is caused by a connection issue](https://github.com/espressif/esp-idf/issues/5140#issuecomment-626561262), but that doesn't make sense since you can obviously get RPM data.

That being said, why do you have `ELM_PORT.begin("AA:BB:CC:11:22:33", true);`? The first argument in `begin()` is the BT name of the ESP32. See [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/src/BluetoothSerial.h). Also, are you sure you need to call `setPin()`?

sheshman, 2020-09-16:
> That isn't my code since you edited it, lol

good point :) 

> why do you have `ELM_PORT.begin("AA:BB:CC:11:22:33", true);`?

AA:BB:CC:11:22:33 is my elm327 dongle's mac address, you were said at the arduino forum "if anybody have connection issue, then use your dongle's mac address instead of name" so i haven't tried with name and used mac address to be sure.

> are you sure you need to call `setPin()`?

my elm327 dongle has pairing password, i mean when i try to connect through my phone it asking for password, that's why i did enable the setPin() but today i'll try without setpin to see does it make any difference.

sheshman, 2020-09-16:
finally had time to test it, when i commented out the setPin() line those error messages are gone (i don't have any idea about how it's connecting to elm327 without password tho), console out put as below;
```
>010C
010C
41 0C 0D 78 
>010D
010D
41 0D 00 

```
41 0C 0D 78 = 3448 and 3448/4 = 862 Rpm which is my car's idle rpm :) until this point we are good.

But at this point another problem just started, i'm plugging in ELM327 dongle to obd port, turning ignition to battery mode (all lights turning on at the dashboard but engine is not starting), plugging esp32 to my notebook and opening arduino terminal with ctrl+shift+m, nothing happens, no message, no error code, no nothing, when i hit to EN button on the ESP32 (which resets the board as you know), then esp32 says "Connecting to ELM327" after that it connects without any problem and run commands, so is that normal behavior? i mean when i start the engine both arduino & esp32 will boot (i'll connect arduino's power to ignition) and as you know OBD2 port allways has power even if you remove your keys, in this case when ESP32 become alive it should connect to ELM327 right away right? But mine waiting to push reset button, am i doing something wrong?

Oh by the way,i'm using the same example code ;
```
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

#define DEBUG_PORT Serial
#define ELM_PORT   SerialBT

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  DEBUG_PORT.begin(115200);
  //ELM_PORT.setPin("1234");
  ELM_PORT.begin("OBDII", true);

  DEBUG_PORT.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);
  }

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}

void loop()
{
  if(DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();

    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }

  if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if(c == '>')
      DEBUG_PORT.println();

    DEBUG_PORT.write(c);
  }
}

```

PowerBroker2, 2020-09-16:
```C++
ELM_PORT.begin("ESP32_BT_NAME", true); // <--- This tells the ESP32's BT driver to display "ESP32_BT_NAME" as the searchable name **for the ESP32** _not_ the ELM327

ELM_PORT.connect("OBDII"); // <--- Connect to the device with name "OBDII"
ELM_PORT.connect("AA:BB:CC:11:22:33"); // <--- Connect to the device with MAC address "AA:BB:CC:11:22:33"
```

I'm not sure about your current issue. Is it a big problem for your application?

sheshman, 2020-09-16:
long story short, my current issue is, esp32 connecting to elm327 dongle without any problem and i'm receiving data from it, but at the beginning it's not connecting until i reset the esp32

sheshman, 2020-09-17:
ok problem solved, it turns out that my esp32 is broken, i've talked with the seller, they've tested and said it's faulty, replaced with new one and worked like a charm.

Thanks for your time, i owe you a beer, you can close the issue.

closed 2020-09-17 by PowerBroker2