PowerBroker2/ELMduino #44: Add spacial PID

Jastreb07, 2020-09-24:
Hi,

how i can add a new PID (221E12) i have foud who i can change the PIDs, but all PID is in 0x???? form and im stuck on this point

PowerBroker2, 2020-09-24:
You can do something like this:

```C++
  if (myELM327.queryPID("221E12"))
  {
    int32_t temp = myELM327.findResponse();

    Serial.print("Raw Response: ");
    for (byte i = 0; i < PAYLOAD_LEN; i++)
      Serial.write(myELM327.payload[i]);
    Serial.println();
    
    if (myELM327.status == ELM_SUCCESS)
      Serial.print("Parsed Response: "); Serial.println(temp);
  }
```

Jastreb07, 2020-09-25:
> You can do something like this:
> 
> ```c++
>   if (myELM327.queryPID("221E12"))
>   {
>     int32_t temp = myELM327.findResponse();
> 
>     Serial.print("Raw Response: ");
>     for (byte i = 0; i < PAYLOAD_LEN; i++)
>       Serial.write(myELM327.payload[i]);
>     Serial.println();
>     
>     if (myELM327.status == ELM_SUCCESS)
>     {
>       Serial.print("Parsed Response: "); Serial.println(temp);
>     }
>     else
>     {
>       Serial.print("ERROR: "); Serial.println(myELM327.status);
>     }
>   }
> ```

'PAYLOAD_LEN' was not declared in this scope

Screenshot: https://prnt.sc/unl6rm

PowerBroker2, 2020-09-25:
Sorry, that was a snippet from old code. Change `PAYLOAD_LEN` to `myELM327.PAYLOAD_LEN`

Jastreb07, 2020-09-25:
Thank its work :)

Jastreb07, 2020-09-25:
> Sorry, that was a snippet from old code. Change `PAYLOAD_LEN` to `myELM327.PAYLOAD_LEN`

im get every time "Parsed Response: 0" how i can fix it?
i have testet the PID on "Torque"-APP and it give me value back

PowerBroker2, 2020-09-25:
What does the Arduino say for the "Raw Response"?

Jastreb07, 2020-09-25:
> What does the Arduino say for the "Raw Response"?

621E2346

PowerBroker2, 2020-09-25:
Hmm, that should parse to `1646142278` (decimal), I can double check everything and let you know

PowerBroker2, 2020-09-25:
What about this?:

```C++
  if (myELM327.queryPID("221E12"))
  {
    uint64_t temp = myELM327.findResponse();

    Serial.print("Raw Response: ");
    for (byte i = 0; i < myELM327.PAYLOAD_LEN; i++)
      Serial.write(myELM327.payload[i]);
    Serial.println();
    
    if (myELM327.status == ELM_SUCCESS)
    {
      Serial.print("Parsed Response: "); Serial.println(temp);
    }
    else
    {
      Serial.print("ERROR: "); Serial.println(myELM327.status);
    }
  }
```

Jastreb07, 2020-09-27:
> ```c++
> if (myELM327.queryPID("221E12"))
>   {
>     uint64_t temp = myELM327.findResponse();
>
>     Serial.print("Raw Response: ");
>     for (byte i = 0; i < myELM327.PAYLOAD_LEN; i++) [...]

call of overloaded 'println(uint64_t&)' is ambiguous
https://prnt.sc/uommak

KoYoT77, 2020-10-09:
> im get every time "Parsed Response: 0" how i can fix it?
> i have testet the PID on "Torque"-APP and it give me value back
> 
I have same problem, all the time response is equal "0".
So I'm curious about solution. Looking forward.

PowerBroker2 -> I have tried to contact you on Arduino.cc forum in private message.

PowerBroker2, 2020-10-09:
Please post exactly what you're trying to do along with your entire sketch in tags along with a description of your physical setup and experiment parameters (i.e. car make/model, is the car on and running, etc). Also, make sure all other bluetooth devices are either disabled or turned off. Testing with different bauds also clears some people's issues.

KoYoT77, 2020-10-14:
**Project:**  Retrive information about DPF,
**Physical setup:** Arduino Nano + BlueTooth HC-05 -> iCar2Pro(ELM327)+Opel Insignia; engine A20DTH.
**Code below**

```
/*
Code base on PowerBroker2 & miguelos6 - Github https://github.com/PowerBroker2/ELMduino/issues/4
HEADER for Engine parameters Jakub Dziworski  - http://jakubdziworski.github.io/
Car: Vauxhal/OPEL; model: Insignia; Engine: A20DTH
Communication->Arduino > BT HC-05 > ELM V-Link ICAR2Pro
Results printed on LCD 16x2
*/

#include <Wire.h>
#include <LiquidCrystal_I2C.h>
#include <SoftwareSerial.h>
#include <ELMduino.h>

LiquidCrystal_I2C lcd(0x27,16,2);  // set the LCD address to 0x27 for a 16 chars and 2 line display
SoftwareSerial mySerial(10, 7); // RX, TX for BlueTooth HC-05
#define ELM_PORT mySerial
ELM327 myELM327;

/*//////////////CONST &  VARIABLE DEFINITION///////////////// */
const long debugRate = 115200;
const long mySerialRate = 38400; //one working BT BOUND 34800,0,0
int32_t dirtLevel = -1;
int32_t burn = -1;
int32_t regenStatus = -1;
int32_t kmsSinceDpf = -1;
  

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
  
  lcd.init();
  lcd.backlight();
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print("...CONNECTING...");
  lcd.setCursor(0,1);  
  lcd.print("BT HC05 > V-LINK");
  Serial.begin(debugRate);
  Serial.println("Serial Monitor - START");
  ELM_PORT.begin(mySerialRate);
  Serial.println("...Connecting to ELM327...");  

/*//////////////CHECK BT CONNECTION///////////////// */
  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("V-LINK - NOT CONNECTED");
    lcd.clear();
    lcd.setCursor(0,0);
    lcd.print("..NOT CONNECTED.");
    lcd.setCursor(0,1);  
    lcd.print("BT HC05 > V-LINK"); 
    while (1);
  }
  Serial.println("V-LINK - CONNECTED SUCCESFUL");
    lcd.clear();
    lcd.setCursor(0,0);
    lcd.print("BT HC05 > V-LINK");
    lcd.setCursor(0,1);  
    lcd.print("....CONNECTED...");

/*//////////////CONFIRM BY FLASHING BUILT LED///////////////// */
flashLEDx5();

[... 230 lines cut ...]
  {
    uint64_t temp = myELM327.findResponse();

    Serial.print("Raw Response: ");
    for (byte i = 0; i < myELM327.PAYLOAD_LEN; i++)
      Serial.write(myELM327.payload[i]);
    Serial.println();

    if (myELM327.status == ELM_SUCCESS)
    {
      Serial.print("Parsed Response: "); Serial.println(temp);
    }
    else
    {
      Serial.print("ERROR: "); Serial.println(myELM327.status);
    }
  }
```
and I got result during verifiying of sketch same as **Jastreb07** 

>     overloaded 'println(uint64_t&)' is ambiguous

So, i'm waiting your feedback.
Thank you in advance
KoYoT77

PowerBroker2, 2020-10-17:
You can typecast temp into a normal int for printing in your case. Also, what exactly is going wrong in your case? Also, can you edit the lib files to print the exact command bytes sent to the ELM327 the next time you test?

cyberelectronics, 2020-10-17:
In my case (Hyundai Kona EV), when the car is turned OFF, the OBD adapter sends something like this (Rawdata, OBD protocol 6):
`03E0:620101FFF7E7`

But when it is ON, "7F2212" frames appears (random length, before or after the actual data 0:620101FFF7E7, where 0: is the indicator for DataFrame 0. Can send even 8 similar dataframes. Each dataframe contains 7bytes, except 0:, which contains only 6bytes).
`7F22127F22127F22127F22127F221203E0:620101FFF7E71:FF8A00000000802:00000ED91814133:141713000011C2`
So your actual data will be after this 7Fxxx bytes.

KoYoT77, 2020-10-19:
>
> 
> You can typecast temp into a normal int for printing in your case. Also, what exactly is going wrong in your case? Also, can you edit the lib files to print the exact command bytes sent to the ELM327 the next time you test?

PowerBroker2 
**What is going wrong**
In my case id doesn't matter what I asked:
`myELM327.queryPID(34, 12916)`
`myELM327.queryPID(34, 12919)`
`myELM327.queryPID(34, 12917)`
answer is all the time the same
>
> Payload received for soot: 7F2222

change of `uint64_t` to `int`
still gave me same answer.
Sam for engine running & not.

I suppose somewhere in my code there is a bug, but I'm really newbie in programming.
So if you be so kind please check it.

KoYoT77, 2020-10-19:
> 
> 
> In my case (Hyundai Kona EV), when the car is turned OFF, the OBD adapter sends something like this (Rawdata, OBD protocol 6):
> `03E0:620101FFF7E7`
> 
> But when it is ON, "7F2212" frames appears (random length, before or after the first actual dataframe 0:620101FFF7E7, where 0: is the indicator for DataFrame 0. Can send even 8 similar dataframes. Each dataframe contains 7bytes, except 0:, which contains only 6bytes).
> `7F22127F22127F22127F22127F221203E0:620101FFF7E71:FF8A00000000802:00000ED91814133:141713000011C2`
> Need to modify the 'payloadlen' value to 200 for 8 dataframes in elmduino.h :
> `bool begin(Stream& stream, char protocol='0', uint16_t payloadLen = 200);`
> 
> So your actual data will be after this 7Fxxx bytes.

I have tried what you suggest, Elmduino.h opened, I look for payloadLen in my lib its value is set to 40.
But there are two declaration of variables:
`uint16_t PAYLOAD_LEN;` called in my sketch for iteration 
`for (byte i = 0; i < myELM327.PAYLOAD_LEN; i++)`
and  
`uint16_t payloadLen = 40` used in begin function of the lib.
But please be forgiving to the beginner if I said something stupid.
Regards

PowerBroker2, 2020-10-19:
`payloadLen` is used by the `begin()` member function to set the value of `PAYLOAD_LEN` (or `myELM327.PAYLOAD_LEN` in the sketch). By default that value is 40, but you can change it to another value, i.e. 200.

cyberelectronics, 2020-10-20:
@KoYoT77 I think we have similar issues.
Your data is there, after 7F.... bytes, but probable the parser stops after a few bytes.
One of my friend sent me a parser, which worked great for ~15seconds :) , after this the ESP32 module crashed (picture attached at the end, where you can see where is the issue, but I don't know how to fix it and he is busy now).
Here is the code for parsing:
```
void clearData() {
	dataFrame0 = "";
	dataFrame1 = "";
	dataFrame2 = "";
	dataFrame3 = "";
	dataFrame4 = "";
	dataFrame5 = "";
	dataFrame6 = "";
	dataFrame7 = "";
	dataFrame8 = "";
  
}

char* appendCharToCharArray(char* array, char c) {
    int len = strlen(array);

    char ret[len+2];

    strcpy(ret, array);    
    ret[len] = c;
    ret[len+1] = '\0';

    return strdup(ret);
}

void addToFrame(int frame, char c) {
	switch(frame) {
		case 0:
			dataFrame0 = appendCharToCharArray(dataFrame0, c);
			break;
		case 1:
			dataFrame1 = appendCharToCharArray(dataFrame1, c);
			break;
		case 2:
			dataFrame2 = appendCharToCharArray(dataFrame2, c);
			break;
		case 3:
			dataFrame3 = appendCharToCharArray(dataFrame3, c);
			break;
		case 4:
			dataFrame4 = appendCharToCharArray(dataFrame4, c);
			break;
		case 5:
			dataFrame5 = appendCharToCharArray(dataFrame5, c);
			break;
		case 6:
			dataFrame6 = appendCharToCharArray(dataFrame6, c);
			break;
		case 7:
			dataFrame7 = appendCharToCharArray(dataFrame7, c);
			break;
		case 8:
			dataFrame8 = appendCharToCharArray(dataFrame8, c);
			break;
	}
}

// get substring between m and n (excluding n) 
char* substr(const char *src, int m, int n) {
	n+=1;
	int len = n - m;
  
	char *dest = (char*)malloc(sizeof(char) * (len + 1));

	for (int i = m; i < n && (*(src + i) != '\0'); i++)
	{
		*dest = *(src + i);
		dest++;
	}

	*dest = '\0';
	  
	return dest-len;
}

void parse(char *raw) {
	int frame = -1;
	
	int len = strlen(raw);
	
	for(int i=0; i<len; i++) {
		
		if(raw[i+1] == ':') { //start frame
			frame = (int) raw[i] - '0';
			continue;
		}
		
		if(raw[i] == ':') {
			continue;
		}
		
		if(frame == -1) {
			continue;
		}
		
		if(raw[i] == '>') {
			frame = -1;
			continue;
		}
		
		addToFrame(frame, raw[i]);
	}
	
}

int convertToInt(char *hexStr) {
	return (int)strtol(hexStr, NULL, 16);
	
}

```
And here is how to use it in the loop(), with some delay/timer (in my case, this will return/print battery voltage):

```
myELM327.sendCommand("AT SH 7E4");      // Set Header BMS (Battery Management System)

if (myELM327.queryPID("220101")) {      // BMS PID
                read_rawdata(); 

                BATTv = ((convertToInt(substr(dataFrame2, 4, 5))<<8) + convertToInt(substr(dataFrame2, 6, 7)))/10;  
                DEBUG_PORT.print("Main Battery Voltage (V): ");
                DEBUG_PORT.println(BATTv);
}

```
Read_rawdata function looks like this (will print the rawdata coming from OBD, then will parse it):
```
void read_rawdata(){
                       // move data from OBD to Rawdata array
                       char rawData[myELM327.recBytes];
                       int n = 0;
                                                                            
                       DEBUG_PORT.print("Payload received: ");
                       
                       for (int i=0; i<myELM327.recBytes; i++) {
                                                                  rawData[n++] = myELM327.payload[i];  
                                                                  DEBUG_PORT.print(myELM327.payload[i]);
                                                               }
                       DEBUG_PORT.println();
                                                                             
                       parse(rawData);  // parse data received from OBD
  }

```
Rawdata from OBD: 
`Payload received: 7F22127F22127F22127F22127F221203E0:620101FFF7E71:FF8A00000000832:00240ED71713133:141713000010C14:22C12800008B005:00450B0000434B6:000019A80000187:1200200EE70D018:7B0000000003E8`

Parse(rawData), will look for dataframes, which begins with 0: 1: 2: 3: etc... then I can select any byte in any dataframe:
Ex.: 
Dataframe2 >>>>>>>>>>>  `2:00240ED7171313` 

`convertToInt(substr(dataFrame2, 4, 5)) `>>>> will return a single byte on location 4 and 5 (from 0) in dataframe 2.
`convertToInt(substr(dataFrame2, 6, 7))` >>>> will return a single byte on location 6 and 7 (from 0) in dataframe 2.

Result >>> 
0E =14; 
D7 = 215;
Formula for Battery voltage = (14 * 256 + 215) / 10.0 = 379.9V  

Now the problem is the parser will crash (possible memory leak) after 10-15seconds, until the crash, the parser works great.

Maybe @PowerBroker2 can help both of us, with this parser :) .

[image]

cyberelectronics, 2020-10-23:
Fortunately, Matthew, who wrote the SafeString library, helped me with this issue. 
Just finished a live test, while charging the car for 50 minutes. 
No reboot or WiFi and BT connection issues at all.
OBD BT queried once/sec and the datas (in multiple groups) was sent every 2 seconds to Blynk.
So here is the working parser for OBD protocol 6 with multiple dataframes (from 0: 1: 2: to 8: ). 
I hope this will help others (maybe @PowerBroker2 will add this to a new release)
Thanks guys!

```
#include "SafeString.h"

createSafeString(dataFrame0, 14);   // will also add space for the terminating null =>15 sized array
createSafeString(dataFrame1, 14);
createSafeString(dataFrame2, 14);
createSafeString(dataFrame3, 14);
createSafeString(dataFrame4, 14);
createSafeString(dataFrame5, 14);
createSafeString(dataFrame6, 14);
createSafeString(dataFrame7, 14);
createSafeString(dataFrame8, 14);

void clearData() {
                    dataFrame0.clear();
                    dataFrame1.clear();
                    dataFrame2.clear();
                    dataFrame3.clear();
                    dataFrame4.clear();
                    dataFrame5.clear();
                    dataFrame6.clear();
                    dataFrame7.clear();
                    dataFrame8.clear();
  
}

void addToFrame(int frame, char c) {
	switch(frame) {
		case 0:
			dataFrame0 += c;
			break;
		case 1:
			dataFrame1 += c;
			break;
		case 2:
			dataFrame2 += c;
			break;
		case 3:
			dataFrame3 += c;
			break;
		case 4:
			dataFrame4 += c;
			break;
		case 5:
			dataFrame5 += c;
			break;
		case 6:
			dataFrame6 += c;
			break;
		case 7:
			dataFrame7 += c;
			break;
		case 8:
			dataFrame8 += c;
			break;
	}
}

void frameSubstr(SafeString& frame, int m, int n, SafeString& subStr) {
    frame.substring(subStr,m,n);   // SafeString substring is inclusive m to n
}

int convertToInt(SafeString& dataFrame, size_t m, size_t n) {
  // define a local SafeString on the stack for this method
    createSafeString(hexSubString, 14);  // allow for taking entire frame as a substring 
    frameSubstr(dataFrame, m, n, hexSubString);
    return (int)strtol(hexSubString.c_str(), NULL, 16);
}

void parse(char *raw) {
	int frame = -1;
	
	int len = strlen(raw);
	
	for(int i=0; i<len; i++) {
		
		if(raw[i+1] == ':') { //start frame
			frame = (int) raw[i] - '0';
			continue;
		}
		
		if(raw[i] == ':') {
			continue;
		}
		
		if(frame == -1) {
			continue;
		}
		
		if(raw[i] == '>') {
			frame = -1;
			continue;
		}
		
		addToFrame(frame, raw[i]);
	}
	
}

void read_rawdata(){
                       // move data from OBD to Rawdata array
                       char rawData[myELM327.recBytes];
                       int n = 0;
                                                                            
                       DEBUG_PORT.print("Payload received: ");
                       
                       for (int i=0; i<myELM327.recBytes; i++) {
                                                                  rawData[n++] = myELM327.payload[i];  
                                                                  DEBUG_PORT.print(myELM327.payload[i]); // Serial print OBD Rawdata
                                                               }
                       DEBUG_PORT.println(); 
                                                                        
                       parse(rawData);  // parse data received from OBD
                        
  }

// loop 
void loop(){
             // insert timer/delay (ex. 1sec)
             ...
             myELM327.sendCommand("AT SH 7E4");       // Set Header BMS 
                                                     
             if (myELM327.queryPID("220101")) {      // BMS PID = hex 22 0101 => dec 34, 257
         
                           read_rawdata(); 
                           BATTv = ((convertToInt(dataFrame2, 4, 5)<<8) + convertToInt(dataFrame2, 6, 7))/10;
                           DEBUG_PORT.print("Main Battery Voltage (V): ");
                           DEBUG_PORT.println(BATTv);
                           clearData();
                        }                                                                                   
    }                                                                                                       
```
[image]
[image]
[image]
[image]
[image]
[image]

PowerBroker2, 2020-10-23:
This is dope! I'll take a closer look later and see if/how I can incorporate it.

However, @cyberelectronics, feel free to fork and PR if you want to do it yourself. Either way, thanks!

cyberelectronics, 2020-10-24:
Thanks, but I don't know how to use github :D .
Also if you can, please add/modify the code for other cases, where the dataframe delimiters are missing (no 0: , when all datas are in a single frame package) like in @KoYoT77 example.
Ex. PID >223275 
Response: 62 32 75 4E
Skip any 7F... bytes at the beginning sent by OBD adapter, and Search/Compare for the first byte of PID 22hex + 40hex = 623275hex >>> 4E..... hex data bytes (dataframe).

KoYoT77, 2020-10-24:
Gents, 
With Jaukb's help - our sketch works.
The issue was in this line of code:
`sprintf(command, SET_HEADER, "7E0");`
Elmduino.h const as below:
`const char * const SET_HEADER                 = "AT SH";    // OBD`
has been edited to this line
`const char * const SET_HEADER                 = "AT SH %s";    // OBD`
finally we have received proper data, same as in the App DPM Monitor.

Thanks to both of you and Jakub too!

cyberelectronics, 2020-11-09:
Thanks guys for your help/library!
Here is the almost finished OBD Google TTS project :) 
https://youtu.be/mUl9IHsN-ao

Jastreb07, 2020-11-09:
Thanks guys, its working now.
I have programmed a Dash Cluster 
The result of my work: https://vm.tiktok.com/ZS4XktLp/

closed 2021-05-22 by PowerBroker2

nices0325, 2021-06-21:
what should I do I wanna use service22 and service01 both?
should I set header to 7E4 to get bms system data and set filter off to get data from service01 and set header to 7E4 again in loop??