PowerBroker2/ELMduino #45: Received:

beeedee, 2020-10-05:
Hi. i am using elm327 wifi and esp32

My code: 
```
#include <WiFi.h>
#include "ELMduino.h"

const char* ssid = "WiFi-OBDII";
const char* password = "your-password";

//IP Adress of your ELM327 Dongle
IPAddress server(192, 168, 0, 10);
WiFiClient client;
ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  Serial.begin(115200);

  // Connecting to ELM327 WiFi
  Serial.print("Connecting to ");
  Serial.println(ssid);

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid);
  // WiFi.begin(ssid, password); //Use this line if your ELM327 has a password protected WiFi

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("Connected to Wifi");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  if (client.connect(server, 35000))
    Serial.println("connected");
  else
  {
    Serial.println("connection failed");
    while(1);
  }

  myELM327.begin(client,5);
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else{
    printError();
  }
}

void printError()
{
  Serial.print("Received: ");
  for (byte i = 0; i < myELM327.recBytes; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();
  
  if (myELM327.status == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.status == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.status == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.status == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(100);
}
```
My Output:

```
Received:
Received:
Received:
Received:
Received:
```
I guess it doesn't meet the following condition
```if` (myELM327.status == ELM_SUCCESS){}```

*How can I list if using custom pid? I think the "at" commands are not valid in the wireless network on elm327.Is there an android app where I can list pids?
*Most importantly, could this problem be caused by custom pid?

PowerBroker2, 2020-10-06:
The second argument to `begin()` is a char, not an int. Try this instead: `myELM327.begin(client, '5');` or simply `myELM327.begin(client);`

beeedee, 2020-10-06:
Oh, sorry. 
Default char '0' not working for my 1996 tigra.
I will try again 😀

beeedee, 2020-10-08:
[image]

[image]

Hi, connection is ok. at least it looks that way.
my new problem rpm is always 0. Should 010C contain a header? And does the parsing of the incoming answers change in case they contain header? I want to send 2 screenshot to help you

[image]
[image]
[image]

beeedee, 2020-10-08:
I accidentally closed

beeedee, 2020-10-08:
Finally, the great idea came to my mind. I looked at the Car Scanner's logs.
I saw only 2101 pid with 104 characters of data being pulled.

Ex. Outputs. 
`>2101
B0F111610103050520034020659069563F6789227E0B00000000260D004549225A64C2670A80003D8E0000838C88107E00000076

>2101
B0F111610103050520034020509069563F678E227E0B00000000260D004F3C245A64C25E0A80003E8F0000838C88107E0000005E

>2101
B0F111610103050520034020489069563F6791227E0B00000000260D005434255A64C2580A80003B8F0000838C88107E0000004E

>2101
B0F1116101030505200340204A8E69563F678D337E0B0000000035220057302D5A64C2510A80003A8F0000838C88107E0000007E

>2101
B0F1116101030505200340205A8E69563F6796177E0C000000001D00006831495A64BB8012790056910000838C88107E000000B7

>2101
B0F111610103050520034020498F69563F676F177E0B000000001C000053383C5A64C17F127900058F0000838C88107E00000015

>2101
B0F111610103050520034020699169563F676D177E0A000000001C00003A54345A64C37D127900008D0000838C88107E00000028

>2101
B0F11161010305052003402089926A563E677A357E0A000000003624003961365A64C4760A8000298D0000838C88107E000000E2

>2101
B0F11161010305052003402065906A563E6776187E0B000000001D0100533D3D5A64C080127900548F0000838C88107E00000091

>2101
B0F1116101030505200340206A916A563E676D177E0A000000001C00003955395A64C47E127900038D0000838C88107E00000033

>2101
B0F111610103050520034020758E6A563E6773177E0A000000001C0000335D335A64C57D127900018C0000838C88107E0000003A

>2101
B0F1116101030505200340207C906A563E6775177E0A000000001C00003360315A64C57C127900008C0000838C88107E00000044
`

and an information I found in my research;

" Opel z16XE Multec-S Desire Rölanti Motor Hızı "," DesIES "," 2101 ", "AJ * 3187/255", "0", "3187", "rpm", "8011F1"
"Opel z16XE Multec-S Motor Hızı", "Motor Hızı", "2101", "(AG * 256 + AH) / 4 "," 0 "," 65536 "," rpm "," 8011F1 ", "2101", "(AG * 256 + AH) / 4", "0", "65536", "rpm", "8011F1", "2101", "(AG * 256 + AH) / 4", "0", "65536", "rpm", "8011F1"

but I could not give a meaning, I am waiting for your help to interpret. I continue to research too.

PowerBroker2, 2020-10-09:
I'm having a very hard time trying to figure out what you're trying to do. Please explain exactly what you're trying to do. Please post your code in tags (no screenshots) and what exactly is working and what isn't. It's entirely possible your car doesn't support the same PID structure as true OBDII ports because of how old it is.

beeedee, 2020-10-09:
All I want from this device is to read the rpm. But 010C does not work. With 2101, all values ​​come at the same time. I managed to link your library to ecu. but I can't read rpm. You may have an idea as you are working on this library. I want to benefit from your information.

I looked at the logs of the "Car Scanner Pro" application.
010C pids not included in the log. All sensors seem to work with 2101 pid.

I have a sample 2101 output here.
`
B0F1116101030505200340207C906A563E6775177E0A000000001C00003360315A64C57C127900008C0000838C88107E00000044
`
The above output contains 38 sensor information along with rpm bytes.It is very difficult to understand which byte is where. I thought you might have encountered this because you are working on this library.

now do you have any idea which bytes are rpm.

PowerBroker2, 2020-10-09:
Ok, I see. Looks like your car predates the PID specification the library was written for and that's why you aren't getting your values. Also, the lib only supports PID responses that are no more than 64 bytes of integer data (not necessarily 64 ASCII bytes), so your 2101 query wouldn't work regardless. Sorry about that

closed 2020-10-09 by PowerBroker2