PowerBroker2/ELMduino #46: Couldn't connect to OBD scanner - Phase 2

adamkovari, 2020-11-20:
**Describe the bug**
Hi, I'm trying to build a bluetooth connection between ELM327( like this 
[image] )
 and an ESP32 Wroom.
I'm using the bluetooth_serial example, and a tried everything to initialize the ELM327 object myELM327.initialize('5'), myELM327.begin(ELM_PORT), myELM327.begin(ELM_PORT, ISO_14230_FAST_INIT), myELM327.begin(ELM_PORT, ISO_14230_FAST_INIT, payloadLen) 
BUT everytime, I got this on serial monitor : "Couldn't connect to OBD scanner - Phase 2" 
@PowerBroker2 Can you please help, what am I doing wrong?

PowerBroker2, 2020-11-20:
Please post your entire code (in tags) and a description of your physical setup. Also, please ensure all other bluetooth devices are either disabled or turned off, ensure your car is on and running, and be sure to try all other baud rates

adamkovari, 2020-11-20:
Hi, here is my code (esp32_bluetooth_serial):

`#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t rpm = 0;

uint16_t BaudRate=115200;
//uint16_t BaudRate=38400;
uint16_t payloadLen=40;

/*
 *  * Protocol - Description
  * 0        - Automatic
  * 1        - SAE J1850 PWM (41.6 kbaud)
  * 2        - SAE J1850 PWM (10.4 kbaud)
  * 3        - ISO 9141-2 (5 baud init)
  * 4        - ISO 14230-4 KWP (5 baud init)
  * 5        - ISO 14230-4 KWP (fast init) -- EZ KELL
  * 6        - ISO 15765-4 CAN (11 bit ID, 500 kbaud)
  * 7        - ISO 15765-4 CAN (29 bit ID, 500 kbaud)
  * 8        - ISO 15765-4 CAN (11 bit ID, 250 kbaud)
  * 9        - ISO 15765-4 CAN (29 bit ID, 250 kbaud)
  * A        - SAE J1939 CAN (29 bit ID, 250* kbaud)
  * B        - User1 CAN (11* bit ID, 125* kbaud)
  * C        - User2 CAN (11* bit ID, 50* kbaud)
 */

char protocol='5';

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(BaudRate);
  SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect("OBDII")) //or "00-00-00-11-11-11" or "00:00:00:11:11:11"
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }
    

  //myELM327.initializeELM(protocol);
  if (!myELM327.begin(ELM_PORT, ISO_14320_FAST_INIT /*or protocol */, payloadLen ))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else
    printError();
}

void printError()
{
  Serial.print("Received: ");
  for (byte i = 0; i < myELM327.recBytes; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();
  
  if (myELM327.status == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.status == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.status == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.status == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(100);
}`

I plugged the ELM327 in my car OBDII connector, and turned on the engine
I downloaded this code above on a esp32 wroom, and then press the en button on esp32 to reset the microcontroller.
I turned off all my bluetooth devices, but i'm parking on the street, so i can't be sure, there isn't available other bluetooth devices

I watch the esp32 microcontroller through my laptop's serial monitor, and the result is: "Couldn't connect to OBD scanner - Phase 2"

I tried with 115200 and 38400 baudrates too.
Thank you for your help!

PowerBroker2, 2020-11-20:
Try commenting out `SerialBT.setPin("1234");` and using protocol `0`

adamkovari, 2020-11-20:
I already tried these I'am afraid :(

PowerBroker2, 2020-11-20:
What's your make/model of your car? Also, can you confirm you've taken care of the following?:
> Also, please ensure all other bluetooth devices are either disabled or turned off, ensure your car is on and running, and be sure to try all other baud rates

adamkovari, 2020-11-20:
peugeot 206 sw - 1.6 HDI diesel - 2005

Alright, tomorrow i'll try find a place, where no other bluetooth devices are available.
Yes the engine was running, while i'm trying to get the data.
I tried only 115200 and 38400 baudrates, what else are you provide?

PowerBroker2, 2020-11-20:
> peugeot 206 sw cool look - 1.6 HDI diesel - 2005

I can't guarantee this lib will work for any vehicles produced earlier than 2008

> I tried only 115200 and 38400 baudrates, what else are you provide?

All other bauds

adamkovari, 2020-11-21:
What is the difference between the cars produced before and after 2008, if they use the same protocol? Can you give me a hint? I want to succed this project.

adamkovari, 2020-11-21:
what I can't understand, that if I run the esp32_test.ino file (from the examples),
i get on the serial monitor: "Couldn't connect to OBD scanner", soo ELM_PORT.connect("OBII") return with false

```c
if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);
  }
```

BUT if i run the ESP32_Bluetooth_serial.ino

i get: "Couldn't connect to OBD scanner - Phase 2"
which is after that ELM_PORT.connect("OBDII") returns with true
```c
 if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }
```

what could be different? (same baudrate115200, engine turned on, all bluetooth are off)

PowerBroker2, 2020-11-22:
> What is the difference between the cars produced before and after 2008, if they use the same protocol? Can you give me a hint? I want to succed this project.

From what I've read, 2008 was the first year the OBDII protocol was made standard on all cars in the US/EU (I do believe some locations/manufacturers adopted it earlier, though). It's because of this that I can't guarantee the lib will work with earlier cars.

The behavior in your last post is definitely odd. Are you sure the BT name of your ELM327 is "OBDII"? Can you check the name with your phone/laptop?

Last ditch effort: Try this sketch:
```C++
  
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

#define DEBUG_PORT Serial
#define ELM_PORT   SerialBT

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  DEBUG_PORT.begin(115200);
  //ELM_PORT.setPin("1234");
  ELM_PORT.begin("ESP32test", true);

  DEBUG_PORT.println("Attempting to connect to ELM327...");

  ELM_PORT.connect("OBDII");

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}

void loop()
{
  if(DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();

    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }

  if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if(c == '>')
      DEBUG_PORT.println();

    DEBUG_PORT.write(c);
  }
}
```

And input the following into the serial monitor:
```
AT Z
AT E0
AT S0
AT AL
AT TP A0
```

**Lastly, post exactly what the Arduino responds with and we can see what's going on.**

adamkovari, 2020-11-22:
Atleast i get this answer with your last sent code, but I'm not sure I wrote the code in the serial monitor precisely.
This is the right way?( i set line ending to carriage return )
AT(space)Z(enter)
AT(space)E0(enter)....and so on?

[image]

Otherwise I found this table for my car type:
Do you know what do this codes mean?
[image]

adamkovari, 2020-11-22:
A new twist today, I downloaded the OBD Auto Doctor software and with my laptop connected to the elm327, it works perfectly.
So i don't really understand, what could be wrong with the esp32.

[image].

I attached the whole obd commands/answer log, but from that it's clear, that the issue is with esp32... :(

[obd_log.txt](https://github.com/PowerBroker2/ELMduino/files/5579944/obd_log.txt)

PowerBroker2, 2020-11-22:
> I attached the whole obd commands/answer log, but from that it's clear, that the issue is with esp32... :(

This might be because the ELM327's Bluetooth device name is not "OBDII" - can you check this?

adamkovari, 2020-11-22:
On laptop and on mobile the ELM327's name is OBDII as well, and I paired with that, before i started OBD Auto Doctor.
Tomorrow i'll check with MAC_Address, and pray...

PowerBroker2, 2020-11-22:
Are you sure your ELM327 is classic bluetooth and not BLE?

adamkovari, 2020-11-22:
Well I'm not, I would never say elm327 uses BLE. I check this! However my laptop is kinda old, I'm not certain it supports BLE

adamkovari, 2020-11-23:
Hi, today it happened the breakthrough! It works.
What I had to do, is use from examples the bt_remove_paired_devices sketch and change #define REMOVE_BONDED_DEVICES 0 to 1.
Thank you for your help!

closed 2020-11-23 by PowerBroker2

rafargp, 2020-12-30:
@adamkovari, Can you post your Code that Worked? I have the same problem but your solution didn't worked for me!

Thanks