PowerBroker2/ELMduino #49: Hi PowerBroker

nices0325, 2020-12-15:
**Describe the bug**
serial monitor keep showing couldn't connect to OBD scanner
**To Reproduce**
Steps to reproduce the behavior.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Equipment**

I used elm327 v1.5. Initially i removed elm327 bluetooth chip and slodered rx,tx,gnd,vcc and connect to arduino uno board.
I used software_serial_test examples to my arduino and serial monitor just keep showing that can't connect to OBD scanner. what'd be the problem.... sorry for my bad english :(

**Code**

serial monitor
Attempting to connect to ELM327...
Couldn't connect to OBD scanner

**Wiring**
Actually I m a damn fucking beginner of arduino and i don' know a shit about what the c language is and so on... I just started to learn about it.. I spent about a weak to two weaks for this problem...
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Additional Context**
PLEASE HELP ME !!!!
**Troubleshooting Connection Issues**
- Try 38400 baud instead of 115200 I TRTIED ALL THE BAUD
- If using an HC-05 for communication, [ensure it's configured properly](https://forum.arduino.cc/index.php?topic=600014.msg4159554#msg4159554)
- Ensure the car is on and running during troubleshooting
- I already have connected to the car but it doesn't work bro..

PowerBroker2, 2020-12-16:
Can you post the exact code you're using (in tags)? Are you testing with the ELM327 connected to the car while the car is on and running? Are you sure you soldered the correct pads?

Try loading a serial pass-through sketch and see if the ELM327 responds to AT commands through the serial monitor. Something like this:

```C++
#include <SoftwareSerial.h>

SoftwareSerial mySerial(2, 3); // RX, TX

void setup()
{
  // Open serial communications and wait for port to open:
  Serial.begin(115200);

  // set the data rate for the SoftwareSerial port
  mySerial.begin(115200);
}

void loop() // run over and over
{
  if (mySerial.available())
    Serial.write(mySerial.read());
  if (Serial.available())
    mySerial.write(Serial.read());
}
```

nices0325, 2020-12-17:
how can i know that it respond to AT commands??? just type at command and push the send button on arduino serial monitor?? or what should i do?

PowerBroker2, 2020-12-17:
Yeah, load the sketch, set the serial monitor to `Carriage Return` line endings, type `ATZ`, and then see if you get something like this:

```
>ATZ
ELM327/ELM-USB v1.0 (c) SECONS Ltd.
```

nices0325, 2020-12-18:
"⸮⸮"
This word shows up... not "ELM327/ELM-USB v1.0 (c) SECONS Ltd."
Is this sure that i soldered the wire wrong way or the elm327 is defective???

would it bo more appropriate to get the obd information from the elm327 bluetooth chip to arduino? then should i install bluetooth receive module like esp32?

nices0325, 2020-12-18:
Actually my goal is to get the OBD informations like rpm, velocity, and abs wheel speed sensor so on... Collect all the sensor values and calculate it so that I Can  get a one value that represent the how well the driver is driving well like "safety driving score"
however, it' s really hard to connect to elm327 wiht arduino at first time...

nices0325, 2020-12-18:
Mr . I haver one more question !

[image]

As I change the baud rate to 38400 and change the programmer: to arduino as isp as you see on the screenshot, finally it worked... I think the programmer set was wrong... what is the difference between the AvrISP mkll and Arduino as isp?? my first set was AvrISP mkll and it didn't work..

nices0325, 2020-12-18:
mr... another problem comes out...
Finally elm327 connected to arduino uno... and I connect to the car but the serial monitor just keep showing elm time out..what's the meaning of the elm327-timeout ???  plz check the screenshout plz..
[image]

PowerBroker2, 2020-12-18:
Anything on the same line after `Received: ` is what exactly was received by the Arduino from the ELM327 (largely unparsed apart from spaces). `ELM_TIMEOUT` means it didn't receive a response from the ELM327. Are you testing with the car on and running?

nices0325, 2020-12-19:
yeah.. I'm sure .. or can you give me one more example showing engine loas plz?? it could be really helpful mr powerbroker2!!!

PowerBroker2, 2020-12-19:
To simplify things, go back to the serial "passthrough" sketch (the one you used to test the `ATZ` command. This time, instead of just `ATZ`, enter each of these commands and let me know what the ELM327 responds with:

```
AT Z
AT E0
AT S0
AT AL
AT TP A0
```

nices0325, 2020-12-21:
Finally I could make it!! I do appreciate a lot!!  
Actually now I'm studying C++ for my first time... It's really hard for me to understand, cause I'm not the major of computer programming... would you mind do me a favor?
Could you get me a simple example code of the ENGINE_LOAD ??? I'm studying a lot, bur it's a little bit challenging for me.
appreciating one more time, I'll expect your reply!! TAHNK YOU sir~~

PowerBroker2, 2020-12-21:
[This thread](https://github.com/PowerBroker2/ELMduino/issues/44) goes into detail on how to query pids other than mph, kph, or rpm. You'll just have to switch out the service number and PID for your `queryPID()` calls to `SERVICE_01` and `ENGINE_LOAD` respectively.

PowerBroker2, 2020-12-21:
Curious, how did you get it to work?

nices0325, 2020-12-22:
Do appreciate your kind inform :) 
Actually I reset the arduino programmer settings to Arduino as ISP , before then it was AVSRMKll This one. I don't have any idea what 's this.. so finally I could handle this

closed 2020-12-22 by PowerBroker2

siwar-w, 2022-05-05:
Hello, I'm trying to make a serial connexion between my esp32 sim800L and elm327 so I removed the wifi and use instead RX et TX i fowlowed the code you post but i get response at ATZ command "?" and i don't know why i'm making sure i wire it correctly can you help me