PowerBroker2/ELMduino #50: hi mr. powerbroker2 . question for the coding..

nices0325, 2020-12-24:
hello mr.

nowadays I'm really struggling to make a code 
my goal is to get some PID responses to collect the information.
Actually I'm not a programmer, so it's damn fuckin super hard for me to coding..

first of all I added some PIDs in .cpp file like this 
float ELM327::engineload()
{
           if( status == ELM_SUCCESS)
                   return findResponse();
           
             return ELM_GENERAL_ERROR;
}

I also add float engineload(); on my .h file

then this is my sketch
#include <SoftwareSerial.h>
#include "ELMduino.h"

SoftwareSerial mySerial(2, 3); // RX, TX
#define ELM_PORT mySerial

ELM327 myELM327;

uint32_t rpm = 0;
uint32_t kph = 0;
uint32_t engineload= 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  Serial.begin(38400);
  ELM_PORT.begin(38400);

  Serial.println("Attempting to connect to ELM327...");

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();
  int32_t tempVEHICLE_SPEED = myELM327.kph();
  float tempENGINELOAD = myELM327.engineload();

//////////////////rpm///////////////////      
  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
///////////////////////////////////////////
///////////////////////kph///////////////////
    if (myELM327.status == ELM_SUCCESS)
    {
      kph = (int32_t)tempVEHICLE_SPEED;
      Serial.print(kph); Serial.println("km/h");
    }

///////////////////////////////////////////////
///////////////////engineload/////////////////
if (myELM327.status == ELM_SUCCESS)
{
  engineload = (uint32_t)tempENGINELOAD;
Serial.print("EngineLoad: "); Serial.println(engineload);
}
////////////////////////////////////////////    
    else
    printError();

    
}

void printError()
{
  Serial.print("Received: ");
  for (byte i = 0; i < myELM327.recBytes; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();
  
  if (myELM327.status == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.status == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.status == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.status == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(100);
}

Acutually It's too hard to understand when to use "float" "uint32_t" "int8_t" like this kind of things. Absolutely I know the concept of these things since I ve been studying.. however, when I get in to the sketch or cpp I BECOME A FUCKING IDIOT.. plz help me EXPERT powerbroker2... 
I will expect your generous reply  thank you~

nices0325, 2020-12-30:
not any advices ??? plz bro

PowerBroker2, 2021-01-04:
Please research basic C++ tutorials. If you still can't figure it out, ask on Stack Overflow or the Arduino forum

closed 2021-01-04 by PowerBroker2