PowerBroker2/ELMduino #53: printerror speed

nices0325, 2021-01-10:
hi mr.powerbroker2

While I'm coding my project, suddenly the printerror interval increased compared to your example code "softwareserial_test"
what would be the problem....

and asking for another little help.... I want  my green and red LED to indicate current status e.g. ELM327 CONNECTED, VALUES ARE RECEIVING, PRINTERROR.... It's a little bit hard for me.. actually i'm not a programmer.. plz understand

#include <SoftwareSerial.h>
#include "ELMduino.h"

SoftwareSerial mySerial(2, 3); // RX, TX
#define ELM_PORT mySerial
#define BUZZER 10

ELM327 myELM327;

int green = 8;
int red = 9;

uint32_t rpm = 0;
uint32_t kph = 0;
uint32_t engineload = 0;
uint32_t enginetemp = 0;
uint32_t throttleposition = 0;
uint32_t intakeairtemp = 0;
int32_t fueltype = 0;

const int beepFrequencyS1 = 1500; //1000Hz, 연결성공시 부저 주파수1
const int beepFrequencyS2 = 2500; //1000Hz, 연결성공시 부저 주파수2
const int beepFrequencyF = 750; //500Hz, 연결실패시 부저 주파수
const int beepDurationS = 250; //1.0sec ,성공시 지속시간, (31~65535 Hz)
const int beepDurationF = 250; //0.25sec ,실패시 지속시간, 

unsigned long pre_time = 0;
unsigned long cur_time = 0;
int ledState = LOW;

void setup()
{
    pinMode(green, OUTPUT);
    pinMode(red, OUTPUT);
    pre_time = millis();
#if LED_BUILTIN
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);
#endif

    Serial.begin(38400);
    ELM_PORT.begin(38400);

    Serial.println("Attempting to connect to ELM327...");

    if (!myELM327.begin(ELM_PORT))
    {
        Serial.println("Couldn't connect to OBD scanner");
        do {
            digitalWrite(red, HIGH);
            digitalWrite(green, LOW);
            tone(BUZZER, beepFrequencyF, beepDurationF);
            delay(850); //ELM327 연결 실패시 0.85초 간격으로 Fail 부저 울림
        } while (1);
    }

    Serial.println("Connected to ELM327");
    digitalWrite(green, HIGH);
    digitalWrite(red, LOW);
    tone(BUZZER, beepFrequencyS1, beepDurationS);
    delay(600);
    tone(BUZZER, beepFrequencyS1, beepDurationS);
    delay(600);
    tone(BUZZER, beepFrequencyS1, beepDurationS);
    delay(600);
    tone(BUZZER, beepFrequencyS2, beepDurationS);
    delay(600);
    noTone(10);

    //ELM327 연결 성공시 Success 부저 4회 울림, led green ON

}

void loop()
{
    cur_time = millis();
    float tempRPM = myELM327.rpm();
    int32_t tempVEHICLE_SPEED = myELM327.kph();
    float tempENGINE_LOAD = myELM327.engineload();
    float tempENGINE_COOLANT_TEMP = myELM327.enginetemp();
    float tempTHROTTLE_POSITION = myELM327.throttleposition();
    float tempINTAKE_AIR_TEMP = myELM327.intakeairtemp();
    float tempFUEL_TYPE = myELM327.fueltype();

    digitalWrite(red, LOW);

    if (cur_time = pre_time >= 500)
    {
        pre_time = cur_time;
        if (ledState == LOW) {
            ledState = HIGH;
        }
        else {
            ledState = LOW;
        }
        digitalWrite(green, ledState);
        
    }
    if (myELM327.status == ELM_SUCCESS)
    {
        rpm = (uint32_t)tempRPM;
        Serial.print("RPM : "); Serial.println(rpm);

        kph = (int32_t)tempVEHICLE_SPEED;
        Serial.print(kph); Serial.println("km/h");

        engineload = (uint32_t)tempENGINE_LOAD;
        Serial.print("EngineLoad : "); Serial.print(engineload); Serial.println("%");

        enginetemp = (uint32_t)tempENGINE_COOLANT_TEMP;
        Serial.print("EngineCoolantTemp : "); Serial.print(enginetemp); Serial.println("˚C");

        throttleposition = (uint32_t)tempTHROTTLE_POSITION;
        Serial.print("ThrottlePosition : "); Serial.print(throttleposition); Serial.println("%");

        intakeairtemp = (uint32_t)tempINTAKE_AIR_TEMP;
        Serial.print("IntakeAirTemp : "); Serial.print(intakeairtemp); Serial.println("˚C");

        fueltype = (int32_t)tempFUEL_TYPE;
        Serial.print("FuelType : "); Serial.print(fueltype);

        if (fueltype == 0)
            Serial.println("Not Available");
        else if (fueltype == 1)
            Serial.println("Gasoline");
        else if (fueltype == 2)
            Serial.println("Methanol");
        else if (fueltype == 3)
            Serial.println("Ethanol");
        else if (fueltype == 4)
            Serial.println("Disel");
        else if (fueltype == 5)
            Serial.println("LPG");
        else if (fueltype == 6)
            Serial.println("CNG");
        else if (fueltype == 7)
            Serial.println("Propane");

            tone(BUZZER, beepFrequencyS2, beepDurationS);

        delay(500);

    }
    else
    printError();
}

void printError()
{
  digitalWrite(green, LOW);
    Serial.print("Received: ");
    for (byte i = 0; i < myELM327.recBytes; i++)
        Serial.write(myELM327.payload[i]);
    Serial.println();

    if (myELM327.status == ELM_SUCCESS)
        Serial.println(F("\tELM_SUCCESS"));
    else if (myELM327.status == ELM_NO_RESPONSE)
        Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
    else if (myELM327.status == ELM_BUFFER_OVERFLOW)
        Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
    else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
        Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
    else if (myELM327.status == ELM_NO_DATA)
        Serial.println(F("\tERROR: ELM_NO_DATA"));
    else if (myELM327.status == ELM_STOPPED)
        Serial.println(F("\tERROR: ELM_STOPPED"));
    else if (myELM327.status == ELM_TIMEOUT)
        Serial.println(F("\tERROR: ELM_TIMEOUT"));
    else if (myELM327.status == ELM_TIMEOUT)
        Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));
        
        tone(BUZZER, beepFrequencyF, beepDurationF);

    delay(10);
}

nices0325, 2021-01-10:
actually it's a question category. sorry for not knowing  how to modify the categories...

PowerBroker2, 2021-01-11:
The no data error means the ELM327 is not getting any data from your car for some reason. Are you testing with your car on and running? Also, please format your code properly

nices0325, 2021-01-11:
sure. Car was running .. would it be the porblem of the ELM327 itself? should i buy one more?

nices0325, 2021-01-11:
Is there any wrong with my code ?

PowerBroker2, 2021-01-11:
it seems more like either a config issue, or your car just doesn't want to send the ELM data. What's your car's make/model/year?

Also, I won't look at your code until you fix its formatting.

nices0325, 2021-01-12:
kia forte 2019..
I tried to buy another elm327 and it works.. I think ELM327 have various issues connecting with arduino ... 

anyway, I have checked the  values from my 
[image]
rpm, vehicle speed and the other values seem to be quite correct but engine load and throttleposition seem to be wrong.. As I fully accelerate to 6000 or 7000 rpm, engine load goes up to as far 18% and throttleposition up to 20%. It doesn't seem to be normal. What would be the problem..

PowerBroker2, 2021-01-12:
The value returned by the engine load pid needs to be divided by 2.55 to be correct. See [here](https://en.wikipedia.org/wiki/OBD-II_PIDs)

closed 2021-01-12 by PowerBroker2

nices0325, 2021-01-12:
sure, I already checked that .. plz check below
float ELM327::rpm()
{
	if (queryPID(SERVICE_01, ENGINE_RPM))
		return (findResponse() / 4.0);

	return ELM_GENERAL_ERROR;
}

float ELM327::engineload()
{
	if (queryPID(SERVICE_01, ENGINE_LOAD))
		return (findResponse() / 2.55);

	return ELM_GENERAL_ERROR;
}

float ELM327::enginetemp()
{
	if (queryPID(SERVICE_01, ENGINE_COOLANT_TEMP))
		return (findResponse() - 40.0);

	return ELM_GENERAL_ERROR;
}

float ELM327::throttleposition()
{
	if (queryPID(SERVICE_01, THROTTLE_POSITION))
		return (findResponse() / 2.55);
	
	return ELM_GENERAL_ERROR;
}

float ELM327::intakeairtemp()
{
	if (queryPID(SERVICE_01, INTAKE_AIR_TEMP))
		return (uint32_t)findResponse();

	return ELM_GENERAL_ERROR;
}

float ELM327::fueltype()

{
	if (queryPID(SERVICE_01, FUEL_TYPE))
		return (int32_t)findResponse();

	return ELM_GENERAL_ERROR;
}

PowerBroker2, 2021-01-12:
`intakeairtemp()` needs a -40 bias, but other than that, everything else is correct

PowerBroker2, 2021-01-12:
Shot in the dark, but I added a ton of new functions for PID processing (basically everything except oxygen sensor stuff). Give [release 2.4.0](https://github.com/PowerBroker2/ELMduino/releases/tag/2.4.0) a shot and see if it helps

nices0325, 2021-01-13:
omg... It's a real dope shit for me... thanks alot bro

PowerBroker2, 2021-01-13:
You're right - I made a typo. It'll be fixed in the next version

nices0325, 2021-01-13:
actually I'm wondering that how could you calculate A and B byte values only using 'findResponse' although it is up to 4byte pid
so that for beginners like me, it's a little bit hard to find out what's value A and B... sorry for ignorance.. 
my final question is, even if the final calculation is using A,B values , how to calculate with only 'findResponse' ... It's too hard for me to get it. I'm sorry