PowerBroker2/ELMduino #54: getResponse and calculate resonseByte A and B

nices0325, 2021-01-13:
Hello powerbroker2
I'm really sorry for bothering you for keep asking questions...
I do appreciate that

Actually It is really easy to use 'findResponse' to get the values
To get more values from PID I should find out some splited values like 
Distance traveled since codes cleared (256*A+B)

I have already checked the close issues about the splited values you wrote... but as mentioned, I'm not a programmer

I added to this my cpp file first.

uint16_t ELM327::runtime()
{
	if (queryPID(SERVICE_01, RUN_TIME_SINCE_ENGINE_START))

		return (uint16_t)findResponse();

	return ELM_GENERAL_ERROR;
}
//then can you tell me how to use responseByte_1 and responseByte_0 using 'findResponse' to calculate the 'runtime' value?

this is my sketch below
uint16_t runtime = 0;

uint16_t tempRUN_TIME_SINCE_ENGINE_START = myELM327.runtime();

    byte B = myELM327.responseByte_0;
    byte A = myELM327.responseByte_1; // I don't have any idea how to use this...

  runtime = (uint16_t)tempRUN_TIME_SINCE_ENGINE_START;
        Serial.print("Engine Runtime : "); Serial.print(runtime); Serial.println("seconds"); // 256*A+B is the correct runtime value

I'm really sorry for my ignorance.. plz give me a little help plz... as I metioned I'm not a programmer... sorry again.

nices0325, 2021-01-13:
It is a question category, sorry for not to know how to change it to question category

PowerBroker2, 2021-01-13:
The new built-in functions should cover this one. Otherwise, do you have a specific question?

nices0325, 2021-01-13:
actually I'm wondering that how could you calculate A and B byte values only using 'findResponse' although it is up to 4byte pid
so that for beginners like me, it's a little bit hard to find out what's value A and B... sorry for ignorance..
my final question is, even if the final calculation is using A,B values , how to calculate with only 'findResponse' ... It's too hard for me to get it.

e.g. for get the rpm value the formula is {256A + B} / {4}.. otherwise I can find out in the cpp file that you just divide 'findResponse' by 4.0 .. does the 256A+B automatically calculated by the findResponse function?? for being 
Newbie,, really sorry to keep asking you a lot. thank you

PowerBroker2, 2021-01-13:
In the RPM function, A is the most significant byte (of meaningful bits) and B is the least significant byte of the value returned by `findResponse()`. This means (256*A + B) is already taken care of by `findResponse()`. Note that A * 256 is the same as left shifting A by 8 bits.

```C++
byte A = 25;
byte B = 30;
uint16_t ex1 = 256*A + B;
uint16_t ex2 = (A << 8) + B;
```
In the above snippet, `ex1=ex2`.

Your code in your first post in this issue looked good, you just didn't do anything with A and B once you found them. However, like I said, you don't really need to now that I added a built-in function for this PID

nices0325, 2021-01-13:
ummm.. a liitle bit confusing in my knowledge.. Would it be easier to just understand 'findResponse' automatically find out the required values?Should it be more helpful to understand "uint64_t ELM327::findResponse() " in cpp file that how it functions?

Anyway, I'm gonna study about MSB and LSB and plz let me ask you again in a few hours.. thanks.

closed 2021-01-16 by PowerBroker2

nices0325, 2021-01-24:
sorry for another question mr.POWERBORKER2
Can I use this library with stn1110? I heared that stn1110 totally compatible with elm327
Cause, ELM327 only works at 38400 baudrate over UART serial.. 
would it be okay?