PowerBroker2/ELMduino #55: Can it run on STN1110?

nices0325, 2021-01-24:
Hi. I'm using ELM327 over UART to my arduino..
However, it only works with 38400baud rate. It seems to be too late for my project.
Can I use this STN1110 with this library?
[image]

PowerBroker2, 2021-01-24:
I've never seen anyone test one with my lib, but from the description I think it would be safe to say it is compatible

nices0325, 2021-01-24:
Thanks for your reply!!
else, I have two more question
First of all, I have connected ELM327 with my arduino board over UART.
Once I connect the obd cable to my car, Serial monitor shows ELM_TIMEOUT error. If I plug out the rx tx cable and plug in, It normally works. What would be the problem... I haven't solved this problem...

Second of all this is my informs of my car below
[image]
[image]
As you see the first row, ELM327 Timeout Error occurs and once I plug out and plug in the rx tx cable It works.

I have no idea why the first value isn't normal and some vlaues return to 0 as you see the oilTemp,oilTemp2 values
Actually, oilTemp2 values -  I coded "return (responseByte_1 - 40.0);" in cpp file, as same as "return (findResponse() - 40.0);" - oilTemp.

Thank for your replay again. Thank you

PowerBroker2, 2021-01-25:
You should be using `responseByte_0` since only 1 byte is returned by the ELM327 for that PID

nices0325, 2021-01-27:
I ordered another ELM327 v1.5 . I 'll check this error once again with my new ELM327. Isn't there any way to solve this connection error with AT COMMAND??

PowerBroker2, 2021-01-27:
I really don't know. The ELM327 protocol in general is a pain to deal with and connection with Chinese knock-offs can be inherently spotty

nices0325, 2021-01-28:
can you recommand the real one(ELM327) ? I have no idea what is the real one

nices0325, 2021-01-28:
Any way I Command (AT Z, AT D) to my ELM327 to reset my elm327 module. Is there any recommanded AT COMMAND to get a good response?

PowerBroker2, 2021-01-28:
I'm not sure where you can buy a real one (maybe DigiKey?), but they're going to be a bit more expensive. As for the AT commands, I'm not sure - check out [this](https://www.sparkfun.com/datasheets/Widgets/ELM327_AT_Commands.pdf)

closed 2021-01-30 by PowerBroker2

nices0325, 2021-02-05:
[image]
As you see my data logger sheet , It takes about 1 minute to get the response data from ELM327. Sometimes it becomes timeout for all time. I think there would be someway to handle it with AT command, but I don't really have no idea to use AT command(really need a description of AT command). 

Would it be better to  use another obd device like stn1110 to run my own car data logger?
waiting for your reply, thank you! If you don't  mind, let me show you my SKETCH!!