PowerBroker2/ELMduino #56: Monitor Status result does not seem correct

stumunro74, 2021-01-30:
**Describe the bug**
Compiler warning when enabled.

C:\Users\Stuart\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp: In member function 'bool ELM327::initializeELM(char)':
C:\Users\Stuart\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp:116:5: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
     if (sendCommand(command) == ELM_SUCCESS)
     ^~
C:\Users\Stuart\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp:120:2: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
  return connected;
  ^~~~~~

**To Reproduce**
Enable compiler warnings.

**Expected behavior**
No compiler warnings.

**Code**
<code>
    if (sendCommand(command) == ELM_SUCCESS)
        if (strstr(payload, "OK") != NULL)
            connected = true;

	return connected;
</code>

**Troubleshooting Connection Issues**
- Try 38400 baud instead of 115200
- If using an HC-05 for communication, [ensure it's configured properly](https://forum.arduino.cc/index.php?topic=600014.msg4159554#msg4159554)
- Ensure the car is on and running during troubleshooting

PowerBroker2, 2021-01-30:
If you change the offending lines to the following, do you still get the error?

```C++
    if (sendCommand(command) == ELM_SUCCESS)
        if (strstr(payload, "OK") != NULL)
            connected = true;

   return connected;
```

stumunro74, 2021-01-31:
The issue was with the following line. 
<code>
	return connected;
</code>
Changing the tab to spaces removes the warning.

<code>
    return connected;
</code>

closed 2021-02-02 by PowerBroker2