PowerBroker2/ELMduino #58: WHAT FUNCTION DO I USE TO CHECK EXISTING DTCS CODES?

kassiooliver, 2021-02-05:


PowerBroker2, 2021-02-05:
For an overview of DTCS codes, check [this out](https://www.nyvip.org/PublicSite/OBDII/diagnostic-trouble-codes.html)

While I haven't tested it myself, according to [this](https://mechanics.stackexchange.com/a/8273), the `monitorStatus()` should allow you to begin checking each trouble code. Then, you send (in ASCII) `03` for each code separately. Since this message has no PID, you'll have to send it manually and parse the response manually. I literally just googled all this 5min ago, but once I get some time (could be months), I'll seriously consider adding this functionality into the lib so users won't have to manually do it all...

rudydelorenzo, 2021-02-23:
Having a way to check codes and subsequently clear them would be wonderful.

Thanks for the library!

TerAtoM, 2023-01-15:
Hello. The description of the library say's "Also, you can use ELMduino to view and clear your car's trouble codes". I assume it is implemented now an I am just do dumb to see it. Did anybody has a working example of it? :)

NickEsColR, 2023-05-24:
I have the same question has @TerAtoM. Is already implemented in the library or we still have to manually do it? 
I check the library src and found the freezeDTC() method to use. my question is if that is the one i can use to get them?
Thank you so much @PowerBroker2 for the effort put in develop the library.

PowerBroker2, 2023-05-24:
For now, try the process [in my first comment](https://github.com/PowerBroker2/ELMduino/issues/58#issuecomment-773766025)

PowerBroker2, 2023-12-07:
> I've implemented both blocking and non-blocking variants and I'm not sure which approach is preferred

I'm ok with having both options available - especially if already implemented. As long as the user uses an FSM like in the multi-PID example sketch, there shouldn't be an issue with the non-blocking version, although most will probably want to use the blocking version.

jimwhitelaw, 2023-12-12:
Updating this discussion to note that a new method currentDTCCodes() has been added that accomplishes this. It's not yet in a tagged release and some additional testing seems prudent, but I have had it working with 2 adapters and 2 vehicles. If anyone else can try this out, it would be appreciated. 

@TerAtoM, I have added two example programs to demonstrate checking for MIL (Check Engine Light) status / code count and fetching any current DTCs.

PowerBroker2, 2023-12-21:
This has been implemented on main and will be included in the next tagged version

closed 2023-12-21 by PowerBroker2

Naveen-S06, 2024-01-22:
> For an overview of DTCS codes, check [this out](https://www.nyvip.org/PublicSite/OBDII/diagnostic-trouble-codes.html)
> 
> While I haven't tested it myself, according to [this](https://mechanics.stackexchange.com/a/8273), the `monitorStatus()` should allow you to begin checking each trouble code. Then, you send (in ASCII) `03` to get the information on each code. Since this message has no PID, you'll have to send it manually and parse the response manually. I literally just googled all this 5min ago, but once I get some time (could be months), I'll seriously consider adding this functionality into the lib so users won't have to manually do it all...

Is it possible to visualize the DTCs instead of just seeing if the DTCs is present or not? I want to see the DTCs... the actual trouble code which we see using a scan tool. Is it possible?

Naveen-S06, 2024-01-23:
> > Is it possible to visualize the DTCs instead of just seeing if the DTCs is present or not? I want to see the DTCs... the actual trouble code which we see using a scan tool. Is it possible?
> 
> [ESP32_Check_DTC.ino](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Check_DTC/ESP32_Check_DTC.ino)

Can you help me out on interfacing the ELM327 with Arduino UNO? Can anyone provide me the pin diagrams for it?

jimwhitelaw, 2024-01-23:
Unless you intend to open up your ELM327 and solder connections to it, you’re likely better off using a Bluetooth connection. You will need to add a device like the HC05/HC06 Bluetooth module your arduino board. There’s lots of tutorials about how to do that. Personally, I think it’s a lot easier to use an esp32 board (~$10) that has BT built in.

Naveen-S06, 2024-01-24:
Can someone please provide me a blog or tutorial video for reading the pids or DTCs using elmduino library. I have interfaced arduino with HC05 module and successfully connected with the ELM327 Bluetooth connector but I'm not sure if the arduino is communicating with the elm? Please help

Naveen-S06, 2024-01-24:
@PowerBroker2 Hey I connected Arduino Uno with HC05 Bluetooth module with pins (HC05 - RX, TX, VCC, GND to Uno - 10, 11, 5v, GND) and I've successfully connected ELM327 bluetooth obd2 tool to arduino. I have this ode to read engine rpm but it is not getting connected to the obd2 scanner. please help me out with this
Code:
#include "ELMduino.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(10, 11); // RX, TX

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  #if LED_BUILTIN
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);
  #endif

  Serial.begin(115200);
  mySerial.begin(115200);

  Serial.println("Attempting to connect to ELM327...");

  if (!myELM327.begin(mySerial, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: ");
    Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}
O/P:
Couldn't connect to OBD scanner