PowerBroker2/ELMduino #59: Help with Reading speed (KPH) instead of RPM

jmandradiar, 2021-02-09:
Hi,
I am a beginner here.

I am interested in using the ESP32_Bluetooth_Serial code. However I want to read speed (KPH) instead of rpm, so can you please help me change the necessary parameters to read the car speed?
(I am getting confused whether to use the term "speed" or "kph")

PowerBroker2, 2021-02-09:
Do you want engine speed or vehicle speed? If you want vehicle speed in kph, you can use the `kph()` function

jmandradiar, 2021-02-09:
I want vehicle speed in kph.
It would be great if you could modify your same ESP32 bluetooth code,
changing the few lines alone for vehicle speed. Would that be possible?

(Just to be safe as I doing it with my car, it's my humble request)

PowerBroker2, 2021-02-09:
Untested and uncompiled:

```C++
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t kph = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempKPH = myELM327.kph();

  if (myELM327.status == ELM_SUCCESS)
  {
    kph = tempKPH;
    Serial.print("RPM: "); Serial.println(kph);
  }
  else
    printError();
}

void printError()
{
  Serial.print("Received: ");
  for (byte i = 0; i < myELM327.recBytes; i++)
    Serial.write(myELM327.payload[i]);
  Serial.println();
  
  if (myELM327.status == ELM_SUCCESS)
    Serial.println(F("\tELM_SUCCESS"));
  else if (myELM327.status == ELM_NO_RESPONSE)
    Serial.println(F("\tERROR: ELM_NO_RESPONSE"));
  else if (myELM327.status == ELM_BUFFER_OVERFLOW)
    Serial.println(F("\tERROR: ELM_BUFFER_OVERFLOW"));
  else if (myELM327.status == ELM_UNABLE_TO_CONNECT)
    Serial.println(F("\tERROR: ELM_UNABLE_TO_CONNECT"));
  else if (myELM327.status == ELM_NO_DATA)
    Serial.println(F("\tERROR: ELM_NO_DATA"));
  else if (myELM327.status == ELM_STOPPED)
    Serial.println(F("\tERROR: ELM_STOPPED"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_TIMEOUT"));
  else if (myELM327.status == ELM_TIMEOUT)
    Serial.println(F("\tERROR: ELM_GENERAL_ERROR"));

  delay(100);
}

```

jmandradiar, 2021-02-09:
Thanks a lot.

PowerBroker2, 2021-02-09:
No prob!

closed 2021-02-09 by PowerBroker2