PowerBroker2/ELMduino #63: V2.5 does not compile on ESP32

chrasim, 2021-03-03:
Hello,

version 2.5 has an issue compiling on ESP32 Arduino:

C:\Users\chris\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp: In member function 'uint64_t ELM327::findResponse()':
C:\Users\chris\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp:2571:27: error: call of overloaded 'println(uint64_t&)' is ambiguous
    Serial.println(response);
                           ^
In file included from C:\Users\chris\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Stream.h:26:0,
                 from C:\Users\chris\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:147,
                 from C:\Users\chris\Documents\Arduino\libraries\ELMDuino\src\ELMduino.h:2,
                 from C:\Users\chris\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp:1:
C:\Users\chris\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h:91:12: note: candidate: size_t Print::println(char)
     size_t println(char);
            ^
C:\Users\chris\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h:92:12: note: candidate: size_t Print::println(unsigned char, int)
     size_t println(unsigned char, int = DEC);
            ^
C:\Users\chris\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Print.h:93:12: note: candidate: size_t Print::println(int, int)
     size_t println(int, int = DEC);

Kind regards

Christoph

PowerBroker2, 2021-03-03:
Terribly sorry about that - I should've caught that before tagging that version. Check out [2.5.1](https://github.com/PowerBroker2/ELMduino/releases/tag/2.5.1), it should work for you. Thanks!

closed 2021-03-03 by PowerBroker2

chrasim, 2021-03-04:
Thanks a lot!