PowerBroker2/ELMduino #64: No Data - protocol 5 - ISO_14230_FAST_INIT

basswe, 2021-03-05:
Hello,
I'am using a esp32-wroom-32 (wemos d1 esp32).
I run the 'esp-serial-example'.
At first with my motorbike (proto 5) i get a time out, also in my car. (CAN proto)

So i changed  the timeout in ELMduino.h to : uint16_t timeout_ms = 5000; //1000;
In my car I read the data now, but my motor bike gives 'NODATA'.
Any help would be apriciate.
Bastiaan

esp32-wroom-32 / elmduino 2.41
Latest 2.51 did not compile.

PowerBroker2, 2021-03-05:
> Latest 2.51 did not compile.

That's very concerning to me - what's the compiler error (in full)?

basswe, 2021-03-05:
Sorry, I forgot to tell, that if I'am using my mobile with a bt-terminal or Torque, I receive data. (with version 2.41)
The compiler error with v.2.51, I will check it later.

basswe, 2021-03-06:
Now with the example of V2.5.1:

My adaptor : (OBDLink LX - STBTCOD 001F00->No Pincode)
My motorbike : URAL Sportsman 2019

After <ELM327 Connected> I added:
myELM327.initializeELM('5');
-----
Connecting on address: 
Result:
Connected to ELM327
Received: SEARCHING...
ERROR: ELM_TIMEOUT
Received: STOPPED
ERROR: ELM_STOPPED
Received: STOPPED
ERROR: ELM_STOPPED
Received: SEARCHING...

Changed the timeout from 1000 to 5000 (ELMDuino.h line 301)

Result:
Connected to ELM327
Received: NODATA
ERROR: ELM_NO_DATA
Received: NODATA
ERROR: ELM_NO_DATA
Received: NODATA

What else can I try to get it working?

Thanks in advance Bastiaan

PowerBroker2, 2021-03-06:
Try the ESP32 test sketch. Look at the header comments for the AT commands to manually use in the Serial Monitor and let me know what the ELM324 responds with. Also, are you testing with your car on and running?

basswe, 2021-03-07:
Thank you for the support,
testing with the AT commands gave as result that there was no data. 
Changed my OBD dongle, stil no data, even with Torque I had no data anymore.
Due the lotst of changes from OBD dongles was the signal ground inside the obd connector broken, so no solid ground.
After I placed another connector, take another clean wemos d1-esp32 , IT IS WORKING. with version 2.5.1.
Again thanks. Bastiaan (Samos Greece)
Sometimes you new a little push from somebody else. :-)

PowerBroker2, 2021-03-07:
Glad it's working - don't wory about it

closed 2021-03-07 by PowerBroker2

nomidaepapi, 2023-06-16:
@PowerBroker2  In the code where do I set the timer timeout to 5000?