PowerBroker2/ELMduino #65: Cant connect to ELM

iJuve, 2021-03-05:
Hey PowerBroker2, thank you for this very usefull library but i cant seem to connect to my ELM in any Baud, im using Uno with HC-05 and i2c display. 
It just cant connect.
Any ideas?
Thank you

PowerBroker2, 2021-03-05:
Are you using the latest version of the library? What HC-05 configuration are you using? Here's how I configured my HC-05 to talk to my ELM327: [Click Here](https://forum.arduino.cc/index.php?topic=600014.msg4159554#msg4159554).

Also, please post your sketch code in tags plus all errors and serial monitor printouts. They help a lot.

closed 2021-03-11 by PowerBroker2

iJuve, 2021-03-23:
Hi PowerBroker2 sorry for the wait, I tried to compile the code you used and i get this error

Serial2 not in scope

Arduino: 1.8.13 (Windows 10), Placa:"Arduino Uno"

C:\Users\Diogo\Desktop\sketch_mar23b\sketch_mar23b.ino: In function 'void setup()':

sketch_mar23b:19:3: error: 'Serial2' was not declared in this scope

   Serial2.begin(38400);

   ^~~~~~~

C:\Users\Diogo\Desktop\sketch_mar23b\sketch_mar23b.ino:19:3: note: suggested alternative: 'Serial'

   Serial2.begin(38400);

   ^~~~~~~

   Serial

C:\Users\Diogo\Desktop\sketch_mar23b\sketch_mar23b.ino: In function 'void loop()':

sketch_mar23b:37:5: error: 'Serial2' was not declared in this scope

     Serial2.write(c);

     ^~~~~~~

C:\Users\Diogo\Desktop\sketch_mar23b\sketch_mar23b.ino:37:5: note: suggested alternative: 'Serial'

     Serial2.write(c);

     ^~~~~~~

     Serial

sketch_mar23b:40:6: error: 'Serial2' was not declared in this scope

   if(Serial2.available())

      ^~~~~~~

C:\Users\Diogo\Desktop\sketch_mar23b\sketch_mar23b.ino:40:6: note: suggested alternative: 'Serial'

   if(Serial2.available())

      ^~~~~~~

      Serial

exit status 1

'Serial2' was not declared in this scope

Este relatório teria mais informação com a
opção «Mostrar mensagens detalhadas durante a
compilação» seleccionada nas Preferências.

PowerBroker2, 2021-03-23:
You're using the wrong example - [try this one](https://github.com/PowerBroker2/ELMduino/blob/master/examples/Arduino_software_serial_test/Arduino_software_serial_test.ino)

iJuve, 2021-03-23:
And do the AT commands to connect to the ELM327 with that code?

PowerBroker2, 2021-03-23:
What AT commands? AT commands for the HC-05 or ELM327?

iJuve, 2021-03-23:
AT commands to connect the HC-05 to the ELM327

PowerBroker2, 2021-03-24:
For that, check out this [link](https://forum.arduino.cc/index.php?topic=600014.msg4159554#msg4159554)