PowerBroker2/ELMduino #66: Using AT functions

MGC1STOKI, 2021-03-11:
Hi there, im new to ELM327s but im a big fan of your library!

How can i use the AT commands, is there a specific function in your library for this?

Currently my 2 uses for these would be to a) ask for the device ID and use its response as a way of checking if the device is connected and b) will be to change/increase the baud rate

Many thanks for any help you can provide as im still very much learning!

PowerBroker2, 2021-03-11:
Since they are global constants, you can use them directly in your sketch. For example, if you want to turn adaptive timing off via AT commands you can simply call the following in your sketch:

```C++
myELM327.sendCommand(ADAPTIVE_TIMING_OFF);
```

Note, though, that some of the AT commands require special inputs. For instance, if you want to assign a new device ID to the ELM327, you would have to also specify a new ID like this:

```C++
char command[10] = { '\0' };
int id = 3;
sprintf(command, STORE_DEVICE_ID, id);
myELM327.sendCommand(command);
```

Does that help?

MGC1STOKI, 2021-03-11:
Amazing thank you, whats the difference between your `myELM327.sendCommand("DISP_ID")` and for example a `myELM327.write("AT I")` would they achieve the same thing?

again excuse my novice approach to this

PowerBroker2, 2021-03-11:
You always want to use `DISP_ID` instead of `"AT I"` since the name `DISP_ID` is much more descriptive and easier to read. The same goes for all the other AT commands. You also should use `sendCommand()` instead of `write()` since `sendCommand()` ensures correct line ending, its return value can be used to determine if the ELM327 responded without error, and also parses the ELM327's response automatically.

If using `sendCommand()`, you'll have to use the class's `payload` char array pointer to access the ELM327's response:

```C++
if (myELM327.sendCommand(DISP_ID) == ELM_SUCCESS)
{
  Serial.println(myELM327.payload);
}
```

closed 2021-03-13 by PowerBroker2