PowerBroker2/ELMduino #67: PID Fuel System Status

technic98, 2021-03-13:
Hello,
I need a example to read the fuel system status. 
I tried:
`if (myELM327.queryPID(SERVICE_01, FUEL_SYSTEM_STATUS))
 {
      if (myELM327.status == ELM_SUCCESS)
      {
           int fuelSystem = myELM327.findResponse();
       }
}`
But there is no valid response.
Can someone give me a short example.

PowerBroker2, 2021-03-13:
```C++
  uint16_t status;
  uint16_t tempStatus = myELM327.fuelSystemStatus();

  if (myELM327.status == ELM_SUCCESS)
  {
    status = tempStatus;
    Serial.print("RPM: "); Serial.println(status);
  }
  else
    myELM327.printError();
```

technic98, 2021-03-13:
Thank you. 
Is the return Fuel System Status #1 or #1 and #2 together?
[https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01_PID_03](url)

PowerBroker2, 2021-03-14:
I don't understand the question, but the function returns the values consistent with  [this encoding](https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01_PID_03)

technic98, 2021-03-14:
The request for this PID returns 2 bytes of data. 
First Byte for Fuel System Status #1 and the second Byte for Fuel System Status #2.
How can I get the first Byte Data to read Fuel System Status #1?

PowerBroker2, 2021-03-14:
The code snippet returns a 16-bit number (2 bytes) that includes both bytes you're looking for. The order of bits/bytes is consistent with [https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01_PID_03](https://en.wikipedia.org/wiki/OBD-II_PIDs#Service_01_PID_03)

closed 2021-03-14 by PowerBroker2