PowerBroker2/ELMduino #68: 2.5.xx memory usage on Uno

crispyjones, 2021-03-14:
I'm building a temp gauge that uses a small 128x32 oled. When I updated the Elmduino libraries from 2.4.1 to anything higher they no longer fit on my arduino Uno. When I compile in 2.5.xx my global variable memory usage is over 84% and kicks low memory warning, when I roll back to 2.4.1 (same sketch) it goes to 50% and compiles/loads without issue.

First off thanks so much for making such an easy to use library. I realize this is mostly a "me problem" and that the Uno is the bare bones end of this platform, and you can't account for how terrible I am at coding. I wanted to post in case someone else runs across the issue or this was not a side effect of the update you were aware of. Thanks again. 

BTW 2.4.1 seems to work just fine for what I'm doing

PowerBroker2, 2021-03-14:
2.5.0 was when I added the debug mode and migrated the `printError()` function into the library (was at the sketch level previously). All those hardcoded strings are most likely the issue. I imagine if you did the `F()` macro around all the debug strings, you'd probably go back to the 50% dynamic memory usage (although your program memory usage will increase). Maybe give that a shot

crispyjones, 2021-03-14:
That did it, back down to 53%. The debug feature is great, that's worth the upgrade. 

Wanted to mention this emulator https://github.com/Ircama/ELM327-emulator, it emulates the car side, so elmduino thinks its talking to a vehicle. Very handy when its too cold to keep running out to the garage to test things. Also handy to test extreme vehicle conditions, don't really want to overheat my car just to test if the part of my code that flashes the temp reading actually works.

closed 2021-03-14 by PowerBroker2