PowerBroker2/ELMduino #69: ELM clone issues

crispyjones, 2021-03-14:
**Describe the bug**
To be clear, this is a bug report on the OBD2 hardware, I believe the PowerBroker library likely deals with a real ELM327 chip correctly.

Many people are likely using cheap ELM327 clones, they appear to respond incorrectly to some AT init sequences. They may also respond with non-ELM text response

**To Reproduce**
Send ATZ command (only ATZ no other init sequences) and see what your ELM device responds with. My device responds with "ELM327 v1.5".  looking at the ELM website [datasheet page](https://www.elmelectronics.com/products/dsheets/) this never exited. My device is some sort of clone/counterfeit ELM327.

**Expected behavior**
the elmduino .cpp file initializes with devices with this sane sequence
ATZ (reset)
AT E0 (echo off)
AT S0 (space off)
AT AL (allow long)
AT TP A0 (automatically try protocols starting at 0 and when connected write to memory)

The last line AT TP A0 locks up my ELM327 and it becomes completely unresponsive requiring a power cycle. I believe the expected behavior would be to respond to a command with "NO DATA". I only have this particular clone, so I can't be sure how a real ELM behaves.

**Equipment**
Probably any sub $15 amazon scanner is likely some sort of clone.

**Additional Context**

My clone *can* deal with the following strings without locking up
AT TP 0 (without the AUTO option)
AT TP A(!0) (starting with any protocol *except* 0)
AT SP 0
AT SP A0

Additionally my clone responds to commands in the unconnected (ie adapter on but not connected to vehicle) with "NOT CONNECTED".  I don't see this documented anywhere and I think it should be replying with "NO DATA", again, I don't have a real ELM327 to test so I'm guessing.

When I have time I'm going to fiddle with the init sequence in .cpp file as well as see if I can make it handle "NOT CONNECTED" sanely. I would also suggest that the behavior of a user specifying a protocol like this `myELM327.begin(ELM_PORT,true,2000,'#')` should be changed. That currently would send AT TP A#, which if 0 is selected on my device results in a lockup. I think it would be safe to assume the user knows the protocol they want and send AT SP #.

I suspect this is causing a lot of the connection/time out issues people are seeing. I have some investigating of my particular clone to do, but I would suggest asking what "version" peoples ELM327 respond with as part of standard practice. 

PS, thanks for updating your library so quickly, it let me use the debug function which made me realize why my code was connecting to the emulator perfectly but not my adapter.

crispyjones, 2021-03-15:
I think my counterfeit ELM327 v1.5 lacks memory. Every time I power cycle it and issue ATDP it says AUTO, ISO 15765-4 (CAN11/500). I can issue AT SP (any_valid#) and it will set the protocol for that session, but a reboot sends it back. I tried ATM1 and then ATSP1, and it still went back to auto after a reboot. I don't believe the M1 command should be necessary, the docs say that SP# should write to the non-volatile memory. The docs also state that protocol 0 isn't written to memory with ATSP0, ATSPA0, ATSPT0, or ATSPTA0, unless the AT SP00 command is issued (which they recommend against). Perhaps this is the issue with AT TP A0, it already is in AUTO and that crashes it?

I would be interested for others to test their ELM327s and see how they behave. I assume a real ELM327 sent AT SP 4 will powerup and reply ISO 14230-4 KWP (5 baud init, 10.4 kbit/s) to ATDP until something else is written to it.

I've changed line 2447 to
	if ((nextIndex(payload, "NODATA") >= 0) || (nextIndex(payload, "NOTCONNECTED"))
and it now treats NOTCONNECTED as a NODATA state.

rp2o2, 2021-03-22:
Hello, Im currently trying to work on a project that uses leds to show my cars rpm. I believe I got the AT+Commands working and the hc-05 is connected to the elm327. And I'm obviously using the ELMduino library but having issues to get it to connect. Im using the example code "Arduino_Software_Serial_test" and as it is connecting it gets through 6 of the protocols and then messes up the next protocol. Then it says it cannot connect. Do you know what the problem is?

///////////////////////////////////////////////////////////////////////////////////
Attempting to connect to ELM327...
Clearing input serial buffer
Sending the following command/query: AT Z
Attempting to connect to ELM327...
Clearing input serial buffer
Sending the following command/query: AT Z
All chars received: 
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT E0
All chars received: 
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT S0
All chars received: 
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT AL
All chars received: 
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT TP A0
All chars received: 
Timeout detected with overflow of 0ms
Setting protocol via AT TP A%c did not work - trying via AT SP %c
Clearing input serial buffer
Sending the following command/query: AT SP 0
All chars received: 
Timeout detected with overflow of 0ms
Setting protocol via AT SP %c did not work
Couldn't connect to OBD scanner
////////////////////////////////////////////////////////////////////////////////////

I have a 2008 Ford Fusion btw

PowerBroker2, 2021-03-22:
The ELM327 isn't responding in a way the Arduino can understand at all. The ELM327 is either connected to another bluetooth device, the HC-05 isn't actually connected to the ELM327, or you're using the wrong baud.

rp2o2, 2021-03-22:
So I believe it to be that my HC-05 is connected. I'm trying to reconnect it but my AT+LINK keeps failing. I've tried doing it under the different baud rates but it always fails. I've seen some of your responses to this particular issue and they do not seem to be working. I could be doing something wrong though.

PowerBroker2, 2021-03-22:
> So I believe it to be that my HC-05 is connected. I'm trying to reconnect it but my AT+LINK keeps failing.

Then it is not connected. Did you follow [these directions?](https://forum.arduino.cc/index.php?topic=600014.msg4159554#msg4159554)

rp2o2, 2021-03-22:
When I try to upload the code I get the error "exit status 1
'Serial2' was not declared in this scope" .

PowerBroker2, 2021-03-22:
In that case, you'll need to use a software serial port using SoftwareSerial.h and replace Serial2. There are plenty of good examples and tutorials online if you need help.

amerr97, 2023-06-26:
I fixed my issue with connection using 2pcb ELM327 PIC18F25K80

closed 2023-06-27 by PowerBroker2