PowerBroker2/ELMduino #72: Compile error occurs as update to latest version

nices0325, 2021-03-31:
Hello PowerBroker!! 
First of all, Thank you for the awesome library!
I was using elmduino v2.4.1

As I upadate to the lastest version of the library, compile error occurs.. what have changed of the new one..

I have found some changes in the sketch --> if (!myELM327.begin(ELM_PORT, true, 2000))
I think this occurs the compile error..

PowerBroker2, 2021-03-31:
Update to the latest version of the library

nices0325, 2021-04-01:
https://pastebin.com/R2CDHYAM

this is my sketch. I'm using esp32 board with this library.. when I update to 2.5.4 version compile error occurs.. I'm not sure what have changed with this new version.. I just exchange the src folder to my existing library thank you very much

PowerBroker2, 2021-04-01:
1.) Does your ESP32 have bluetooth? If so, why not use the built-in bluetooth functionality of the ESP32 instead of using a UART?
2.) I can't get your code to compile due to the many atypical and hard to find libraries you're using. To simplify things, test the appropriate [example sketch](https://github.com/PowerBroker2/ELMduino/tree/master/examples) and see if it compiles for you. If not, we have a problem, if the example compiles, you need to check your project sketch for inconsistencies

closed 2021-04-08 by PowerBroker2