PowerBroker2/ELMduino #73: Tank Fuel Level on Renault Kangoo

GiacomoGenovese, 2021-04-03:
Hi all,
I cannot read Fuel Level on Renault Kangoo (Diesel).
Did anyone have the same problem and solve it?

I'm using this OBD II reader: [https://www.amazon.it/Yizhet-Bluetooth-Scanner-Android-Dispositivo/dp/B082XWXC45/ref=sr_1_3_sspa?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=3IMCWW5QD97FB&dchild=1&keywords=obd2+elm327+bluetooth&qid=1617445185&sprefix=obd2+elm%2Caps%2C210&sr=8-3-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUFESkhERUVMMDBHOSZlbmNyeXB0ZWRJZD1BMDIzNjk5MEdRUlQ2WVo1NThVRyZlbmNyeXB0ZWRBZElkPUEwMTgyODY1Uko2NTlLUFVKU0dJJndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ==](url)

Thanks in advance for any support you'll give.
Best

PowerBroker2, 2021-04-03:
> Thanks in advance for any support you'll give.

First you'll need to give some support ;)

What Arduino are you using? What does your wiring look like? Does your Arduino connect properly to the ELM327? Are you able to get the example sketch to work and display RPM? If not, what have you tried to do to fix it? If so, what does your project sketch look like (post in full and in tags)? What year is your car?

GiacomoGenovese, 2021-04-03:
Hi,
thank you for your fast response.

Thanks to your code, I've got all request and get all the value in a FIAT (rpm, Tank level, speed, etc...) as well as for some friend's cars like Jeep Renegade and Honda Jazz (2010). When I try the same in the Renault Kangoo (2019, diesel) I get speed and rpm but tank level. the value is not available.

PowerBroker2, 2021-04-03:
Ok, glad you have a good connection to the ELM327 (that's usually the most difficult part) - Can you post your code (in tags) plus a printout of the debug statements. Make sure the library's class is set in debug mode to get the printouts:

https://github.com/PowerBroker2/ELMduino/blob/4eddfb758813180371d36399985fd7f92c00a178/src/ELMduino.cpp#L6-L44

GiacomoGenovese, 2021-04-10:
I had to wait for the weekend to work on it.
Here the debug. `Thanks`

```
16:40:12.795 -> Clearing input serial buffer
16:40:12.795 -> Sending the following command/query: AT D
16:40:12.928 -> 	Received char: \r
16:40:12.928 -> 	Received char: O
16:40:12.928 -> 	Received char: K
16:40:12.928 -> 	Received char: \r
16:40:12.928 -> 	Received char: \r
16:40:12.928 -> 	Received char: >
16:40:12.928 -> Delimiter found
16:40:12.928 -> All chars received: OK
16:40:12.928 -> Clearing input serial buffer
16:40:12.928 -> Sending the following command/query: AT Z
16:40:12.928 -> 	Received char: A
16:40:12.928 -> 	Received char: T
16:40:12.928 -> 	Received char:  
16:40:12.928 -> 	Received char: Z
16:40:12.928 -> 	Received char: \r
16:40:13.788 -> 	Received char: \r
16:40:13.788 -> 	Received char: \r
16:40:13.788 -> 	Received char: E
16:40:13.788 -> 	Received char: L
16:40:13.788 -> 	Received char: M
16:40:13.788 -> 	Received char: 3
16:40:13.918 -> 	Received char: 2
16:40:13.918 -> 	Received char: 7
16:40:13.918 -> 	Received char:  
16:40:13.918 -> 	Received char: v
16:40:13.918 -> 	Received char: 1
16:40:13.918 -> 	Received char: .
16:40:13.918 -> 	Received char: 5
16:40:13.918 -> 	Received char: \r
16:40:13.918 -> 	Received char: \r
16:40:13.918 -> 	Received char: >
16:40:13.918 -> Delimiter found
16:40:13.918 -> All chars received: ATZELM327v15
16:40:13.918 -> Clearing input serial buffer
16:40:13.918 -> Sending the following command/query: AT E0
16:40:14.017 -> 	Received char: A
16:40:14.017 -> 	Received char: T
16:40:14.017 -> 	Received char:  
16:40:14.017 -> 	Received char: E
16:40:14.017 -> 	Received char: 0
16:40:14.017 -> 	Received char: \r
16:40:14.017 -> 	Received char: O
16:40:14.017 -> 	Received char: K
16:40:14.017 -> 	Received char: \r
16:40:14.017 -> 	Received char: \r
16:40:14.017 -> 	Received char: >
16:40:14.017 -> Delimiter found
16:40:14.017 -> All chars received: ATE0OK
16:40:14.111 -> Clearing input serial buffer
16:40:14.111 -> Sending the following command/query: AT S0
16:40:14.191 -> 	Received char: O
16:40:14.191 -> 	Received char: K
16:40:14.191 -> 	Received char: \r
16:40:14.191 -> 	Received char: \r
16:40:14.191 -> 	Received char: >
16:40:14.191 -> Delimiter found
16:40:14.191 -> All chars received: OK
16:40:14.278 -> Clearing input serial buffer
16:40:14.278 -> Sending the following command/query: AT AL
16:40:14.278 -> 	Received char: O
16:40:14.278 -> 	Received char: K
16:40:14.278 -> 	Received char: \r
16:40:14.278 -> 	Received char: \r
16:40:14.360 -> 	Received char: >
[... 271 lines cut ...]
16:41:55.354 -> 	Received char: 0
16:41:55.354 -> 	Received char: D
16:41:55.354 -> 	Received char: 0
16:41:55.354 -> 	Received char: 0
16:41:55.354 -> 	Received char: \r
16:41:55.517 -> 	Received char: \r
16:41:55.517 -> 	Received char: >
16:41:55.517 -> Delimiter found
16:41:55.517 -> All chars received: 410D00
16:41:55.517 -> Expected response header: 410D
16:41:55.517 -> Single response detected
16:41:55.517 -> 64-bit response: 
16:41:55.517 -> 	responseByte_0: 0
16:41:55.517 -> 	responseByte_1: 0
16:41:55.517 -> 	responseByte_2: 0
16:41:55.517 -> 	responseByte_3: 0
16:41:55.517 -> 	responseByte_4: 0
16:41:55.517 -> 	responseByte_5: 0
16:41:55.517 -> 	responseByte_6: 0
16:41:55.557 -> 	responseByte_7: 0
16:41:55.557 -> RPM: 850
16:41:55.557 -> Tank Level: 0
16:41:55.557 -> Vehicle Speed: 0
16:41:56.047 -> ........................
```

PowerBroker2, 2021-04-10:
Since you are getting a `NO_DATA` error, the car does not support that PID. Sorry, but all I can do to help is point you to some online resources:

[https://mechanics.stackexchange.com/questions/13153/obd-2-fuel-reading-says-no-data](https://mechanics.stackexchange.com/questions/13153/obd-2-fuel-reading-says-no-data)

[https://stackoverflow.com/questions/44870312/how-to-get-obd2-fuel-level-reading-from-wagon-r-2010-model](https://stackoverflow.com/questions/44870312/how-to-get-obd2-fuel-level-reading-from-wagon-r-2010-model)

[https://mechanics.stackexchange.com/questions/53418/calculating-current-fuel-level-with-obd-data](https://mechanics.stackexchange.com/questions/53418/calculating-current-fuel-level-with-obd-data)

closed 2021-04-10 by PowerBroker2