PowerBroker2/ELMduino #74: Transmition rate

Niv1900, 2021-04-04:
Hello @PowerBroker2, 

I just discovered this awesome library ! 

In the most basic example I was wondering if I can speed up the communication between my ESP32 and the Bluetooth ELM327 that I use. 

For instance, when I use this example : 

```
#include <Arduino.h>
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT SerialBT
#define DEBUG_PORT Serial
#define LED_BUILTIN 2

ELM327 myELM327;

char output[300];

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
    pinMode(LED_BUILTIN, OUTPUT);
    digitalWrite(LED_BUILTIN, LOW);
#endif

    DEBUG_PORT.begin(115200);
    ELM_PORT.begin("OBDII", true);
    SerialBT.setPin("1234");

    if (!ELM_PORT.connect("OBDII"))
    {
        DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
        while (1)
            ;
    }

    if (!myELM327.begin(ELM_PORT, false, 1000))
    {
        Serial.println("Couldn't connect to OBD scanner - Phase 2");
        while (1)
            ;
    }

    Serial.println("Connected to ELM327");

    digitalWrite(LED_BUILTIN, HIGH);
}

void loop()
{
    float tempRPM = myELM327.rpm();
    float load = myELM327.engineLoad();
    float temp = myELM327.engineCoolantTemp();
    int32_t kph = myELM327.kph();
    float throttle = myELM327.throttle();
    float fuel = myELM327.fuelLevel();
    float airTemp = myELM327.ambientAirTemp();
    float oilTemp = myELM327.oilTemp();

    if (myELM327.status == ELM_SUCCESS)
    {
        sprintf(output, "%06.3f,%06.3f,%06.3f,%06.3f,%06.3f,%06.3f,%06.3f,%06.3f",tempRPM,load,temp,kph,throttle,fuel,airTemp,oilTemp);
        Serial.println(output);
    }
}
```

I'v got some new value every two seconds : 

```
19:11:00.587 -> 706.000,25.882,54.000,01.784,02.353,00.000,-40.000,-40.000
19:11:02.382 -> 705.000,25.882,54.000,01.784,02.353,00.000,-40.000,-40.000
19:11:04.120 -> 702.000,24.706,54.000,01.784,02.353,00.000,-40.000,-40.000
19:11:05.925 -> 712.000,23.922,54.000,01.784,02.353,00.000,-40.000,-40.000
19:11:07.696 -> 712.000,22.745,55.000,01.784,02.353,00.000,-40.000,-40.000
19:11:09.471 -> 704.000,22.745,54.000,01.784,03.137,00.000,-40.000,-40.000
19:11:11.239 -> 703.000,23.922,55.000,01.784,02.745,00.000,-40.000,-40.000
19:11:13.013 -> 711.000,22.745,55.000,01.784,02.353,00.000,-40.000,-40.000
19:11:14.818 -> 699.000,22.745,55.000,01.784,02.353,00.000,-40.000,-40.000
19:11:16.561 -> 705.000,22.745,56.000,01.784,02.353,00.000,-40.000,-40.000
19:11:18.337 -> 693.000,22.745,56.000,01.784,02.353,00.000,-40.000,-40.000
19:11:20.109 -> 695.000,21.961,56.000,01.784,02.353,00.000,-40.000,-40.000
19:11:21.881 -> 697.000,21.961,56.000,01.784,02.745,00.000,-40.000,-40.000
19:11:23.683 -> 00.000,00.000,56.000,01.784,05.882,00.000,-40.000,-40.000
19:11:25.493 -> 00.000,00.000,56.000,01.784,05.882,00.000,-40.000,-40.000
19:11:27.266 -> 00.000,00.000,57.000,01.784,05.882,00.000,-40.000,-40.000
19:11:29.043 -> 00.000,00.000,57.000,01.784,06.667,00.000,-40.000,-40.000
```

Is this related to this issue : #19 ? 

I have made a quick test with the dual core example but I'm not sure it will solve my goal to get 10Hz readings on several variables. 

I tried to search for this question, without luck, sorry if it's already been asked.

PowerBroker2, 2021-04-04:
Yes, the way this library works, it has to block the code until it gets a full response from the ELM327. This means, however, that the library operates as fast as the ELM327 can respond, meaning the limiting factor actually is the ELM327 itself. If your ELM327 responded faster, you'd get a higher sample/printout rate.

Does that make sense?

Niv1900, 2021-04-05:
100% clear, thank you so much !

closed 2021-04-05 by Niv1900